iiab/iiab #660: web interface to upload books into Calibre 3.16 - how to make it work?

holta, 2018-01-28:
Error msg in browser:
```
Failed to upload the file: SAMPLEBOOK.pdf
You must be logged in to make changes to the library

Failed to communicate with "/cdb/add-book/2/n/SAMPLEBOOK.pdf/calibre", with status: [403] Forbidden

Anonymous users are not allowed to make changes
```
Is this more serious that a permissions issue, when the above directory (/cdb) does not exist on Raspbian 2017-11-29 where Calibre 3.16 is installed?

In any case, https://calibre-ebook.com/whats-new says:
```
New features [in Calibre 3.16]

Content server: Allow adding and deleting of books using the web interface

Use the + icon in the top bar of the book list to add new books and the trash icon in the top bar of the book details page to a delete a book. Note that only authenticated users are allowed to perform these actions, so you have to setup user accounts for the server to use these features.

holta, 2018-01-28:
This had looked *extremely* promising:

- Calibre Desktop Application -> Preferences -> Sharing over net -> User accounts -> (add Admin/changeme as username/password)
- systemctl restart calibre-serve

But it did not work (the http://box:8080 error remains the same, as pasted in above, on IIAB 6.5/master on Raspbian Desktop 2017-11-29...)

tim-moody, 2018-01-28:
I would be inclined to believe it when it says you are not logged in. The directory is probably defined in apache and not physical.

floydianslips, 2018-01-28:
I created a user and logged in as the user and was able to add content. It is not the ideal way probably but it worked. 
create user
calibre-server --userdb /location you want the userdb/users.sqlite --manage-users

calibre-server --port 8080 /library/calibre/ --userdb /location of newly created userdb/users.sqlite --enable-auth

systemctl start calibre-serve.service

You will need to log in using the newly created credentials in a pop up window in the browser, but you can use the new upload button.

holta, 2018-01-29:
@floydianslips saved the day, at several Mexican schools, thank you.

FYI I used /library/calibre/users.sqlite

And I created these 2 users:

- box/box (for kids)
- Admin/changeme (for teachers)

holta, 2018-01-29:
Aside: Calibre's client software is quite incredible (Mexican teachers want to use it on Raspbian Desktop and Windows etc!) BUT they need to be warned that it is configuring web users in the wrong place for IIAB:

- /home/pi/.config/.../calibre-server.sqlite
- /root/.config/.../calibre-server.sqlite

So I am asking teachers to stick with Admin/changeme when logging into http://box:8080 for now.  Also the iiab-menu file needs to be fixed from 8010 to 8080 within /opt/iiab/iiab-menu/config.json

tim-moody, 2018-01-29:
changing /opt/iiab/iiab-menu/config.json will only work before first install and will block git pulls. after change /library/www/html/iiab-menu/config.json.

if idmgr is gone for good, then we could default to 8080.

holta, 2018-02-02:
Keeping this tkt open so we can try to:
- Automate username/password creation of box/box and Admin/changeme
- Revise port 8010 to 8080 in /opt/iiab/iiab-menu/config.json (and hence /library/www/html/iiab-menu/config.json) — is it possible/realistic to read this variable from local_vars.yml during installation?
- Is it possible you get Calibre's client software to modify our own /library/calibre/users.sqlite instead of it's default in /home/pi/.config/.../calibre-server.sqlite or /root/.config/.../calibre-server.sqlite ?
- Explore why http://box/books doesn't quite work as http://box:8080 does

holta, 2018-06-12:
Evolving as part of #747 -> #830

closed 2018-06-12 by holta