iiab/iiab #661: IIAB 6.5 microSD's cannot be moved from RPi3 to RPi Zero W?

holta, 2018-01-29:
Mexico's various 128GB SanDisk IIAB 6.5 microSD's cannot be moved from RPi3 to RPi Zero W: the symptom is that the SSID ("Internet in a Box") is visible but client machines cannot connect.

Whereas this worked in June of 2017, with Haiti's 64GB GB microSD's.

Please help investigate, those who can!

holta, 2018-02-04:
Clarif: it's entirely possible that Raspbian Desktop (as used in Mexico, underlying IIAB) never ran properly on RPi Zero W.

But whatever the underlying reason: RPi Zero W's inability to deliver WiFi AP Mode (i.e. a functional hotspot, works in certain conditions but not others) needs investigation...

holta, 2018-02-05:
Raspbian Desktop from earlier in 2017 (IIAB 6.2 or 6.3) works to create a fully functional hotspot in an RPi Zero W.

Hypothetically it's possible that WiFi-installed IIAB 6.5 (all I've tested so far, inside a RPi Zero W) continues to behave differently from Ethernet-installed IIAB 6.5 ?

In any case: thanks all who can evaluate what works on RPi Zero W and what doesn't!

tim-moody, 2018-02-10:
installed 6.5 on rpi3 using medium over wire
moved to W
no ssid
accessed over usb wire OK
hostapd not running; started it
able to ssh in over wifi
over wifi connection reset on http
same over wired connection

holta, 2018-02-10:
Is it possible that all 3 Raspbian Stretch releases (2017-08-16, 2017-09-07, 2017-11-29) are themselves causing underlying incompatibility with RPi Zero W?

tim-moody, 2018-02-10:
medbox is based on 2017-09-07 and I just reverified that it works on W

tim-moody, 2018-02-10:
2017-11-29 is a suspect. seems there were extensive changes and nothing new in 2 1/2 months could represent struggles

holta, 2018-02-10:
> medbox is based on 2017-09-07 and I just reverified that it works on W

Good news.

> 2017-11-29 is a suspect

Can you apply all "apt update; apt dist-upgrade" patches to a copy of medbox and see if it still works in a RPi Zero W?

Installing IIAB 6.4 onto 2017-11-29 could also be a valuable test?

tim-moody, 2018-02-10:
N.B. firmware-libertas firmware-misc-nonfree firmware-realtek in list of upgrades

tim-moody, 2018-02-10:
took content-ready 6.4 and ran apt update and dist-upgrade on rpi3 then moved to W
it booted and I accessed it via ssh and http OK

6.5 on 9-7 seems more promising

tim-moody, 2018-02-10:
root@raspberrypi:/opt/iiab/iiab# ./iiab-install
Placed /etc/ansible/facts.d/local_facts.fact into position.
Found Kernel 4.9.41-v7+

EXITING: Kernel 4.9.59-v7+ or higher required with Raspbian.
PLEASE RUN 'apt update' then 'apt install raspberrypi-kernel' then reboot.
IIAB INSTALL INSTRUCTIONS: https://github.com/iiab/iiab/wiki/IIAB-Installation

tim-moody, 2018-02-10:
raspberrypi-kernel is already the newest version (1.20171029-1).

so waste of time

holta, 2018-02-10:
> EXITING: Kernel 4.9.59-v7+ or higher required with Raspbian.

If nec this check can be commented out from /iiab-install (it's in place for KA Lite).

holta, 2018-02-10:
>> EXITING: Kernel 4.9.59-v7+ or higher required with Raspbian.
>
> If nec [during testing] this check can be commented out from /iiab-install (it's in place for KA Lite).

Correction: @jvonau asked for this minimum kernel check for bridging @ https://github.com/iiab/iiab/issues/482 ("Raspbian kernel update breaks bridging").

Whereas the KA Lite issue (https://github.com/iiab/iiab/issues/392, resolved by Been Nuttall) appears only tangentially related.

holta, 2018-03-12:
FYI IIAB MicroSD cards can be moved back and forth between RPi 3 and RPi Zero W using PR #697 (anticipated to become the foundation of IIAB 6.6, starting sometime next week!)

holta, 2018-04-26:
@deldesir confirms this is no longer an issue &mdash; _profound thanks to @jvonau and @georgejhunt whose very hard work made this possible in recent weeks._

closed 2018-04-26 by holta