iiab/iiab #662: Teacher want Accents to work when searching for cities in OpenStreetMap

holta, 2018-01-29:
See @georgejhunt heroic fix @ #659 that goes 90% of the way, but leaves teachers frustrated when their city shows up (autocomplete works!) but something about the geo-location fails, such that a map of their city never appears.

This is based on the 10.7GB National Geographic tiles from http://dev.worldpossible.org/cgi/viewmod.pl?module_id=100

tim-moody, 2018-01-29:
if this is a bug it is at source. remember that the search db only has the 5000 largest cities in the world.

I don't see that auto complete works on iiab.me. try León. maybe no accents in db. oaxaca search returns full name with an accent that fails to match to a map. just entering oaxaca goes to the map.

holta, 2018-01-30:
Autocomplete works at iiab.me/modules/en-worldmap/map.html if you enter "Acap" or "acap"

Observe the 3 autocomplete results — some of which are clickable — but not Acapulco, as it's full name contains with accent(s).

This is an upstream bug no doubt, but universally requested by teachers in different states in Mexico.

(Let's also consider upgrading from 5000 cities to 15,000 cities — which Mexican schools would appreciate greatly if it's doable!)

tim-moody, 2018-01-30:
the original iiab maps does know the difference between León and Leon

tim-moody, 2018-01-30:
sqlite> select * from features where name like 'oaxaca%';
3522507|Oaxaca de Juárez|Oaxaca de Juarez|Byen Oaxaca,Guaxaca,Huaxyacac Juarez,Huaxyacac Juárez,Lungsod ng Oaxaca,OAX,Oachaka,Oachaka nte Chouares,Oahaka,Oakhaka,Oakhaka de Khuares,Oakhaka-de-Khuares,Oakhakae-de-Khuares,Oaksaka de Khuarez,Oaxaca,Oaxaca City,Oaxaca by,Oaxaca de Juarez,Oaxaca de Juárez,Oaxaka de Juarez,Oaxaka de Juárez,Oaxaka de Xuares,Santa Maria,awakhaka d khwarz,awkhaka,oagsaka,oahaka,vahaka de ju'ereza,vahaka de khu'arisa,vasaka de hu'arejha,wa ha ka,wa ha ka shi,Οαχάκα ντε Χουάρες,Оаксака де Хуарез,Оахакæ-де-Хуарес,Оахака,Оахака де Хуарес,Оахака-де-Хуарес,Օախակա դե Խուարես,אואחאקה,أوخاكا,اوآخاکا د خوارز,اوکساکا سٹی,ܘܐܟܐܩܐ ܕܝ ܟܘܐܪܝܣ,वाशाका दे हुआरेझ,वाहाका डे जुएरेज़,ਵਾਹਾਕਾ ਦੇ ਖ਼ੁਆਰਿਸ,วาฮากา,གའཀགྲོང་།,ოახაკა-დე-ხუარესი,ወሓካ ዴ ዋሬዝ,オアハカ,瓦哈卡市,오악사카|17.06542|-96.72365|P|PPLA|MX||20|067|||262566||1567|America/Mexico_City|2015-06-22

holta, 2018-01-30:
Amazing...how close we are?

I hope we can make kidproof geography a focus given every school I've talked to recently (elementary, middle, high) takes this very seriously.

Whether or not locally [comprehensive detail can be layered on](https://github.com/iiab/iiab-factory/tree/master/content/subset-osm), deepening these 10 base (National Geographic) layers within the local country/region!

 CC @m-anish

tim-moody, 2018-01-30:
rachel's search php is the source of the search problem. we need to define the search algorithm and then re-implement it. iiab's approach was better, but the ui is flawed. (actually we have not really tried Joel's subsequent changes.

how close are we? not very.

holta, 2018-01-30:
> how close are we? not very

Could we restrict (typing) accents (into the textfield search box) in the short term?

Since the cities are showing up correctly, except for the very first stage of visualizing them?

(And work with @jasco too, for the longer-term solution below?)

> rachel's search php is the source of the search problem. we need to define the search algorithm and then re-implement it. iiab's approach was better, but the ui is flawed. (actually we have not really tried Joel's subsequent changes.

holta, 2018-02-07:
> how close are we? not very

@tim-moody is it possible you misunderstood?  Accented cities ARE appearing correctly in the dropdown.

The only thing that's not happening is the very, very final step &mdash; displaying the MAP of such accented cities &mdash; i.e. when users click on those accented cities in the dropdown.

(This would seem to be a straightforward bug very possibly: if these accented cities' latitudes/longitudes are indeed known...seemingly these lat/long's are just getting messed up during the final tile-rendering stage?)

holta, 2018-02-08:
@tim-moody explained during our community/team call (http://minutes.iiab.io) that he sees no easy way to fix this, without reworking the PHP logic &mdash; or possibly hacking the database to include plaintext unaccented citynames as well?

In any case in Spanish-speaking countries, it would appear that something like 1/3 of all cities currently do not display as a result of this bug.

**Crude-but-useful Workaround for Teachers:** just type the first few letters of the city into the textfield / search box, while observing the dropdown/autocomplete.  Stop typing before it's too later, i.e. "oax" or "tuxtla" will show 2 cities that begin with those letters.  You can try this live at http://dev.worldpossible.org/mods/en-worldmap-10/map.html

Then hit Enter, causing both cities' locations to be correctly displayed, with a red teardrop over each!  _Not perfect but better than nothing :-)_

tim-moody, 2018-02-08:
this sounds like a good workaround and is even simpler than you say. type 'oaxaca' and press enter and the  map will be positioned to Oaxaca. But pause and a drop down will appear and selection of an item in that dropdown with an accent will fail. Basically, if you press enter the entire search list is marked on the map with the map positioned to the item at the top of the list. Try typing 'par' and enter and zoom out to see the whole world.

holta, 2018-02-08:
> type 'oaxaca' and press enter and the map will be positioned to Oaxaca

Good Catch: it tolerates even more letters, so that multiple cities need not be shown.

Conclusion: teacher training can showcase users typing in all letters **up to the 1st accent** &mdash; and then hit Enter (or click on the magnifying glass) &mdash; _but do not click on the actual accented city name!_

This doesn't solve all cases (i.e. when city names like "León" contain accents on the left side in the very 1st few letters, they're impossible to display) but still this workaround-for-teachers is a massive win, allowing geography classes to take this much more seriously.

holta, 2018-04-26:
IIAB 6.5 has brought us many improvements here.  Thanks to the many contributors who made this happen.

Hopefully an even more complete solution will allow students/teachers to find OSM cities that contain accents in many languages &mdash; even without these recent training/UX hacks &mdash; as part of IIAB 6.6 or 7.0.

holta, 2018-08-25:
This appears partially or largely solved by Vector-based OSM approach @ #877 ?!

holta, 2018-09-02:
Seem good enough!

(But please re-opened as nec!)

closed 2018-09-02 by holta