iiab/iiab #669: is Ansible 2.4.3 running out of memory when installing IIAB 6.5/master on RPi3?

holta, 2018-02-10:
If #666 is the devil's number, we can revert it :)

Quick summary (of memory running out when running the "network" role) @ https://github.com/iiab/iiab/pull/666#issuecomment-364667294

holta, 2018-02-10:
@tim-moody @georgejhunt can you paste in the error message onto this ticket if poss?

tim-moody, 2018-02-10:
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: OSError: [Errno 12] Cannot allocate memory

holta, 2018-02-10:
> OSError: [Errno 12] Cannot allocate memory

Unrelated to IIAB, others experienced this with Ansible 2.4.1:
https://github.com/ansible/ansible/issues/32854

And yet this is (alleged) to be improved in Ansible 2.4.3:
https://github.com/ansible/ansible/issues/31673

Thanks all for more reports helping us isolate the issue & how to reproduce it.

holta, 2018-02-10:
@jvonau : Ansible 2.5 was released in beta form yesterday @ https://releases.ansible.com/ansible/ so let's look into how we might pin scripts/ansible to 2.4.x possibly?

holta, 2018-02-11:
A workaround for RPi &mdash; and Ubuntu 16.04 LTS too if nec &mdash; until Ansible 2.4.3 is fixed:

> wget http://download.iiab.io/packages/ansible_2.4.2.0-1ppa~xenial_all.deb
> apt install ./ansible_2.4.2.0-1ppa~xenial_all.deb

I've cached Ansible 2.4.2's PPA above, as it's expected to disappear from https://launchpad.net/~ansible/+archive/ubuntu/ansible/+files/ansible_2.4.2.0-1ppa~xenial_all.deb shortly.

holta, 2018-02-12:
Excellent News: Ansible's @sivel replied to me @ https://github.com/ansible/ansible/issues/31673#issuecomment-365011700

He explains this memory allocation problem is likely https://github.com/ansible/ansible/issues/35796 which was fixed late last week and has been backported to stable-2.4 for an upcoming 2.4.4 release.  It is also in stable-2.5 for the 2.5.0.b1 release (and in their devel branch).

holta, 2018-02-12:
FYI Ansible 2.5.0's target release is March 2018, according to the official roadmap/checklist:
http://docs.ansible.com/ansible/devel/roadmap/ROADMAP_2_5.html

[January 23rd's Feature Freeze annc](https://groups.google.com/forum/#!topic/ansible-devel/vGCQQY5472s) from the release manager reaffirms this general schedule:
- 2.5.0 b1 was released last Friday (Feb 9)
- 2.5.0 b2 is scheduled for Thur (Feb 15)
- 2.5.0 rc1 is scheduled for next Thur (Feb 22)

holta, 2018-02-15:
This tkt can be re-opened later if absolutely nec:

IIAB 6.5 should be released in sometime in March—but it remains an open question if with Ansible 2.4.4 or 2.5.0 (or whatever) will be ready at that time.

holta, 2018-03-12:
Am re-opening this tkt as [Ansible 2.4.4 and 2.5.0](https://releases.ansible.com/ansible/) release appear imminent in coming weeks, regardless how this affects IIAB 6.5 and 6.6

holta, 2018-03-16:
Ansible 2.5 RC3 released:

https://releases.ansible.com/ansible/ansible-2.5.0rc3.tar.gz

holta, 2018-03-25:
Error has returned on Ansible 2.5.0 on 2018-03-13 on RPi3 &mdash; 38min into a run of "./iiab-install --reinstall" :
```
TASK [network : Copy /etc/sysconfig/dhcpd file] ********************************
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: OSError: [Errno 12] Cannot allocate memory
fatal: [127.0.0.1]: FAILED! => {"msg": "Unexpected failure during module execution.", "stdout": ""}
```

holta, 2018-04-07:
> why is 2.5 50% larger than 2.4.4 and does that affect memory usage?

No idea: Thx everybody for keeping an eye on this+other consequences of our transition from Ansible 2.4.x (IIAB 6.5) to Ansible 2.5.x+ (IIAB 6.6).

jvonau, 2018-04-07:
Anish reported the same crash under 2.4.4 also

> OK 16. ./ansible-2.4.x 
OK 17. cd /opt/iiab/iiab/
XX 18. ./iiab-install
TASK [network : Copy /etc/sysconfig/dhcpd file] *****************************************
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: OSError: [Errno 12] Cannot allocate memory
fatal: [127.0.0.1]: FAILED! => {"failed": true, "msg": "Unexpected failure during module execution.", "stdout": ""}
    to retry, use: --limit @/opt/iiab/iiab/iiab-stages.retry

holta, 2018-04-19:
> Anish reported the same crash under 2.4.4 on ubuntu 16.04 also

@m-anish can you retest with [Ansible 2.5.1](https://github.com/iiab/iiab/pull/740#issuecomment-382581867) released yesterday?  Which can be installed in either of 2 ways:
1. Run `/opt/iiab/iiab/scripts/ansible` (recommended!)
2. `pip install ansible==2.5.1`

holta, 2018-04-20:
@m-anish Ansible has had many memory leaks recently :) so I'm not sure which you ran into, but FYI the most recent fix (https://github.com/ansible/ansible/issues/38532#issuecomment-383040559) JUST missed inclusion in Ansible 2.5.1 released 2 days ago.  And so will be part of Ansible 2.5.2 expected "3 weeks later" by/around mid-May.

holta, 2018-04-26:
> @m-anish Ansible has had many memory leaks recently :) so I'm not sure which you ran into, but FYI the most recent fix (ansible/ansible#38532 (comment)) JUST missed inclusion in Ansible 2.5.1 released 2 days ago. And so will be part of Ansible 2.5.2 expected "3 weeks later" by/around mid-May.

@m-anish: Ansible 2.5.2 is now released fixing memory leaks, try it if poss!  (I've confirmed `./iiab-install --reinstall` runs [IIAB 6.5] to completion on 2.5.2 on Ubuntu 16.04...of course stick with Ansible 2.4.4 if you require ./runtags to work with multiple roles...instead of IIAB 6.6's new ./runrole)

holta, 2018-06-10:
Ansible 2.5.3 & Ansible 2.5.4 resolve even more memory issues for RPi, so these issues seem very much under control now.

closed 2018-06-10 by holta