iiab/iiab #684: search crashed kiwix-serve

holta, 2018-02-19:
### Expected behavior and actual behavior.

When any recent kiwix-serve (e.g. from 2017-12-27) @tim-moody can crash kiwix-serve repeatedly if not always by entering a search on "pneumoni" (without the final 'a' !) when searching within wikipedia_en_medicine_2017-10 or wikipedia_en_medicine_2018-01

(Press Enter immediately after typing in "pneumoni")

Sometimes it claims there is no FT index, which is false.

### Steps to reproduce the problem.

Might be a timing thing...intermittent :/

### Specifications like the version of the IIAB, operating system version, or hardware details.

Tested on Raspbian 2017-09-07.

@tim-moody asks if we should follow MySQL and add a restart:
# Restart crashed server only, on-failure would also restart, for example, when
# my.cnf contains unknown option
Restart=on-abort
RestartSec=5s

holta, 2018-02-19:
@georgejhunt apologies: on today's community/team call (http://minutes.iiab.io) Tim & the group are hoping you might have some recommendations reviewing this idea!

holta, 2018-02-21:
> @tim-moody asks if we should follow MySQL and add a restart:
> ```
> # my.cnf contains unknown option
> Restart=on-abort
> RestartSec=5s

@tim-moody might possibly have a PR ready for tmrw's community/team call (http://minutes.iiab.io)

holta, 2018-02-26:
@tim-moody do you know if this can officially be closed now, as @kelson42 suggests, with PR #691 now merged?

Plz clarify your the readiness of your PR #687 too if poss!

tim-moody, 2018-02-26:
I have not tested this. #687 is independent of whether it works and only becomes unnecessary if kiwix-serve never crashes again. It is ready to merge.

holta, 2018-03-07:
Appears fixed by #691 kiwix-tools 2017-12-27 -> 2018-02-26 (when testing against large Wikipedias...)

closed 2018-03-07 by holta