iiab/iiab #698: iiab-cmdsrv not enabled on ubuntu-18 -- maybe other os's

georgejhunt, 2018-03-14:
(I do not know why this is surfacing now)
This is a bug in ansible -- line 251 fails to create a link in /etc/systemd/system/multi-user.target.wants:

line 251 roles/cmdsrv/tasks/main.yml:
- name: Enable iiab-cmdsrv service
  service: name=iiab-cmdsrv
           enabled=yes
           state=restarted

root@box:/etc/systemd/system/multi-user.target.wants# ls
apache2.service		  netfilter-persistent.service
atd.service		  NetworkManager.service
avahi-daemon.service	  ondemand.service
bind9.service		  open-vm-tools.service
chrony.service		  pollinate.service
console-setup.service	  pppd-dns.service
cron.service		  remote-fs.target
dhcpd.service		  rsync.service
ebtables.service	  rsyslog.service
iiab-startup.service	  snapd.autoimport.service
irqbalance.service	  snapd.core-fixup.service
isc-dhcp-server6.service  snapd.service
kalite-serve.service	  ssh.service
kiwix-serve.service	  systemd-resolved.service
lxcfs.service		  udev-reload.service
lxd-containers.service	  ufw.service
mariadb.service		  unattended-upgrades.service
ModemManager.service	  vnstat.service
munin-node.service	  wpa_supplicant.service

root@box:/etc/systemd/system/multi-user.target.wants# systemctl status iiab-cmdsrv
● iiab-cmdsrv.service - Provides the IIAB Command Server
   Loaded: loaded (/etc/systemd/system/iiab-cmdsrv.service; enabled-runtime; ven
   Active: inactive (dead)

I broke line 251 ff into two separate plays -- one for enable and one for start -- still failed

Then I  called systemctl directly, and the link was generated:
- name: Enable iiab-cmdsrv service
  shell: systemctl enable iiab-cmdsrv

After re-running /opt/iiab/iiab-admin-console/install (with the above direct call to systemctl)
root@box:/opt/iiab/iiab-admin-console# ls /etc/systemd/system/multi-user.target.wants/
apache2.service		  netfilter-persistent.service
atd.service		  NetworkManager.service
avahi-daemon.service	  ondemand.service
bind9.service		  open-vm-tools.service
chrony.service		  pollinate.service
console-setup.service	  pppd-dns.service
cron.service		  remote-fs.target
dhcpd.service		  rsync.service
ebtables.service	  rsyslog.service
iiab-cmdsrv.service	  snapd.autoimport.service
iiab-startup.service	  snapd.core-fixup.service
irqbalance.service	  snapd.service
isc-dhcp-server6.service  ssh.service
kalite-serve.service	  systemd-resolved.service
kiwix-serve.service	  udev-reload.service
lxcfs.service		  ufw.service
lxd-containers.service	  unattended-upgrades.service
mariadb.service		  vnstat.service
ModemManager.service	  wpa_supplicant.service
munin-node.service
root@box:/opt/iiab/iiab-admin-console#

holta, 2018-03-14:
Thx @georgejhunt !

Can you or @tim-moody or anybody confirm this bug in https://releases.ansible.com/ansible/ansible-2.5.0rc2.tar.gz ?

(Whoever knows the best way to run Ansible 2.5.x on Ubuntu 18.04 please let us know, I'd like to post it to https://github.com/ansible/ansible/milestone/11 if poss!)

holta, 2018-03-15:
(We presume @georgejhunt is running Ansible 2.4.2 as most all of us are running that at the moment)

tim-moody, 2018-03-15:
I have not tried ubuntu 18 and have not seen this on rpi

holta, 2018-03-16:
> Whoever knows the best way to run Ansible 2.5.x on Ubuntu 18.04 please let us know

Ansible 2.5.0 RC3 is the latest as mentioned in #669.

holta, 2018-03-16:
> Whoever knows the best way to run Ansible 2.5.x on Ubuntu 18.04 please let us know

"pip install ansible==2.5.0rc3" looks sufficient according to https://libraries.io/pypi/ansible ?

See also https://pypi.python.org/pypi/ansible

holta, 2018-04-26:
> @georgejhunt & all who can get access to Ubuntu 18.04: let's try to confirm this is resolved on the newly released Ansible 2.5.1, prior to 18.04's release in 6 days if possible (hopefully our own release of IIAB 6.5 will be right around then too!)

@georgejhunt & all who are able to test Ubuntu 18.04 LTS upon its release in coming hours/days:

[IIAB 6.5 install scripts](http://download.iiab.io/6.5/rpi/) now mandate ansible-2.4.x so is this perhaps no longer an issue?  (When installing a fresh IIAB 6.5/master on Ubuntu 18.04...)

ASIDE: IIAB 6.6 (presumably equivalent to iiab/iiab's master branch starting next week) may be an entirely different story, as [its own install scripts](http://download.iiab.io/6.6/rpi/) install the latest Ansible (currently Ansible 2.5.1).

holta, 2018-04-30:
Appears solved — but don't hesitate to re-open as necessary.

closed 2018-04-30 by holta