iiab/iiab #709: Consider the latest kiwix-tools prior to IIAB 6.5 final release

holta, 2018-03-22:
If so, upgrade from 2018-02-26 to the latest from http://download.kiwix.org/nightly/ in both these places:
- [roles/kiwix/defaults/main.yml](https://github.com/iiab/iiab/blob/master/roles/kiwix/defaults/main.yml)
- http://download.iiab.io/packages/

holta, 2018-03-24:
New versions of kiwix-tools 0.3.0 appear here since 2018-03-21, thanks to @mgautierfr:
http://download.kiwix.org/release/kiwix-tools/

Earlier releases:
http://download.kiwix.org/bin/

Builds from the most recent 30 days:
http://download.kiwix.org/nightly/

holta, 2018-03-26:
Likely IIAB 6.5 will endorse Ansible 2.4.2 and/or Ansible 2.4.4

Whereas IIAB 6.6 will build off Ansible 2.5.0 and successors.

Ref's
#711 "restore scripts/ansible now that Ansible 2.5.0 is released & smoke-tested"
#712 "support Ansible 2.5.x: include_role to import_role in Stages 3-9"
#714 "Enhancement - Generate Dynamic Playbook"

closed 2018-03-27 by holta

tim-moody, 2018-03-27:
why do we not do an upgrade? (force install is only true if not already installed)

commit dd9ec834 (2018-03-27): Merge pull request #717 from holta/kiwix-tools_0.4.0

kiwix-tools 2018-02-26 -> 0.4.0 for armhf & linux64

closes #709