iiab/iiab #710: "Failed to start usbmount@dev-sda.service." during every IIAB boot on RPi

holta, 2018-03-23:
Error occurs on all boots, after installing IIAB 6.5/master on Raspbian Lite 2018-03-13 on a (regular!) RPi 3, using 1-line install script http://download.iiab.io/6.5/rpi/load.txt

FYI "systemctl start usbmount@dev-sda.service" also fails (after boot).  Inspired by /var/log/boot.log which recommended "systemctl status usbmount@dev-sda.service"

This seems to be a very new IIAB bug arising in recent days (or weeks?)

Somewhat Related:
#682 "automount USB exFAT sticks (& possibly portable NTFS disks too)"
#695 "Option to show entire USB stick/drive content (at http://box/usb)"

holta, 2018-03-26:
Notes:
- does not prevent actual usage http://box/usb (content IS still visible)
- warning goes away if rebooting when there's no stick inserted

georgejhunt, 2018-03-29:
Note:
systemctl disable usbmount\@.service makes the error message dissappear

holta, 2018-03-29:
> `systemctl disable usbmount@dev-sda.service` makes the error message disappear

Is [roles/usb-lib/tasks/main.yml](https://github.com/iiab/iiab/blob/master/roles/usb-lib/tasks/main.yml) the place to make this happen?

holta, 2018-03-29:
@jvonau experiment shows another method to make red error go away...adding `After=rc-local.service` to the end of [Unit] section in /etc/systemd/system/usbmount@.service so that it appears as follows:
```
[Unit]
BindTo=%i.device
After=%i.device
After=rc-local.service

...
```
Where should this be implemented / in which playbook?

holta, 2018-03-29:
(Resolved by @jvonau's PR #722 "delay usbmount's actions during boot to be as late as possible")

closed 2018-03-29 by holta

tkantor81, 2019-05-08:
Error is still visible.

holta, 2019-12-12:
TK Kang mentions this is an issue when installing IIAB 7.1 pre-releases on on Debian 10?

commit 59f1aba9 (2018-03-29): Merge pull request #722 from jvonau/usb-boot

delay usbmount's actions during boot to be as late as possible

closes #710