iiab/iiab #714: Enhancement - Generate Dynamic Playbook

tim-moody, 2018-03-26:
Give a list of desired tags to run as currently done by runtags, generate a dynamic top level playbook to include the vars section, aggregate 0 and the role of each passed tag, assuming that tag name and role name are the same.

holta, 2018-03-27:
Given that Ansible 2.5 has broken ./runtags (neither #712 nor #716 fully resolve this just yet) @tim-moody is also exploring a possibly much simplified replacement command (`./runrole`, that works quite separately from Ansible's tag system).  Perhaps with /opt/iiab/iiab/iiab-run-one-role.yml along these lines:
```
---
- hosts: all
  become: yes

  vars_files:
  - roles/0-init/defaults/main.yml
  - vars/default_vars.yml
  - vars/{{ ansible_local.local_facts.os_ver }}.yml
  - vars/local_vars.yml

  tasks:

  - name: 0-init
    include_role:
      name: 0-init
    tags: 0-init

  - name: Runtag
    include_role:
      name: {{ run_role }}

ansible-playbook -i ansible_hosts iiab-tmp-run.yml --connection=local -e "run_role=$1"
```
If running multiple roles is important, perhaps something like...
```
ansible-playbook -i ansible_hosts run_roles.yml  --connection=local -e "role_list=kiwix,kalite"
```
List's syntax needs fixing, perhaps...
```
role_list=['kiwix','kalite']
```
If multiple roles are even necessary.  Regardless, our 11 aggregate roles (Stage 0-9 and network) can be invoked, using `./runrole 7-edu-apps` etc.

tim-moody, 2018-03-27:
the multi-role syntax does not work

holta, 2018-03-29:
@tim-moody reports that his draft `./runrole` currently has a problem getting access to critical vars in [roles/0-init/defaults/main.yml](https://github.com/iiab/iiab/blob/master/roles/0-init/defaults/main.yml)

Reason: [default_vars.yml](https://github.com/iiab/iiab/blob/master/vars/default_vars.yml) are not shared (as if private: True) ?

So apparently `./runrole kiwix` works but `./runrole kalite` does not.

One workaround might be putting all the variables from roles/0-init/defaults/main.yml into default_vars.yml ?

holta, 2018-03-29:
Let's confirm that the following...
```
  vars_files:
  - roles/0-init/defaults/main.yml
```
...at the very top of ./runrole (similar to https://github.com/iiab/iiab/blob/master/iiab-stages.yml#L6) is insufficient to get these var values?

(Perhaps this is because of dynamic `include_*` failing to inherit attributes in Ansible 2.5?  As opposed to the static `import_*`)

jvonau, 2018-03-30:
[upstream bug](https://github.com/ansible/ansible/pull/36372)

holta, 2018-04-19:
Upstream bug https://github.com/ansible/ansible/pull/36372 (looping in include_role:) has been fixed in the newly released Ansible 2.5.1

holta, 2018-05-18:
> Upstream bug ansible/ansible#36372 (looping in include_role:) has been fixed in the newly released Ansible 2.5.1

Better news with [Ansible 2.5.3](https://github.com/ansible/ansible/blob/stable-2.5/changelogs/CHANGELOG-v2.5.rst) now released:

> loop_control - update template vars for loop_control fields on each loop iteration (https://github.com/ansible/ansible/pull/39818).

> Fix name parameter templating in include_role module
https://github.com/ansible/ansible/pull/36372

tim-moody, 2018-06-12:
I already closed this once. Now again.

closed 2018-06-12 by tim-moody