iiab/iiab #715: Pin IIAB 6.5 to Ansible 2.4.x in scripts/ansible ?

holta, 2018-03-26:
@jvonau do you know how to do this in [scripts/ansible](https://github.com/iiab/iiab/blob/master/scripts/ansible) ?

PS pip within scripts/ansible-2.2.0 -> ansible-2.4.2 -> ansible-2.4.4 (soon) is one option.  But there's prob a better way, in preparation for 2.4.5 backported bug fixes etc.

holta, 2018-03-26:
Related:
#709 "Consider the latest kiwix-tools prior to IIAB 6.5 final release"
#669 "is Ansible 2.4.3 running out of memory when installing IIAB 6.5/master on RPi3?"

jvonau, 2018-03-29:
Looks like upstream ansible as reworked the repo urls again from
http://ppa.launchpad.net/ansible/ansible/ubuntu
to:
http://ppa.launchpad.net/ansible/ansible-2.5/ubuntu 
http://ppa.launchpad.net/ansible/ansible-2.4/ubuntu 

See and expand the 'Technical details about this PPA' link
https://launchpad.net/~ansible/+archive/ubuntu/ansible-2.5
https://launchpad.net/~ansible/+archive/ubuntu/ansible-2.4

Modifying the respective lines in scripts/ansible should pin you to 2.4 or 2.5 

Preventing ansible from updating from 2.4.2 that is being installed to 2.4.3 until 2.4.5 is released is the issue. I'd ensure that /etc/apt/sources.list does not contain ansible defination and /etc/apt/sources.list.d/ansible*.list does not exist to prevent apt from seeing the ansible repo and thus any available update available from upstream until 2.4.5 is released. 

Personally I feel the out of memory issue uncovered in #669 can be mitigated with a more adequate amount of swap space and 2.5 has show the same behaviour now.

holta, 2018-03-29:
Excellent summary above by @jvonau.

If Ansible 2.4.4 is released shortly, this might be best for IIAB 6.5 legacy systems, given the serious [maximum recursion depth](https://github.com/iiab/iiab/pull/716#issuecomment-376406165) issues recently uncovered in 2.4.2

Conversely rapid consolidation around 2.5.x (which already today solves essentially all our known issues) might serve us all strengthening community focus (certainly for IIAB 6.6) _...as soon as we transition from ./runtags to ./runrole or similar._

holta, 2018-04-05:
FWIW Ansible 2.4.4 (v2.4.4.0-1) is now released:
- https://releases.ansible.com/ansible/
- https://launchpad.net/~ansible/+archive/ubuntu/ansible-2.4
Its 70+ bug fixes are listed here:
- https://github.com/ansible/ansible/blob/stable-2.4/CHANGELOG.md

holta, 2018-04-05:
FYI ./iiab-install and ./runtags and ./runrole all work well with the new Ansible 2.4.4

(Tests were completed on Ubuntu, similar tests are ongoing on Raspbian Lite...)

holta, 2018-04-06:
Done.

See load scripts in http://download.iiab.io/6.5/rpi/ which now use scripts/ansible-2.4.x from PR #711

closed 2018-04-06 by holta