iiab/iiab #718: hostapd config

tim-moody, 2018-03-28:
why are there two config files that are nearly identical?
hostapd.conf and iiab.hostapd.conf

holta, 2018-03-28:
Problems getting the internal WiFi AP (hostapd) enabled on new IIAB installs?

@floydianslips & @kanagit also — had such problems in the past month.

Until this solved properly, here are workarounds documented in #655:
- cd /opt/iiab/iiab; ./iiab-network
- systemctl enable hostapd

jvonau, 2018-03-29:
@tim-moody   Given that hostapd.conf is an ansible template that is populated only once during iiab-install or iiab-network is run once some other method is used to edit hostapd.conf knowing what the starting point was post ansible becomes more difficult. Just an easy method to have backup copy around that is pre-populated with the values that were declared at ansible runtime.
@holta  Post or sent some logs and I will have a look. Phoning the doctor and saying it hurts here does not provide enough data to solve the problem. I suspect that the install is reporting appliance mode due to some change that was introduced after I wrote routine for the 'keep wifi available as upstream' that you requested.

holta, 2018-03-29:
Clarif of @tim-moody's observation &mdash; these 2 files are different:

https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/hostapd.conf.j2
https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/iiab-hostapd.conf.j2

holta, 2018-03-29:
On the call we discussed renaming iiab-hostapd.conf.j2 to iiab-hostapd-discovered.conf.j2 but in the end decided against this (2 files to handle separate logic) as either/appropriate one is copied into a single file.

closed 2018-03-29 by holta