iiab/iiab #723: large SD's don't boot in RPi Zero W when image small?

holta, 2018-03-29:
Due to "fsck.repair=yes rootwait" in /boot/cmdline.txt ?

@tim-moody theory: writing small images to large (e.g. 128GB) microSD causes checkdisk/fsck to time out on RPi Zero W, failing to boot when it gives up after several minutes.

@georgejhunt wonders if we can increase the timeout?

tim-moody, 2018-03-29:
This is all anecdotal. A large (say 128G) card on rpi W can fail to boot.

This can happen if there is a large amount of space to resize and also if fsck.repair needs to run.

jvonau, 2018-03-29:
Looking at a [boot sequence from a pri](https://raspberrypi.stackexchange.com/questions/78099/how-can-i-lower-my-boot-time-more) it appears that the resizing service will run much later than systemd-fsck-root.service that does the root file system check. Just a thought, perhaps iiab-rpi-root-resize.service could employ a Before=systemd-fsck-root.service line to have the resizing take place before the file system integrity check is preformed?

tim-moody, 2018-03-29:
worth a try, but I think the biggest problem is how long even the fsck service can take if the card is large.

holta, 2018-03-30:
@jvonau writes: "Tim Looking at the long fsck check failure on boot, perhaps sd-min should do the tune2fs before running e2fsck to prevent having the image contain what might be considered a dirty filesystem?"

holta, 2018-06-14:
@tim-moody is this failure (caused by a slow fsck / disk check) at all related to yours?

[image]

holta, 2018-06-20:
RPi Zero W's very often prevent themselves from booting, due to the problem shown in the above screenshot &mdash; no doubt triggered by these slow/cautious disk checks that happen on every 3rd or 4th boot &mdash; causing very serious problems when you have no screen or keyboard (as is normal with Zero W's, hence this is very serious!)

Pasted in explicitly here:
```
[ TIME ] Timed out waiting for device dev-disk-by\x2partuuid-36d137ba\x2d01.device.
[DEPEND] Dependency failed for File System Check on /dev/disk/by-partuuid/36d137ba-01.
[DEPEND] Dependency failed for /boot.
[DEPEND] Dependency failed for Local File Systems.
[DEPEND] Dependency failed for Mark the need to relabel after reboot.
[ TIME ] Times out waiting for device dev-serial1.device.
```
CLARIF: this happens even if IIAB contains very few services at all (e.g. Calibre, Kiwix, MediaWiki and nothing else!)

jvonau, 2022-04-05:
Think this was solved in #2465

holta, 2022-04-05:
> Think this was solved in #2465

Let's hope so.

(@tim-moody please speak up if there's more to this?)

closed 2022-04-05 by holta