iiab/iiab #743: Instructions  to create sdcard on mac

tim-moody, 2018-04-11:
A potential user wanted to write an iiab image to an sd card using a mac, but the instructions were not detailed enough to do it.

holta, 2018-04-26:
@tim-moody am moving this to [IIAB 6.6's Milestone](https://github.com/iiab/iiab/milestone/3) if that's ok?

arky, 2018-07-02:
@tim-moody  @holta 
I had similar request from Philippines. So, this is my solution I found.  
[Etcher for Windows, Linux and MacOS](https://etcher.io/)
I have also added a note in IIAB Installation wiki page.

holta, 2018-07-02:
> I had similar request from Philippines. So, this is my solution I found.
Etcher for Windows, Linux and MacOS

Thanks a million @arky &mdash; while several of us use Win32 Disk Imager (https://sourceforge.net/projects/win32diskimager/) I've noticed my roommates much prefer http://etcher.io which is very clean, and has more options (and more OS support!)

> I have also added a note in IIAB Installation wiki page.

Huge Thanks for Helping Here: that legacy section https://github.com/iiab/iiab/wiki/IIAB-Installation#take-a-short-cut definitely needs cleaning up after 3 years (mainline installation approach has changed over the past half-decade!) so let's keep on it (:

holta, 2018-07-18:
Any progress to report?

Hope it Ok am moving this to moving this to the [IIAB 6.7 Milestone](https://github.com/iiab/iiab/milestone/4) as our [IIAB 6.6](http://download.iiab.io/6.6/) final release is now quickly approaching?

jvonau, 2020-04-29:
close

holta, 2020-04-29:
Any progress to report?  Can we close this now?

holta, 2020-04-29:
Let's assume [Raspberry Pi Imager](https://www.raspberrypi.org/blog/raspberry-pi-imager-imaging-utility/) (which works on macOS) will solve this sufficiently going forward.

closed 2020-04-29 by holta