iiab/iiab #767: OpenVPN connection false positives

holta, 2018-04-30:
Thanks to @jvonau who confirms OpenVPN traffic is not flowing in either direction on affected machines. Such machines claim ("ip a") they are connected to OpenVPN with a VPN IP address for many days.  But this is a false positive as traffic does not flow.  Coincidentally both machine I've seen are running Ubuntu 16.04 LTS &mdash; and have been rebooted multiple times).  Details from @jvonau pasted in here:

> /var/log/openvpn.log
Apr 29 03:28:31 2018 OpenVPN 2.3.10 x86_64-pc-linux-gnu [SSL (OpenSSL)] [LZO] [EPOLL] [PKCS11] [MH] [IPv6] built on Jun 22 2017
Sun Apr 29 03:28:31 2018 library versions: OpenSSL 1.0.2g  1 Mar 2016, LZO 2.08
Authenticate/Decrypt packet error: packet HMAC authentication failed
/var/log/apt/history.log
Start-Date: 2018-04-20  06:28:30
Commandline: /usr/bin/unattended-upgrade
Upgrade: openssl:amd64 (1.0.2g-1ubuntu4.11, 1.0.2g-1ubuntu4.12), libssl-dev:amd64 (1.0.2g-1ubuntu4.11, 1.0.2g-1ubuntu4.12), libssl-doc:amd64 (1.0.2g-1ubuntu4.11, 1.0.2g-1ubuntu4.12), libssl1.0.0:amd64 (1.0.2g-1ubuntu4.11, 1.0.2g-1ubuntu4.12)
End-Date: 2018-04-20  06:28:47
>
> Think upgrade changed something in the libraries and some option is no longer available or openvpn needs to be rebuild against the newer libraries

jvonau, 2018-04-30:
Ubuntu 18.04 ships
>root@box:/home/iiab-admin# apt list openssl*
Listing... Done
openssl/bionic,now 1.1.0g-2ubuntu4 amd64 [installed]
openssl1.0/bionic 1.0.2n-1ubuntu5 amd64

Raspbian 9 ships
>root@raspberrypi:/opt/pi-gen# apt list openssl*
Listing... Done
openssl/stable,now 1.1.0f-3+deb9u2 armhf [installed,automatic]
openssl-blacklist/stable 0.5-3 all
openssl-blacklist-extra/stable 0.5-3 all

the above are using 1.1.0 while the problem 16.04 machines use 1.0.2 for the ssl libraries, wonder if downgrading openss to1.0.2g-1ubuntu4.11 from the newest 1.0.2g-1ubuntu4.12 makes the issue go away... If that fixes the problem then that would point to the need for openvpn to be updated by upstream Ubuntu.

holta, 2018-05-01:
> Even more serious, ping box.lan (and its equivalent ping 172.18.96.1) are both broken, experiencing 100% packet loss on these Ubuntu 16.04 + IIAB 6.5 machines in question.

FYI @jvonau is making good progress solving the above (NetworkManager issue @ PR #769).

So far this appears unrelated to the OpenVPN failure (#767) on the very same Ubuntu 16.04 machines :/

holta, 2018-05-01:
PR #769 tested across 3 OS's & merged &mdash; so "ping box.lan" and "ping 172.18.96.1" now work on all known IIAB 6.5 machines.

(Any other ideas that could explain these OpenVPN failures on Ubuntu 16.04?)

holta, 2018-05-03:
Is https://askubuntu.com/questions/1027870/openvpn-authenticate-decrypt-packet-error-packet-hmac-authentication-failed suffering the same failure as we are, on 2018-04-24 ?

_If so let's follow up on his/her ticket, raising the prominence of this serious issue, until upstream maintainers can hopefully offer us an explanation-_

holta, 2018-05-03:
Thanks @floydianslips who's currently trying @jvonau's test of an older openssl on an Ubuntu 16.04 VM, so see if that (hopefully!) brings OpenVPN back to life...

holta, 2018-05-04:
Ubuntu Forums asked me to repost here:
https://askubuntu.com/questions/1032047/openvpn-suddenly-failing-on-ubuntu-16-04-since-openssl-update

holta, 2018-05-11:
ssh over OpenVPN to an Ubuntu 16.04 machine running IIAB 6.5/master now suddenly works.

I have absolutely no idea why :-(

holta, 2018-05-11:
Resolved itself, for reasons I do not understand:

/var/log/apt/history.log claims /usr/bin/unattended-upgrade ran twice in the 2 prior days, which might mean Ubuntu 16.04 fixed itself with online updates??  And yet openssl still shows:
```
root@box:~# apt -a list openssl
Listing... Done
openssl/xenial-updates,xenial-security,now 1.0.2g-1ubuntu4.12 amd64 [installed]
openssl/xenial 1.0.2g-1ubuntu4 amd64
```
Am moving this ticket to IIAB 6.6 so we can keep an eye on it if this or similar flares up again!