iiab/iiab #775: iiab-cmdsrv doesn't auto-start on Ubuntu 18.04

holta, 2018-05-03:
Manually starting `systemctl status iiab-cmdsrv` works AFTER boot &mdash; but iiab-cmdsrv isn't starting DURING boot of Ubuntu 18.04/Server and 18.04/Desktop (each of which contain the latest IIAB 6.5/master).

@jvonau tried commenting out Line 3 here: (derived from https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/templates/iiab-cmdsrv.service.j2)
```
[Unit]
Description=Provides the IIAB Command Server
#After=syslog.target network.target local-fs.target
After=multi-user.target
Requires=multi-user.target

[Service]
Type=forking
Environment="CMDSRV_DIR=/opt/admin/cmdsrv"
PIDFile=/run/iiab-cmdsrv.pid
ExecStart=/etc/init.d/iiab-cmdsrv-init  start
ExecStop=/usr/bin/iiab-cmdsrv-ctl STOP

[Install]
WantedBy=multi-user.target
```
Unfortunately iiab-cmdsrv still does not start at boot time on Ubuntu 18.04.

@jvonau wonders if something (like) `DefaultTimeoutStartSec=90s` might possibly help, at the end of the [unit] section above ??

holta, 2018-05-03:
@tim-moody shout if you have ideas here...this appears to be the main remaining blocker to releasing IIAB 6.5 :-)

PS both Ubuntu 18.04 machines can be accessed remotely if that helps.

holta, 2018-05-04:
@tim-moody @georgejhunt 

If systemd is blocking auto-start of iiab-cmdsrv (for whatever reason, as @jvonau has found...)

Should we "systemctl enable rc-local" then put "#!/bin/bash" and "systemctl start iiab-cmdsrv" into /etc/rc.local &mdash; as an interim solution so we can release IIAB 6.5?

https://askubuntu.com/questions/9853/how-can-i-make-rc-local-run-on-startup

tim-moody, 2018-05-04:
This doesn't add up:

"systemctl disable iiab-cmdsrv-init"
how did it get enabled. it is not a service.

I don't see how the path changes make the reboot work as they are all absolute.

iiab-cmdsrv-init (sourced from iiab-admin-console/roles/cmdsrv/templates/iiab-cmdsrv-init.j2#L68, see Line 68) contains bug "xscd-cmdsrv-ctl" which should be "iiab-cmdsrv-ctl"

True and should be fixed or better removed, but iiab-cmdsrv-init STOP is never invoked, so irrelevant to the reported problem.

REMOVE all executable bits from /etc/systemd/system/iiab-cmdsrv.service
True, this is recommended, but how is it relevant to the reported problem unless treated differently in ub 18 than other oses.

So, I am not  clear what it is that fixed the problem.

holta, 2018-05-04:
> "systemctl disable iiab-cmdsrv-init"
how did it get enabled. it is not a service.

This seems to be a side effect of it being stored in an illegal directory.

Certainly Jerry's fix works, having been tested on both 18.04/Desktop and 18.04/Server (which are quite different from one another!)

In any case init.d is on the road to obsolescence, and we don't have much choice here :)

jvonau, 2018-05-04:
>"systemctl disable iiab-cmdsrv-init"
how did it get enabled. it is not a service.

Think this is a by-product systemd's [systemd-sysv-generator](https://www.freedesktop.org/software/systemd/man/systemd-sysv-generator.html) which might be handling things differently between 16.04 and 18.04.

jvonau, 2018-05-04:
Raspbain 
> apt list systemd
Listing... Done
systemd/stable,now 232-25+deb9u2 armhf [installed]

Ubuntu 16.04 
> apt list -a systemd
Listing... Done
systemd/xenial-updates,now 229-4ubuntu21.2 amd64 [installed]

Ubuntu 18.04
> apt list systemd
Listing... Done
systemd/bionic,now 237-3ubuntu10 amd64 [installed]

holta, 2018-05-04:
https://github.com/systemd/systemd/blob/master/NEWS might be relevant.

(org.freedesktop.systemd1.Manager.AttachProcessesToUnit ?)

holta, 2018-05-05:
Consider this fix...that @jvonau has tested on Raspbian:
PR https://github.com/iiab/iiab-admin-console/pull/59

holta, 2018-05-06:
@tim-moody please consider merging PR https://github.com/iiab/iiab-admin-console/pull/60 so all can move forward with IIAB 6.5 RC2, Thx!

Aside: PR https://github.com/iiab/iiab-admin-console/pull/61 split off as you requested &mdash; please revise that as appropriate.

closed 2018-05-07 by tim-moody