iiab/iiab #776: WordPress Menu text/logo needed in Kenya

holta, 2018-05-03:
@kananigit can you refine the best WordPress menu text/icon below, as is required in Kenya, so that @tim-moody can upstream that into the official https://github.com/iiab/iiab-menu ?

Whereas https://github.com/iiab/iiab-menu/blob/master/menu-files/menu-defs/en-wordpress.json is in fact available today ("Write a personal article (or journal blog post) to share with your community!") these are other options:
http://iiab.me/iiab-menu/menu-files/menu-defs/en-wordpress2.json
http://iiab.me/iiab-menu/menu-files/menu-defs/es-wordpress.json

_Live @ the bottom of iiab.me/chiapas for Mexico._

Here are 2 older WordPress logos I'm not in love with:
https://github.com/iiab/iiab-menu/blob/master/menu-files/images/wordpress-logo.png
https://github.com/iiab/iiab-menu/blob/master/menu-files/images/wordpress.png

Personally I prefer this one: (feel free to reduce it / thumbnail it, if 77KB is too large and you want custom text for Kenya that you want Tim to merge?)
http://iiab.me/iiab-menu/menu-files/images/wordpress-icon.png

ggcarlosr, 2018-05-04:
Hello, I am a designer that you need, I am willing to help, what would your requirements be?

kananigit, 2018-05-04:
@holta 
I like the menu item you selected too. 
http://iiab.me/iiab-menu/menu-files/images/wordpress-icon.png

Am not the best designer out there but @ggcarlosr  here can help us. Am thinking something along this lines

[image]

Can have a "w" for wordpress too. Any ideas ??????

ggcarlosr, 2018-05-04:
ok, what do you want me to do to that circle, I add the "W" as the Wordpress?

ggcarlosr, 2018-05-04:
I see that you have a very powerful repository Git, as a designer I can collaborate not only in what you currently need, I can also create a logo change for your page and a header design that is coupled with the new logo, all without any compromise. If you need my help let me know, I am very interested in this repo

holta, 2018-05-04:
That'd be amazing Carlos, as our most urgent need (in truth is) an Internet-in-a-Box logo for the entire project :-)

(Definitely please do contact me be email holtATlaptop.org if you also have time there !)

ggcarlosr, 2018-05-04:
can we communicate through discord app, do you have a user there? my username is @ggcarlosr#3234

ggcarlosr, 2018-05-04:
my email is elwu3y@gmail.com

ggcarlosr, 2018-05-04:
Ok the circle you want it of that same color that you indicate? or I can provide a color more adapted to the web.

kananigit, 2018-05-04:
Doesn't have to be exactly same. You can provide one that is more adopted
to the web

ggcarlosr, 2018-05-04:
ok I will develop it and send it here. what resolution in specific requires

kananigit, 2018-05-04:
Thanks. About the resolution, well, not exactly sure the ones Tim uses for
our menus, but if you look at the logos Adam posted above you can make one
with similar resolution.

holta, 2018-05-04:
Exactly: http://iiab.me/iiab-menu/menu-files/images/wordpress-icon.png I love but it (and others!) need to be shrunk down to about 100x100 pixels.

So these logos all load quickly in developing world situations on olders phones etc..._while remaining attractive :)_

ggcarlosr, 2018-05-04:
I sent them 2 logo, the one that they sent me adapted to a resolution of 100x100 and one with blue flat color, color that predominates in its web. They notify me of any concerns. 
[DOWNLOAD HERE](https://drive.google.com/open?id=1BZn8j3KRBMTHTVOX2PJbIERLzb9tW_PS)

ggcarlosr, 2018-05-04:
I would like to know what your requirements are for the logo of your repo, if you have something in mind or are open to any work that presents them.

holta, 2018-05-04:
> I sent them 2 logo, the one that they sent me adapted to a resolution of 100x100 and one with blue flat color, color that predominates in its web.

Cdn you upload those to this GitHub ticket? (As attachments if necessary!)

ggcarlosr, 2018-05-04:
I'm going to try it I'm not very good at this alone in design...

holta, 2018-05-17:
@kananigit please re-open this as nec!

closed 2018-05-17 by holta