iiab/iiab #789: Logo Proporsal

ggcarlosr, 2018-05-12:
that such a brother here opened this issue to show some ideas that have been reflected so far,

[image]

holta, 2018-05-12:
Thanks Carlos!  @darkenvy likes the 2nd from the right — where the hexagon and the mouse pointer are both hollow/white:
- He suggests removing both the inner gray dot and semi-circle?
- He suggests "Internet in a Box" be aligned as in your top proposal.  So the 'x' is directly below the 't'.
Perhaps the rightmost logo has potential too, with dot and semi-circle likewise removed?
- if so, should the smaller mouse pointer (that we like) perhaps become all white?

ggcarlosr, 2018-05-13:
I'm glad you answered, I thought you forgot about the makeover. Here I send you the corrections requested, I hope I could have done everything as indicated-

[image]

holta, 2018-05-13:
Thanks Carlos!

@darkenvy was wondering how it would look with the _outer circle preserved_.  (That would mean the circle that goes 3/4 of the way around, i.e. bit more than 270 degrees).

ggcarlosr, 2018-05-13:
Ok here is updated, with the semi circle.

[image]

holta, 2018-05-13:
Clarification: earlier logo designs were lost, due to Anish's server failure in India: #383

ggcarlosr, 2018-05-13:
I wait for your point of view of the last correction sent.

holta, 2018-05-13:
@ggcarlosr I think Reno (@darkenvy) was interested in the larger circle/arc, that goes 75% or 80% around the mouse pointer (and with a larger diameter!)

But hopefully he will respond here, in the coming 24 hours :-)

ggcarlosr, 2018-05-13:
ok I'm going to close the circle around the pointer as indicated

tim-moody, 2018-05-13:
I still like a literal internet in a box, with the words, using something like 
[image] for internet

holta, 2018-05-18:
@darkenvy @floydianslips @kananigit & all please add logo ideas/suggestions for Carlos!

holta, 2018-05-19:
> I like the second logo in @ggcarlosr 's last post ^.

@darkenvy & All: can you think of a way to de-emphasize the central arc/circle (conveying "direct action" of clicking a mouse-or-such is a good thing!) but trying to make much more soft/natural?

(I not sure eliminating this arc/circle is the right move &mdash; but it seems to command too much visual attention as is??)

ggcarlosr, 2018-05-20:
Seeing your comments I think you all agree with the second option, do you think that would be the definitive logo for IIAB?

tim-moody, 2018-05-20:
So to summarize:
I like 'reinforcing representations'. Otherwise why have a graphic, or perhaps replace it with a cat, which I think kids like, or some other animal not yet taken, like a Kuala or an Ibex! (get it).
It might make sense just to have the name. The S Asia press release just had that.
The overall shape should be roughly square as rectangles cause problems with partners.
BTW, for mobile I am coming to the conclusion that we shouldn't even have a logo in the menu
https://en.m.wikipedia.org/wiki/Main_Page
But for branding we definitely need one as soon as we have a web site.
Is the hexagon over used?
https://www.google.ca/search?q=images+hexagon+logo&tbm=isch&tbs=rimg:CWFnuN2DvzTSIjggTe7uDtbXG_1YT8zmrwjuaIJvz8QqYxHnPUCfqkG35Rc3Eusy02IIMTRuk2jxlrfLM9eStkzdFEioSCSBN7u4O1tcbEcQFMpRj4e3ZKhIJ9hPzOavCO5oRrBkVCFsJTKAqEgkgm_1PxCpjEeREDsb7IoaWcAyoSCc9QJ-qQbflFEYGiDvL3a_13lKhIJzcS6zLTYggwRZOZZyfZA-DkqEglNG6TaPGWt8hHTOtBtnsIdEyoSCcz15K2TN0USEW78KelKwpld&tbo=u&sa=X&ved=2ahUKEwifmoqA_5TbAhUp5IMKHW9mC-8Q9C96BAgBEBs&biw=1920&bih=1020&dpr=1

ggcarlosr, 2018-05-21:
What changes do you suggest with respect to the hexagon? That in itself is a representation of a box, although we see it

closed 2019-10-18 by holta