iiab/iiab #796: Rapid networking & other IIAB config changes that can't wait ~20min for Ansible

holta, 2018-05-14:
Am trying with @tim-moody to prioritize crucial features where implementers don't have time to wait ~20min for Ansible to finish.  Hopefully several of these improvements are possible for IIAB 6.6!?

Example things that (might!) be changeable in less than 1 minute -- order VERY preliminary below!  Are these good ideas?

1) SSID name change
2) Upstream SSID/password
3) VPN on/off + VPN handle
4) Fixed IP address to DHCP and back
5) Other network stuff...network sensing might be nec??
6) Turn services on/off

holta, 2018-05-14:
Quick thoughts from @jvonau:

>1 & 3 easy. 4 & 5 & 6 network services (squid dhcpd/dnsmasq) are in iiab-network. 2 is easy on raspbian but the others not so easy. 6 could just be calls to systemctl to enable/start and disable/stop

m-anish, 2018-05-15:
5 would be good to have

katneer91, 2018-05-15:
Like

tim-moody, 2018-05-15:
I think that state and environment are two different things. I would like to see a state file (json/yml/ini) that holds name-value pairs for all the main features of iiab, but is not as chatty as iiab.ini, which was really intended for reporting. That file could be loaded as facts by ansible and also used (read/write) by non-ansible scripts/applications that change state. It would include networking (but not working variables), installed and (actually) enabled, and other things that define the current state of iiab at any point in time.

holta, 2018-07-18:
Important...but not achievable for [IIAB 6.6](https://github.com/iiab/iiab/wiki/IIAB-6.6-Release-Notes) so moved to [IIAB 6.7](https://github.com/iiab/iiab/milestone/4)

holta, 2018-08-11:
If any such redesign of /etc/iiab/iiab.env and related files moves forward, it should take into account that 3 different people got stuck as a result of iiab.env &mdash; i.e. not know that such a centrally-important file even exists:

#880 "Install script fails to create /opt/iiab/downloads folder"

holta, 2019-04-27:
Related:

#1520 Admin Console & instructions should offer iiab-hotspot-on | iiab-hotspot-off functionality more intuitively

jvonau, 2020-04-29:
0-7 above have been achieved.

holta, 2020-04-29:
Great news indeed!

Public-spirited documentation should be added soon now, clarifying the most common public needs.

As part of #2096 "NGINX & ./runrole & ./iiab-network [Dual WiFi] & IIAB Maps need approachable docs for newbie/legacy implementers"

holta, 2020-05-08:
> 0-7 above have been achieved.

Great news, and certainly good enough for the imminent [IIAB 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes) !

@jerry where are the best places to monitor future networking possibilities, perhaps along the lines of Ubuntu's issue-tracking tickets you've linked to from #2270 ?

closed 2020-05-08 by holta