iiab/iiab #798: Sugarizer 0.9 -> 1.0

holta, 2018-05-15:
@georgejhunt would you have time to take a look at this small but important task?

Sugarizer should be upgraded from 0.9 to 1.0 and quickly tested on Raspbian and Ubuntu 18.04 LTS, as Sugarizer was released yesterday Monday 2018-05-14:

https://github.com/llaske/sugarizer/releases

holta, 2018-05-15:
Stretch Goal if volunteers can be found:

Testing of Sugarizer would be wonderful on Debian 9.4 also (and CentOS 7.5 if possible?)

holta, 2018-05-16:
Thanks to @georgejhunt who determined:

> sugarizer on rpi is not going smoothly. The image appears to have a file in the root of the tar.gz called "server". In version 0.9, server is a directory (rather than a file). In version -0.9, server is the root of the express framework. The redefinition causes our playbook to fail.

> On sugarizer v0.9, the <sugarizer root>/server/server.ini file appears to have the same content as <sugarizer root>/server on v1.0.
>
> we might be able to move, and rename, but I think it would be better to move the issue upstream...

@llaske do you have suggestions for evolving IIAB's [existing integration of Sugarizer 0.9](https://github.com/iiab/iiab/tree/master/roles/sugarizer), so that we might release Sugarizer 1.0 as part of IIAB 6.5 in coming days (for Mexican & Peru governments) rather than in the next release cycle?

llaske, 2018-05-16:
A major change between Sugarizer v0.9 and Sugarizer v1.0 is that the Server part is not included into Sugarizer Client. Another change is that Sugarizer Server no longer need Apache. All the server is now fully exposed from node.js.
The new repository for Sugarizer Server with instruction to install it here: https://github.com/llaske/sugarizer-server
It could be installed from Docker - included on the RaspberryPI. A current GSoC project plan to create a dedicated image on RaspberryPI.
@georgejhunt, do not hesitate to ask if I could help you to package Sugarizer v1.0 in IIAB 6.5.

georgejhunt, 2018-05-17:
I need a refresher. I think Jerry was involved in getting nodejs to work on
sugarizer. There was something about the re-install of nodejs causing a
failure of IIAB installs.

Also, I think Adam decided to eliminate mongodb, because of unpredictable
pre-allocation of too much disk space.

The new sugarizer-server has statistics, and usage summaries, which require
mongodb.  Are we willing to stick with no mongodb? We do not know what it
will take to get the new server code to ignore absence of mongodb.

There are separate code paths through the existing playbook, depending on
platform.  Are we willing to limit our 6.5 sugarizer offering to raspbian?
If not, the work expands.

I'm willing to work an hour or two per day on sugarizer changes, and
testing. But its complicated. I think it will delay release of 6.5. With
any luck, there may be other essential upgrades which we can throw into 6.5.

holta, 2018-05-17:
Excellent questions @geogejhunt:

1) I think you're 100% correct we should try for Sugarizer 1.0 on Raspbian for starters.

2) I don't know if MongoDB should be restored as part of IIAB.  As its random disk-preallocations (intensively documented in 2017, each install was different, often costing us 1GB with no explanation...) are not acceptable on Raspberry Pi installs, where SD storage is expensive.

3) @jvonau observes that Sugarizer 1.0 now defaults to Port 8080, colliding with IIAB's Calibre.

We need to change this in https://github.com/llaske/sugarizer-server/blob/master/env/sugarizer.ini immediately after cloning Sugarizer.

4) Please send specific nodejs / dependency questions to @jvonau who is quite an expert here.

tim-moody, 2018-05-17:
I vote against putting this is iiab 6.5.
I vote for reinstating mongodb to give sugarizer the benefit of the doubt.
In all instances we should use the latest version of node and mongo that are supported by both the os and sugarizer. We should get the port softcoded upstream.

georgejhunt, 2018-05-17:
> Hopefully this is solvable in 2018, but here's the background on MongoDB's
> prealloc files that were randomly created (or not!) in a nondeterministic
> way in 2017, during each IIAB installation:
>
> #254 <https://github.com/iiab/iiab/issues/254>

Did my sugarizer duty for today. Got it installed via a playbook (see
required changes at
https://github.com/iiab/iiab/compare/master...georgejhunt:sugar?expand=1).

The good news is that when I executed "npm install" a second time, on this
version or raspian, it did not appear to create an error (I did not "echo
$?", so npm return variable is still an open issue). The playbook
workaround that Jerry put in to keep from trying to install npm also
appeared to work.

Was able to get a sugar login screen. Created a user. But got an internal
error -500 when I clicked "Done". My guess is that the error is related to
not having a mongodb connection available. I will need to enable logging to
go the next little bit.

Done for today

holta, 2018-05-17:
> Got it installed via a playbook (see required changes at
https://github.com/iiab/iiab/compare/master...georgejhunt:sugar?expand=1).
>
> The good news is that when I executed "npm install" a second time, on this
version or raspian, it did not appear to create an error (I did not "echo
$?", so npm return variable is still an open issue). The playbook
workaround that Jerry put in to keep from trying to install npm also
appeared to work.
>
> Was able to get a sugar login screen. Created a user. But got an internal
error -500 when I clicked "Done". My guess is that the error is related to
not having a mongodb connection available. I will need to enable logging to
go the next little bit.

Amazing !

jvonau, 2018-05-18:
testing mongodb by un# the statza in roles/0-init/tasks/main.yml and then runrole mongo results in 
>ls -lah /var/lib/mongodb/
total 33M
drwxr-xr-x  4 mongodb mongodb 4.0K May 18 00:26 .
drwxr-xr-x 60 root    root    4.0K May 18 00:25 ..
drwxr-xr-x  2 mongodb nogroup 4.0K May 18 00:26 journal
-rw-------  1 mongodb nogroup  16M May 18 00:26 local.0
-rw-------  1 mongodb nogroup  16M May 18 00:26 local.ns
-rwxr-xr-x  1 mongodb nogroup    0 May 18 00:26 mongod.lock
drwxr-xr-x  2 mongodb nogroup 4.0K May 18 00:26 _tmp

looks better than what was reported in #254

holta, 2018-05-18:
> looks better than what was reported in #254

Great news.  What OS did you test?

jvonau, 2018-05-18:
On a RPi, but something is not right with the the our modified systemd and the stock(in /lib/systemd/system/) unit file. Mongo doesn't stay running but I can start the service from the cmdline with mongod --fork --port 27018 --dbpath /library/dbdata/mongodb --logpath /var/log/mongodb/mongod.log and that stays running but does not get rid of the 500 error noted above either, not sure what the issue is at this point. Done for the night.

llaske, 2018-05-18:
> The new sugarizer-server has statistics, and usage summaries, which require mongodb. Are we willing to stick with no mongodb? We do not know what it will take to get the new server code to ignore absence of mongodb.

@georgejhunt is right, instead of Sugarizer Server v0.9, Sugarizer Server v1.0 need mongodb, not only for statistics but also because there is a new login screen to automatically local browser journal into the server.

georgejhunt, 2018-05-18:
Running server from command line, I get

 root@box:/library/www/html/sugarizer-server# node sugarizer.js
{ Error: Cannot find module '../build/Release/bson'
    at Function.Module._resolveFilename (module.js:476:15)
    at Function.Module._load (module.js:424:25)
    at Module.require (module.js:504:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous>
(/library/www/html/sugarizer-server-1.0/node_modules/mongodb/node_modules/bson/ext/index.js:15:10)
    at Module._compile (module.js:577:32)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3) code: 'MODULE_NOT_FOUND' }
js-bson: Failed to load c++ bson extension, using pure JS version
Sugarizer Server is listening on port 8089...
Presence Server is listening on port 8039...

There's a dependency declared in package.json on mongodb "1.4.x" , but my
mongodb version is 2.4.14 -- so I changed package.json

    "dependencies": {
        "bson": "1.0.x",
        "express": "3.x",
        "http": "0.0.x",
        "ini": "1.1.x",
        "cors": "2.8.x",
        "moment": "2.19.3",
        "jwt-simple": "^0.2.0",
        "mongodb": "1.4.x" => "mongodb": "2.4.x",

Then I did a "npm rebuild",
But it did not fix the problem, so I changed it back.

But in actual fact there is no ipv4 port 8089 open:

root@box:/library/www/html/sugarizer-server# netstat -nat|grep 8089
tcp6       0      0 :::8089                 :::*                    LISTEN

root@box:/library/www/html/sugarizer-server#

So our error 500, is probably issued by apache2. We may need to move
sugarizer out of the document-root of apache, to keep from getting confused.

I think the line in the server startup:

    at Object.<anonymous>
(/library/www/html/sugarizer-server-1.0/node_modules/mongodb/node_modules/bson/ext/index.js:15:10)

needs to be corrected. It refers to the connection to mongodb. IIAB has
ipv6 turned off in most roles, so getting port 8089 to work on ipv4 seems
important.

Partial sugarizer.ini is as follows:

[information]
name = Sugarizer Server
description = Your Sugarizer Server

[web]
port = 8089

[security]
min_password_size = 4
max_age = 172800000
https = false
certificate_file = ../server.crt
key_file = ../server.key
strict_ssl = false

[client]
path t Object.<anonymous>
(/library/www/html/sugarizer-server-1.0/node_modules/mongodb/node_modules/bson/ext/index.js:15:10)
= ../sugarizer/

[database]
server = localhost
port = 27018
-- INSERT --                                                  18,1
Top

I feel way out of my comfort zone.

Guess I've done what I can for today.

llaske, 2018-05-18:
The error message just mean that the bson module is not compiled but the message

> using pure JS version

means that the module will works in interpreted mode.
So as the lines:

> Sugarizer Server is listening on port 8089... 
> Presence Server is listening on port 8039...

implied, it means that Sugarizer Server run correctly and should be accessible on port 8089.
Could you try a to launch node in background:

`node sugarizer.js &`

then try:

`wget http://localhost:8089`

to see what happen.

Are you sure that the 8089 port is visible from the outside ?
i.e. do you have an Nginx service running ?

georgejhunt, 2018-05-18:
Lionel's comment nudged me to flush iptables, and again try
http://10.10.123.206:8089.

And I was able to log in and bring up the activity circle. A smattering of
activities all appeared to work.

So we are getting closer

holta, 2018-05-18:
> I was able to log in and bring up the activity circle. A smattering of activities all appeared to work.

Extremely promising!  I'll remain hopeful for Sugarizer 1.0 in Chiapas, Mexico where RPi3 installations begin in 10 days from today.

Even if only partial support is possible.  For example SD cards are big enough in Mexico, that 1GB wasted due to prealloc files might be perfectly tolerable (during these initial deployments at approx 2 + 5 schools) e.g. if pending MongoDB issues are not resolved by then.

Let's discuss more Monday during our community/team call (http://minutes.iiab.io) &mdash; _@llaske are free to join a call any particular time(s) Monday (May 21st) to try to resolve pending issues?_

holta, 2018-05-24:
See PR #814 by @georgejhunt looking good towards implementing Sugarizer 1.0 (with MongoDB) on Raspbian and Ubuntu 18.04, even if now quite there yet &mdash; we could use input from @llaske on the [error appearing to be at line 237 of api/controlers/users.js](https://github.com/iiab/iiab/pull/814#issuecomment-391538650).

holta, 2018-07-09:
_CLARIFICATION that the MongoDB failure remains on Ubuntu 18.04: (not just the Node.js 8.x issue?!)_

Whereas on Raspbian/RPi "systemctl status mongodb" at least works, after IIAB 6.6/master attempts to install Sugarizer.  

Unfortunately on Ubuntu 18.04 it does even get this far &mdash; "./runrole sugarizer" completes (as if installing Sugarizer 1.0 properly) but then the result is:
```
root@box:~# systemctl status mongodb
● mongodb.service - High-performance, schema-free document-oriented database
   Loaded: loaded (/etc/systemd/system/mongodb.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2018-07-09 13:44:37 EDT; 2min 12s ago
  Process: 9326 ExecStart=/usr/bin/mongod -f /etc/mongod.conf (code=exited, status=100)
 Main PID: 9326 (code=exited, status=100)

Jul 09 13:44:36 box.lan systemd[1]: Started High-performance, schema-free document-oriented database.
Jul 09 13:44:36 box.lan mongod[9326]: note: noprealloc may hurt performance in many applications
Jul 09 13:44:36 box.lan mongod[9326]: 2018-07-09T13:44:36.735-0400 I CONTROL  [main] log file "/var/log/mongodb/mongod.log" exists; moved to "/var/log/mongodb/mongod.lo
Jul 09 13:44:37 box.lan systemd[1]: mongodb.service: Main process exited, code=exited, status=100/n/a
Jul 09 13:44:37 box.lan systemd[1]: mongodb.service: Failed with result 'exit-code'.
```

holta, 2018-07-09:
CLARIF 2: Sugarizer 1.0 (and MongoDB) do work on Debian 9.4 (as confirmed by PR #882) presumably because that OS contains much older libraries/dependencies like Node.js 4.8.2

(Unfortunately however this does not help Internet-in-a-Box implementers who've most all upgraded to Ubuntu 18.04 and very recent Raspbian.)

holta, 2018-07-10:
CLARIF: it's important we dump Node 4.x as it was EOL'd (end-of-life'd) on 2018-04-03.

Consequently both Ubuntu 18.04 and Raspbian made the (smart) decision to skip Node.js 6.x and went directly to Node.js 8.x, so it's time we get with the program :)

FWIW, according to https://en.wikipedia.org/wiki/Node.js#Releases :
- Node.js 6.x EOL's April 2019
- Node.js 8.x EOL's December 2019
- Node.js 10.x EOL's April 2021

holta, 2018-07-11:
An ancient version of npm will break your Sugarizer 1.0 installation as documented at https://github.com/iiab/iiab/pull/814#issuecomment-404297798.

So here is a comparison of {MongoDB, Node.js, npm} version numbers, as (currently) offered by mainline OS's when newly installed &mdash; right after updates were applied using "apt update; apt -y dist-upgrade".

...Raspbian Lite 2018-06-27: (after apt updates)
```
root@box:~# apt -a list mongodb nodejs npm
Listing... Done
mongodb/stable 1:2.4.14-4 armhf

nodejs/stable 8.11.1~dfsg-2~bpo9+1 armhf
nodejs/stable 4.8.2~dfsg-1 armhf

npm/stable 1.4.21+ds-2 all
```
...Ubuntu 18.04/Server: (after apt updates)
```
root@ubuntu-18-server:~# apt -a list mongodb nodejs npm
Listing... Done
mongodb/bionic 1:3.6.3-0ubuntu1 amd64

nodejs/bionic 8.10.0~dfsg-2 amd64

npm/bionic 3.5.2-0ubuntu4 all
```
...Debian 9.4/Desktop: (after apt updates)
```
root@debian-9:~# apt -a list mongodb nodejs npm
Listing... Done
mongodb/stable 1:3.2.11-2+deb9u1 amd64

nodejs/stable 4.8.2~dfsg-1 amd64
```
The key appears to be to (1) AVOID installing Raspbian's npm 1.4.21 and (2) DO use the version of npm that's now generally included within most OS's Node.js packages IF it's recent enough (3) OR IF ABOLUTELY NEC run "curl -sL https://deb.nodesource.com/setup_8.x | bash -" to populate /etc/apt/sources.list.d with nodesource.list containing something like:
```
deb https://deb.nodesource.com/node_8.x stretch main
deb-src https://deb.nodesource.com/node_8.x stretch main
```
...then run "apt update; apt install nodejs" (to install Node.js 8.x which comes with npm 5.x).

holta, 2018-07-11:
FYI the above nodejs packages offered by Raspbian (`stable 8.11.1~dfsg-2~bpo9+1 armhf`) and Debian 9.4 (`stable 4.8.2~dfsg-1 amd64`) do NOT include npm unfortunately :-(

So we're obliged to continue with Strategy Number (3) in many/most cases (as @jvonau pioneered for IIAB in 2017) but now with 8.x instead of 6.x:
```
curl -sL https://deb.nodesource.com/setup_8.x | bash -
```

holta, 2018-07-17:
Remaining Sugarizer 1.0 -> 1.0.1 issues summarized here:
https://github.com/iiab/iiab/pull/888#issuecomment-404370082

Highest Oriority Issues:
#898 Fix http://box/sugarizer (worked with Sugarizer 0.7, 0.8, 0.9)
#915 Yank power from MongoDB and see if Sugarizer 1.0 works on next boot(s) 
#899 Enable Debian 9 rsyslog service

holta, 2018-07-27:
Moving the "final polish" checklist to a new ticket for IIAB 6.7, as Sugarizer 1.0.1 now works quite well as part of IIAB 6.6.

closed 2018-07-27 by holta