iiab/iiab #799: Standardize upgrade/reinstall of IIAB services (server apps)

holta, 2018-05-16:
Finally we have quite rigorous documentation in 2018, explaining how to upgrade or reinstall most every IIAB service &mdash; for operators doing maintenance online *or* offline:

http://wiki.laptop.org/go/IIAB/FAQ#Can_I_upgrade_or_reinstall_server_apps.3F

However the approach above is very different for each &mdash; and sometimes imperfect &mdash; these different tacks/tricks should be harmonized into a more uniform approach.

_Yeah we can do better (-:_

CLARIF: IIAB services or server apps are defined in ([Ansible roles](https://github.com/iiab/iiab/tree/master/roles).  The main ones are summarized for teachers at http://wiki.laptop.org/go/IIAB/FAQ#What_can_I_do_with_E-books_and_Internet-in-a-Box.3F

holta, 2018-09-26:
SEE near-duplicate tkt:

#1041 "IIAB Apps need: Upgrades, Reinstalls, Backups, Accounts, Content, Audits (provisioning norms needed!)"

jvonau, 2020-04-29:
Think the 'needs install' and 'needs upgrade' can be clearly distinguished now with iiab_state. Continue the follow up in 1041

closed 2020-04-29 by holta