iiab/iiab #807: City Search failing in en-worldmap-10 on Ubuntu 18.04/Server

holta, 2018-05-20:
City search is not working on new U18/Server:
```
http:// iiab.me /modules/en-worldmap-10/map.html
```
Is #659 related?

Is "apt install php7.2-sqlite3" plausible?

tim-moody, 2018-05-20:
does ubuntu-18 need php7.2 declaration?

holta, 2018-05-21:
1) THX @tim-moody for the manual workaround on IIAB . ME where city search now works!

2) When searching on "Paris", the pointer above each of these 7 cities is broken — can you tell in the HTML what's causing this newly broken GIF/PNG ?

holta, 2018-05-21:
> 2. When searching on "Paris", the pointer above each of these 7 cities is broken — can you tell in the HTML what's causing this newly broken GIF/PNG ?

CONTEXT: these red "drops" (above each search result city, showing its place on the map) appear in Firefox on Windows and Internet Explorer on Windows.

_However these red "drops" do not appear in Firefox or Chrome on Android 7.0 &mdash; only a broken image is shown above each city in Android :-(_