iiab/iiab #816: Much nicer "calibre-web" front-end for Calibre

holta, 2018-05-24:
Thanks @darkenvy who discovered this much cleaner front-end, that allows anonymous browsing of books (avoids the need for accounts like box/box, mx/mx and m/m!) as well as a cleaner UX to edit ebook metadata etc:

https://github.com/janeczku/calibre-web
https://github.com/janeczku/calibre-web/blob/master/readme.md

This appeared on GitHub around August 2015 and was very actively developed by several people in 2016 and 2017.  Is installation sane and or complex?  Is installation possible/wise without Docker?

darkenvy, 2018-05-24:
I didn't see anonymous browsing mode. Should I look harder?

I threw it up the two different versions on http://noo.sh:31415 and
http://noo.sh:31416/ . Let me know what you think. They are both docker
images and I understand the concerns there. However, this shouldn't be too
hard to install natively.

holta, 2018-05-25:
@darkenvy we have a strong vote for the new UX [add-on [calibre-web](https://github.com/janeczku/calibre-web)] from @nataren who writes:

> Definitely http://noo.sh:31416 is SO much nicer. Let's use it if possible!

Plz LMK if you make any headway installing this natively on top of IIAB 6.5 (or any Calibre 3.23) as the Mexican and Peruvian schools would clearly both *love!*

holta, 2018-05-29:
The latest version of calibre-web has 2 known concerns:
- choosing "Language" field is much harder than within Calibre 3.23 itself, as there is no autocomplete
- data entered via calibre-web is not appearing within the same backend when viewed using Calibre 3.23 itself (http://noo.sh:31415)

darkenvy, 2018-05-29:
The second issue is probably a non-issue. If you use calibre-web we should
use it and it only. That way it is compatible with itself (and therefor not
a problem)

The first issue, sure. That is a downgrade. But is a feature non-the-less.

I advocate for the new version and am working on getting it working on the
pi. We can always do feature requests in the repo ;)

holta, 2018-05-30:
> I advocate for the new version and am working on getting it working on the
pi. We can always do feature requests in the repo ;)

This will be a Dream Come True, around the entire planet, if it works for BOTH:
* Teachers (empowered to change book metadata, add+remove books)
* Kids (with a dead-simple UX of any kind: so long as books are findable/sortable/searchable AND visually browsable by cover art for less literate folk !)

holta, 2018-06-03:
@darkenvy I've discovered Calibre is in active use on Internet-in-a-Box in Chiapas, Mexico, where teacher(s) are adding books using Calibre's existing web interface:

Does it make sense to transition them to calibre-web instead in the coming year?
And if so can this be added to [Calibre's existing playbook](https://github.com/iiab/iiab/tree/master/roles/calibre)?

holta, 2018-07-10:
@darkenvy : @nataren & I are wondering if we should hold off on adding calibre-web until the co-evolution of both products (Calibre & calibre-web) becomes even more clear in coming months?

What is your opinion?

(Certainly calibre-web is more beautiful, but if it's an ongoing Ansible maintenance/integration nightmare, teachers are certainly very happy with Calibre's own simple-if-ugly UI today!)

arky, 2018-07-16:
@holta I quickly looked at the code of the Calibre-web. It is clean and simple and has some very interesting features. 

It is possible to easily integrated as a service in IIAB. Perhaps shipping early preview of calibre-web could encourage the calibre-web developers and provide valuable feedback.  I could take a shot at this in a week or two. 

It uses port 8083 by default.  Is that acceptable or do you have any other suggestion?

holta, 2018-07-16:
> It is possible to easily [integrate] as a service in IIAB. Perhaps shipping early preview of calibre-web could encourage the calibre-web developers and provide valuable feedback. I could take a shot at this in a week or two.

Awesome: keep @darkenvy and @nataren in the loop, as both as very interested!

> It uses port 8083 by default. Is that acceptable or do you have any other suggestion?

Whatever you pick please publish it here and let me know if you modify https://github.com/XSCE/xsce/wiki/IIAB-Networking &mdash; GitHub's wiki does not send notifications very unfortunately :-(

arky, 2018-07-16:
Shall do @holta Please kindly assign this task to me. Thanks!

arky, 2018-07-17:
@holta Members of the IIAB organization with read permission to the repo should be able to self-assign issues.  Perhaps it is turning on the right settings in the github permission settings. 

https://help.github.com/articles/assigning-issues-and-pull-requests-to-other-github-users/

holta, 2018-07-17:
> @holta Members of the IIAB organization with read permission to the repo should be able to self-assign issues. Perhaps it is turning on the right settings in the github permission settings.
>
> https://help.github.com/articles/assigning-issues-and-pull-requests-to-other-github-users/

Brilliant!  I and others had given up on this being possible over the past year.  I hope people can self-assign issues etc now.

Currently there are 8 of us in IIAB's "Community" team, making  this possible:

https://github.com/orgs/iiab/teams/community (others are welcome to join!)

holta, 2018-07-18:
@arky am deferring to IIAB 6.7, as short+memorizable URL's like http://box/books are a higher priority for the teacher's I've spoken with.

(But don't let that stop you if you make progress on calibre-web too!)

arky, 2018-08-14:
@holta @darkenvy @nataren  I have a working playbook for Calibre-web here: 
https://github.com/arky/iiab/tree/calibre-web

Need to iron out issues related to provisioning. You can login using the default using following Username: admin Password: admin123 

Let me know if you find any bugs. Thanks!

holta, 2018-08-14:
> @holta @darkenvy @nataren I have a working playbook for Calibre-web here:
https://github.com/arky/iiab/tree/calibre-web

Great News !

> Need to iron out issues related to provisioning.

Schools often ask to upgrade to the latest [Calibre](https://calibre-ebook.com/whats-new) a few months later, as it's never sitting still :-)

Might it be possible to upgrade Calibre alone, while preserving calibre-web?  Or is it likely safer to upgrade both to the latest in most such cases?  (e.g. by running either of their individual Ansible playbooks/roles perhaps, while not losing E-Books / accounts / settings?)

Or if certain upgrade approaches are impossible or way too risky, please help me document those cases within http://FAQ.IIAB.IO Question 23 = "Can I upgrade or reinstall server apps?"

> You can set the default 'Location of the database' /library/calibre-web/
> Then you can login using the default Username: admin Password: admin123

Using the IIAB norm would be ideal, if at all possible!
```
Username: Admin
Password: changeme
```

arky, 2018-08-14:
After a break, I read through calibre-web source code. I realized that reads CALIBRE_DBPATH environment variable. 

I have added this variable to calibre-web.service file along with the -p command line option. 

Let me know this fixes the problem with provisioning the default IIAB Username: admin and Password: changeme.

holta, 2018-08-14:
> I read through calibre-web source code. I realized that cli.py reads CALIBRE_DBPATH environment variable.
>
> I have added this variable to calibre-web.service file along with the -p command line option.

Excellent news (nobody's asking for perfection, only iterating towards it ;)

Got time to generate a PR in coming days &mdash; making it easier for others to try out with [IIAB 6.6](https://github.com/iiab/iiab/wiki/IIAB-6.6-Release-Notes) ?

arky, 2018-08-15:
@holta Aye, aye mon cap'tan 
Sent a PR #999  after smoke-testing on Debian 10. I hope the community and the upstream developers would iron out the bugs with provisioning database.

holta, 2018-08-30:
@arky: @jvonau asks if a Python virtual environment (virtualenv) might be possible here, so that IIAB's Python system libraries remain safe ?

Would this be possible as a future upgrade for Calibre-Web, to make its integration with IIAB more robust?

(After looking into it last year, he believes Pypa.io can dangerously muddy the waters as to who is installing which Python package versions: https://github.com/iiab/iiab/issues/920#issuecomment-417057553)

holta, 2018-08-30:
PS @jvonau may be able to help mock up such a virtualenv here in the coming week, if that helps.

holta, 2018-08-30:
http://box/books works as part of the imminent IIAB 6.6 release, congrats to all!

Still: this ticket has been re-opened and is being moved to IIAB 6.7 so we can address @jvonau's larger question about integrity of Python packages across IIAB...

arky, 2018-08-30:
@holta @jvonau I think python flask web apps are implemented this way. Please kindly talk to upstream calibre-web developers.

Recommend closing this bug and continue the conversation about pypi in #920

I am getting really overwhelmed and confused with cross chatter between bugs. Perhaps I just need to get some more sleep.