iiab/iiab #822: IIAB 6.5 takes 45 sec to shut down (Rasp Lite on RPi3)

holta, 2018-05-28:
In the past (2017, and earlier in 2018) shutdowns consistently took 10-20 seconds.

(Not sure if Raspbian's own networking changes are to blame here ?)

holta, 2018-06-12:
Many related details within #656 ('IIAB 6.5 often takes 1.5MIN to shut down ("plymouth-reboot.service" on screen)')

holta, 2018-06-15:
IIAB is capable of a rapid shutdown during iiab-install's initial stages 0, 1, 2 and 3.

_Bissecting a [MIN-sized install](http://download.iiab.io/6.6/load-min.txt) on RPi provides strong evidence that the 3X slowdown (from ~15sec shutdowns to ~45sec shutdowns) arises during Stage 4, 7, 8 or in the networking stage._

holta, 2018-08-23:
FYI shutdown times are starting to spiral out of control again:

- At the end of 2017 and beginning of 2018, this had been contained to 10-to-20 seconds to shutdown, as measured across diversity of IIAB-on-RPi-3 shutdowns.
- Today a BIG-sized IIAB 6.6/master on RPI 3 B+ can take as many as 90 seconds to shut down, as I just now measured :-(

Conclusion: Definitely this needs another look, as we begin the IIAB 6.7 developer cycle!

holta, 2018-08-23:
FYI shutdown times were tested across 4 different Raspberry Pis, each running [BIG-sized](http://wiki.laptop.org/go/IIAB/local_vars_big.yml) IIAB 6.6/master:

    70s Rasp Desktop, RPi 3 B+
    76s Rasp Lite, RPi 3
    79s Rasp Lite, RPi 3 B+
    85s Rasp Lite, RPi 3 B+

Conclusion: average shutdown time was 77.5 seconds.

arky, 2018-08-27:
@holta I noticed the squid http proxy does take time to shutdown properly ( https://github.com/iiab/iiab/blob/master/vars/local_vars_big.yml#L61-L63 )

It is not the right tool but what is the output of  'systemd-analyze blame'

holta, 2020-09-11:
Occasional flare-ups still occur (very slow shutdowns of RPi IIAB's) so this is just a bunch of general tips...towards helping everyone diagnose/solve these situations:

_'Shutdown Taking Too Long? Here’s How to Investigate and Fix Long Shutdown Time in Linux'_

https://itsfoss.com/long-shutdown-linux/

holta, 2020-09-15:
Current Example:

#2526 'Lokole delays IIAB shutdowns: "A stop job is running for Celery service (1min 34s / 1min 34s)"'

jvonau, 2021-08-17:
With squid's disk cache now enabled I would expect this power off issue to pop back up when squid is in use.

holta, 2021-08-18:
> With squid's disk cache now enabled in #2948 I would expect this power off issue to pop back up when squid is in use.

Excellent warning.
That folks need to watch out for.

We can and should document this if it's confirmed.

jvonau, 2025-01-27:
status?

holta, 2025-01-27:
No longer common, and likely no longer relevant.

closed 2025-01-27 by holta