iiab/iiab #823: RPi3 cannot handle more than ~20 WiFi connection from Android?

holta, 2018-05-30:
Currently being discussed on xsce-devel mailing list...

Unlike in 2016, 32 simultaneous WiFi connections to RPi 3 (running the latest Raspbian Lite from 2018-04-18) no longer seems possible :/

holta, 2018-07-18:
Tangentially Related: TK Kang is observing total WiFi failure about an hour after boot (of IIAB 6.6 on Raspbian 2018-06-27) with symptoms similar to: https://github.com/raspberrypi/linux/issues/2453

Hopefully he and @jvonau can write up a more proper bug report ticket, so others can verify whether this is-or-is-not related to upstream firmware issues with Raspberry Pi's Wi-Fi?

jvonau, 2018-07-18:
Perhaps disabling the promiscuous mode hack from #638 might help or hurt but lets take that out of the picture as the firmware/kernel have been updated since the introduction of that 'fix'

holta, 2018-07-19:
As TK has reconfirmed the pattern in comprehensive tests on both Raspbian Lite and Raspbian Desktop... 

> With Internet access from Ethernet cable and connection to the the RPi3 Wi-Fi, will fail after an hour. Sometimes can't connect to the Wi-Fi AP and if it is connected it is not able to access http://box, etc.
> 
> If I boot with Ethernet cable unplugged  (i.e. no Internet), the wifi access and http://box, etc will remain stable for hours.

...I've published this as a "Known Issue" at the bottom of the Release Notes:

https://github.com/iiab/iiab/wiki/IIAB-6.6-Release-Notes#known-issues

holta, 2019-12-21:
Raspberry Pi 4's tested in classrooms in Oaxaca State, Mexico did not allow more than 16 simulataneous WiFi connections to its internal hotspot.

Whereas this is not true on Raspberry Pi 3, which allows up to 32 simultaneous WiFi connections, even with the very same microSD card/image as used in the above RPi 4's.

_The Raspberry Pi 3 B+ should be verified, to reconfirm whether this one still supports 32?_

@tim-moody is investigating different ways to make this essential load-testing (of Raspberry Pi's internal WiFi hotspot/AP) happen, alongside threads like https://raspberrypi.stackexchange.com/questions/50162/maximum-wi-fi-clients-on-pi-3-hotspot

Background: Mexico rollout & testing used Raspbian With Desktop with kernel 4.19.86 as supplied by the `rpi-update` command, as was required to work around Raspbian bug #1993.

holta, 2019-12-26:
Question relating to RPi 4 supporting only 16 simultaneous WiFi connections: (whereas RPi 3 supported 32 simultaneous WiFi connections)

https://github.com/seemoo-lab/nexmon/issues/371

holta, 2019-12-26:
Possibly related:

Raspberry Pi 4 BCM43455 NVRAM Addition Hits Linux-Firmware Git
https://www.phoronix.com/scan.php?page=news_item&px=Raspberry-Pi-4-NVRAM-FW-Git

holta, 2020-01-04:
Absolutely central issue, but [IIAB 7.1](https://github.com/iiab/iiab/milestone/6) cannot realistically fix it this month, so let's stay on it during the [IIAB 7.2](https://github.com/iiab/iiab/milestone/7) release cycle!

mitra42, 2020-01-05:
Just adding myself on this, to follow progress, can't help much with testing.

holta, 2020-02-25:
Results after 3 hours of testing Raspbian's hostapd alongside dnsmasq today:

- **RPi 3 allowed a MAXIMUM of 24 simultaneous WiFi connections.**  (FYI its IIAB microSD was built on Raspbian Desktop 2019-09-26, rpi-updated to kernel 4.19.86)

- **RPi 3 B+ allowed a MAXIMUM of 16-to-18 simultaneous WiFi connections.**  (FYI its IIAB microSD was built on Raspbian Desktop 2019-09-26, rpi-updated to kernel 4.19.86)

  Connection count was consistently limited to 18, then 17, then 16 simultaneous WiFi connections (for about 20 min each).

  When the observed limit was 18 (during the first ~20 minutes) we watched the actual number very briefly fall to 17 about every minute or two.

- **RPi 4 (2GB RAM) allowed a MAXIMUM of 18-19 simultaneous WiFi connections.**  (FYI its IIAB microSD was built on Raspbian Desktop 2020-02-13, WITHOUT running rpi-update)

  Connection count was consistently limited to 18 most of the time, but sometimes 19 for shorter periods of time.

CONTEXT:
- 32+ WiFi client devices included 15 RPi 4's, 17 Ubermix (Linux) laptops, and a wide assortment of smartphones, a MacBook, a Windows 10 laptop etc:
- The following 2 lines were needed in /etc/hostapd/hostapd.conf

   ```
   ctrl_interface=/var/run/hostapd
   ctrl_interface_group=0
   ```
- We then ran `systemctl restart hostapd`
- The number of simultaneous WiFi connections to each RPi was continuously monitored by counting MAC addresses using `hostapd_cli all_sta | grep : | wc`

mitra42, 2020-02-25:
Was there a reason that you skipped Rpi4 with rpi-update ?

holta, 2020-02-25:
> Was there a reason that you skipped Rpi4 with rpi-update ?

Long story but we'll explain more later in the week after running more tests, hopefully including that test too if there is time!

holta, 2020-02-25:
Further reconfirming yesterday's test results...

- **RPi 4 (2GB RAM) allowed a MAXIMUM of 18 simultaneous WiFi connections.** (Using the _EARLIER_ version of the Raspbian Desktop OS from 2020-09-26, updated to kernel 4.19.86 using rpi-update)

jvonau, 2020-02-25:
I view hostapd_cli as a debugging tool, ctrl_interface provides a management interface but is not needed to configure a simple access point but opens the door for live external changes via hostapd_cli, this could be viewed as problematic with pi having access to sudo. `hostapd_cli list_sta` returns just the 'associated' list of clients without the need to grep

holta, 2020-02-26:
Important News:

- **RPi 4 (2GB RAM) allowed a maximum of 32 simultaneous WiFi connections (to the RPi 4's internal WiFi hotspot) when running IIAB on Ubuntu 19.10.1**

  This is almost double the ~18 WiFi connections allowed by Raspbian!

Central Questions:

- What firmware/driver is making this superior functionality possible within Ubuntu 19.10.1?

- Who released it and when?

- What firmware/driver causes Raspbian to support only ~18 WiFi connections on this very same hardware?

holta, 2020-02-26:
Tangentially related:

#2265 Raspbian blocks hostapd when IIAB microSD is built in RPi 4 and then moved to RPi 3 or 3 B+ (or Zero W?)

jvonau, 2020-02-26:
> Important News:
> 
>     * **RPi 4 (2GB RAM) allowed a maximum of 32 simultaneous WiFi connections (to the RPi 4's internal WiFi hotspot) when running IIAB on Ubuntu 19.10.1**
>       FWIW Ubuntu 19.10.1 is currently running a 5.3.x Linux kernel, instead of Raspbian's 4.19.x kernel.
> 
> 
> Central Questions:
> 
>     * What [WiFi firmware/driver](https://wiki.ubuntu.com/ARM/RaspberryPi#WiFi) is making this superior functionality possible within Ubuntu 19.10.1?
> 

From the link above:
    linux-firmware-raspi2 - GPU firmware bootloader files and WiFi firmware configuration files
    linux-raspi2 (linux-image-raspi2) - Linux kernel with patches from https://github.com/raspberrypi/linux

>     * Who released it and when?
The answer lies within dmesg, look for lines like this:
[    4.124828] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.137439] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04

holta, 2020-03-09:
@georgejhunt's related Ubuntu 19.10.1 research:

PR #2269 "Enable aarch64 in kiwix, disable stem.php"

tim-moody, 2020-03-09:
> The answer lies within dmesg, look for lines like this:

```
[   11.944398] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   11.975764] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4

```

georgejhunt, 2020-03-12:
The equivalent dmesg from 20.04:
```
[    9.438936] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.475030] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
```

holta, 2020-03-12:
> @georgejhunt's related Ubuntu 19.10.1 research:
> 
> PR #2269 "Enable aarch64 in kiwix, disable stem.php"

Thanks to @georgejhunt who is further documenting his Ubuntu 20.04 LTS research (of their daily pre-release builds) here:
- #2270 "Ubuntu 20.04 aarch64 early testing"
- #2271 "What's the best way to monitor RPi's CPU temp when running Ubuntu 19.10.1+ ?"

holta, 2020-04-02:
Very new WiFi firmware is seen below...running on the latest Raspbian Lite 2020-02-13 after it was updated by `apt update; apt -y dist-upgrade`

> root@box:~# dmesg | grep brcmfmac
...
[    4.107998] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.121606] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2

> root@box:~# uname -a
Linux box.lan 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

holta, 2020-04-12:
@shanti-bhardwa indicated (earlier this week) that Ubuntu 20.04 pre-releases for RPi may have moved wifi firmware forward (from 19.10.1's "Mar 1 2015") to "Feb 27 2018" according to his dmesg output here:

> [    9.901286] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.922853] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32

jvonau, 2020-04-13:
https://bugs.launchpad.net/ubuntu/+source/linux-firmware-raspi2/+bug/1862760

holta, 2020-04-13:
> https://bugs.launchpad.net/ubuntu/+source/linux-firmware-raspi2/+bug/1862760

Great find!

jvonau, 2020-04-26:
Adam noted these over time:
> [72259.791893] ieee80211 phy0: brcmf_run_escan: error (-52) 
[72259.794685] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)

While I noted these on boot
>[   21.271547] device ap0 entered promiscuous mode
[   21.492845] br0: port 1(ap0) entered blocking state
[   21.492856] br0: port 1(ap0) entered forwarding state
[   21.567374] ieee80211 phy0: brcmf_cfg80211_disconnect: error (-52)
[   21.879347] ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[   21.879402] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)

I think these are related to how netplan constructs its conf file used with wpa_supplicant. Without declaring the county code beforehand I think the ieee80211 routine would try to quiz the upstream AP as to what the regulatory domain that should be used but can't as hostapd already started the device's firmware with the country code set. The upstream wifi device still works and I have not noted this error while raspbian is running as the country code is already set in the running wpa_supplicant.conf file. Updated LP 1862760 with my thoughts.

holta, 2020-04-26:
> Adam noted these over time:
> 
> > [72259.791893] ieee80211 phy0: brcmf_run_escan: error (-52)
> > [72259.794685] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)

This seems to happen every 4-5 hours on average, according to the output of `dmesg` summarizing the first 43 hours after boot (of machine 183-rpi4-2g-u20srv-BIG-2337-0423 = 10.8.0.66) which was largely idle during those 43 hours:

> [ 5238.939880] ieee80211 phy0: brcmf_run_escan: error (-52)
[ 5238.942428] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[72259.791893] ieee80211 phy0: brcmf_run_escan: error (-52)
[72259.794685] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[79084.879964] usb 1-1.4: USB disconnect, device number 3
[119814.152785] ieee80211 phy0: brcmf_run_escan: error (-52)
[119814.155361] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[127638.124394] ieee80211 phy0: brcmf_run_escan: error (-52)
[127638.126930] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[127639.135015] ieee80211 phy0: brcmf_run_escan: error (-52)
[127639.140722] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[139446.536809] ieee80211 phy0: brcmf_run_escan: error (-52)
[139446.539168] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[139447.544562] ieee80211 phy0: brcmf_run_escan: error (-52)
[139447.549764] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[149879.966982] ieee80211 phy0: brcmf_run_escan: error (-52)
[149879.969182] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[149880.974777] ieee80211 phy0: brcmf_run_escan: error (-52)
[149880.979801] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)

holta, 2020-04-26:
@jvonau suggested this as a possible workaround &mdash; to the "brcmf_run_escan: error (-52)" errors on the new Ubuntu 20.04 LTS on Raspberry Pi:

- Edit `/run/netplan/wpa-wlan0.conf` adding `country=US` below line `ctrl_interface=/run/wpa_supplicant`
- Run these 2 commands:
  - systemctl daemon-reload
  - systemctl restart netplan-wpa-wlan0
- Unfortunately this last step did not work for me (not sure why!) but for @jvonau it did work...look for the word 'COUNTRY' which you should find...in the output of these commands:
  - systemctl status netplan-wpa-wlan0
  - journalctl -b 0 -u netplan-wpa-wlan0.service -l --no-pager | grep -i country

holta, 2020-05-09:
For mysterious reasons, older WiFi firmware from "Oct 22 2019" has suddenly appeared on one individual's "Raspbian with desktop" (stage4) 2020-02-13 in the output of iiab-diagnostics (which runs `dmesg | grep brcm`) here:

> [    7.608434] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd

Per: https://github.com/iiab/iiab/issues/2391#issuecomment-625989542

Whereas my own "Raspbian with desktop" (stage4) 2020-02-13 shows WiFi firmware from "Mar 2 2020" here:

> [    4.064555] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2

>

holta, 2020-05-22:
I'm moving this to the [IIAB 7.2](https://github.com/iiab/iiab/milestone/7) Milestone as it's likely impossible to (re)test the hotspot capabilities of Raspbian & the new Ubuntu 20.04 (on hardware like RPi 3, RPi 3 B+ and RPi 4) prior to [IIAB 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes)'s release expected before the end of month!

holta, 2020-06-22:
There's serious concern that Raspberry Pi OS (and presumably Ubuntu-on-RPi too later?) might soon prevent more than 7 simultaneous WiFi clients / connections:

https://github.com/raspberrypi/firmware/issues/1403

mitra42, 2020-06-22:
Have you asked Raspberry ? I made the connection for you about 6 months ago but you didn't want to pursue it till you'd done more testing.

holta, 2020-06-22:
@mitra42 I've not spoken with the Raspberry Pi foundation.

If this latest development (a hard limite of 7 WiFi client devices) is rolled into the Raspberry Pi OS (in coming weeks presumably?) it will be genuinely disastrous, from the perspective of schools all around the world :-(

mitra42, 2020-06-22:
Matt (ED of RPI foundation) said: It's great to hear from you. I don't have any info on the issue you're having with the 4's. Have you already been to the forums with the issue? https://www.raspberrypi.org/forums/ That's probably the best bet.

AFAIK you never took him up on that suggestion, which would have then opened the possibility to escalate it.

georgejhunt, 2020-06-29:
Here are firmwares from ubuntu, that display beacons, and pass data on top of a rapsberry OS 2020-05-07 userland file system.

1. Verified in Feb to support 32 clients (when on top of a 64bit ubuntu 19.10 userland): http://download.iiab.io/packages/brcmfmac43455-sdio.bin.ubuntu.1910 
  
 root@ubuntu:~# strings /lib/firmware/brcm/brcmfmac43455-sdio.bin | grep Version
43455c0-roml/sdio-ag-p2p-pno-aoe-pktfilter-keepalive-mchan-pktctx-proptxstatus-ampduhostreorder-lpc-pwropt-txbf-wnm-okc-ccx-ltecx-wfds-wl11u-mfp-tdls-ve
Versio: 7.45.18.0 CRC: d7226371 Date: Sun 2015-03-01 07:31:57 PST
Ucode Ver: 1026.2 FWID: 01-6a2c8ad4

2. From latest Ubuntu 20.04 (client count unknown): http://download.iiab.io/packages/brcmfmac43455-sdio.bin.ubuntu.20.04

root@box:/lib/firmware/brcm# strings brcmfmac43455-sdio.bin|grep Version
43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-lpc-pwropt-43455_ftrs-wfds-mfp-dfsradar-wowlpf-idsup-idauth-noclminc-clm_min-obss-obssdump-swdiv
Version: 7.45.154 (r684107 CY) CRC: b1f79383 Date: Tue 2018-02-27
03:18:17 PST Ucode Ver: 1043.2105 FWID 01-4fbe0b04

To test: Copy one of the above to /lib/firmware/brcm/brcmfmac43455-sdio.bin (changing the name), and reboot

shanti-bhardwa, 2020-07-16:
tested with iiab 7.1 running on raspbian-lite on Pi4 with "out of the box" Raspberry Pi OS (32-bit) Lite
Minimal image based on Debian Buster, Version:May 2020,Release date:2020-05-27,Kernel version:4.19.

Can ONLY connect 4 WIFI desktop clients based on various models of PI hardware (PI Zero W, PI 2B, PI 3B, PI3B +, PI 4) running "out of the box" desktop Raspberry Pi OS (32-bit) with desktop, Image with desktop based on Debian Buster, Version:May 2020, Release date:2020-05-27,Kernel version:4.19.

My hostapd is based on ap0 with channel g - 2.4Ghz 

5th device cannot connect. 

Repeated the tests with the same clients, but this time iiab 7.1 hostapd server running on Ubuntu 20.04 Server Image + Mate Desktop on Lenovo X240 laptop with x86_64bit version 5.4.0.40-generic Kernel.
So far have managed to connect 10 desktop clients, including Galaxy-S7 phone, Android Tablet and Amazon Fire Kindle.

So I am confident that my test approach is correct. Will write up on how I have done the testing and some of the results.

The issue is with PI Raspbian Firmware

tim-moody, 2020-07-21:
So is the consensus that we should always use the firmware in ub19.10, even though it is old?

holta, 2020-07-22:
How many WiFi client devices can a Raspberry Pi hotspot (hostapd) support?

PRELIM TESTING RESULTS FROM @holta & @jvonau, using 35 actual XO-1.5 laptops running OLPC OS 13.2.11:

3 B+ (assumed) & RPi 4 (tested) -- version & date found by running `strings /lib/firmware/brcm/brcmfmac43455-sdio.bin | grep Version`
| Max WiFi Clients | Firmware Version | Compile Date |
| --------------- | ------------- | ---- |
| 32 | 7.45.18.0 | 2015-03-01 |
| 15 | 7.45.154 | 2018-02-27 |
| 3-5-crashes-firmware | 7.45.202 | 2020-03-02 <<< used by RaspiOS 2020-05-27 + apt updates as of 2020-07-22 |
| 6 | 7.45.206 | 2020-03-23 |
| 6? | 7.45.214 | 2020-05-22 |

Zero W (tested) & RPi 3 (tested) -- version & date found by running `strings /lib/firmware/brcm/brcmfmac43430-sdio.bin | grep Version`
| Max WiFi Clients | Firmware Version | Compile Date |
| --------------- | ------------- | ---- |
| 10 | 7.45.98.94 | 2019-10-22 <<< used by RaspiOS 2020-05-27 + apt updates as of 2020-07-22 |
| 28 | 7.45.98.38 | 2017-10-23 |

ABOVE 7.45.98.38 CAN BE INSTALLED USING THESE INSTRUCTIONS FROM THE BOTTOM OF https://www.raspberrypi.org/forums/viewtopic.php?t=249750 ...
- wget https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-brcm80211_20161130-3+rpt3_all.deb
- dpkg -i firmware-brcm80211_20161130-3+rpt3_all.deb
- reboot

@jvonau question for another day: would enabling WPA2 authentication reduce the number of WiFi client devices possible?

Ref: https://github.com/raspberrypi/firmware/issues/1403

jvonau, 2020-07-22:
Remember ap0 doesn't connect upstream, wlan0 does and that might cost a connection, something to test to find out if it does.

holta, 2020-07-22:
1) Just FYI I've (begun) cleaning up the table describing published IIAB-on-RPi prefab images, so we can (start) to get the word out to developing world schools/clinics about this very sad and very dramatic WiFi capacity reduction befalling their Raspberry Pi's:

   https://github.com/iiab/iiab/wiki/Raspberry-Pi-Images:-Summary

   (Due to Cypress firmware "regressions" over the past 2-3 years: apparently they are taking away RAM to use it for other bleeding-edge features, effectively sabotaging developing world education, as suggested off of raspberrypi/firmware#1403.)

2) Internet-in-a-Box will seek a short-term workaround, likely reverting to 2015/2017 WiFi firmware for now (similar to Ununtu 19.10.1 on RPi) so that schools/educators &mdash; and medical community hotspot operators &mdash; can at least get their Raspberry Pi community/learning hotspots working again.

3) Longer term we need to ask the Raspberry Pi Foundation to clarify directly when they talk to Cypress, what's really going on here.  i.e. WHY educational applications (like learning hotspots, medical hotspots, and the like) are being rendered unworkable after years of success here.  

   And what can be done to reverse this ugly and frankly tragic pattern :/

holta, 2020-07-25:
Conclusion: Testing over the past 4 days makes very clear that most every IIAB implementer needs the 2 firmware files from Ubunu 19.10.1 -- let's move forward to make that happen now.

So what role/tasks/file.yml should...?

- download both 19.10.1 WiFi firmware files (from http://d.iiab.io/packages) placing them in /lib/firmware/brcm
- lock these 2 files in place by running...
  `apt-mark hold firmware-brcm80211` (on RaspiOS-on-RPi)
  or...
  `apt-mark hold linux-firmware-raspi2` (on Ubuntu-for-RPi)

Tangentially Related:

PR #2470 "No /etc/network/interfaces.d/iiab [fixes RPi 3 B+, allowing WiFi firmware & hostapd to launch reliably on boot]"

jvonau, 2020-07-27:
Using `apt-mark hold firmware-brcm80211` might block a future kernel update should upstream change the dependence to include an absolute or minimium version match. Think it would be just easier to copy the firmware in place as needed, just put the logic check for the firmware/user's wish or just blindly replace as required in iiab-hotspot-on. The instructions echoed to the user in iiab-hotspot-on instructs to reboot, that covers a firmware replacement's needs. Don't over think it, kiss it.

holta, 2020-07-30:
After 10 days of very thorough testing, it seems the IIAB default should be to drop these 3 files into /lib/firmware/brcm &mdash; for most all installs onto RPi:

- http://d.iiab.io/packages/brcmfmac43455-sdio.bin_2015-03-01_7.45.18.0_ub19.10.1 (allows 32 WiFi connections to RPi 3 B+ & RPi 4)
- http://d.iiab.io/packages/brcmfmac43430-sdio.bin_2018-09-11_7.45.98.65 (allows 30 WiFi connections to RPi Zero W & RPi 3)
- http://d.iiab.io/packages/brcmfmac43430-sdio.clm_blob_2018-09-28_7.45.98.65 (necessary for the file just above)

Of course renaming them appropriately.

The hard part remains how to "lock" them in place, protecting the Access Point's ability to serve 30-32 clients &mdash; even if unattended upgrades or unexpected apt updates spontaneously later overwrite any of these 3 files.

I like @jvonau's suggestion that we consider implementing some such protection logic in [/sbin/test-wifi](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/test-wifi) associated with [wifi-test.service](roles/network/templates/hostapd/wifi-test.service.j2) ...such that these 3 files are mandated on every boot (and perhaps verified for undesired changes?).

After restoring the correct such firmware file(s), while a rare event, would also require a 2nd reboot.  As every IT person knows: rebooting is often necessary, if at first you don't succeed...  (So long as we avoid a worst-case "thrashing" scenario e.g. if some apt-like update keep trying to re-upgrade firmware file(s) after every boot.  Apt should not try to re-update / re-upgrade on every boot, so I think we are safe here.)

holta, 2020-08-01:
1) Is [/sbin/test-wifi](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/test-wifi) the best place to copy the 3 firmware files during every RPi boot, e.g. better than [/usr/libexec/iiab-startup.sh](https://github.com/iiab/iiab/blob/master/roles/2-common/templates/iiab-startup.sh) and other places?

   Something like these 4 lines should do the job?

   ```
   cd /lib/firmware/brcm
   rsync -tb --suffix=.`date +%Y-%m-%d_%H:%M:%S` brcmfmac43455-sdio.bin_2015-03-01_7.45.18.0_ub19.10.1 brcmfmac43455-sdio.bin
   rsync -tb --suffix=.`date +%Y-%m-%d_%H:%M:%S` brcmfmac43430-sdio.bin_2018-09-11_7.45.98.65 brcmfmac43430-sdio.bin
   rsync -tb --suffix=.`date +%Y-%m-%d_%H:%M:%S` brcmfmac43430-sdio.clm_blob_2018-09-11_7.45.98.65 brcmfmac43430-sdio.clm_blob
   ```

   ( 'rsync -t' preserves timestamp when copying.  'rsync -b' creates a backup file, whenever overwriting.  'rsync -c' = 'rsync --checksum' could also be considered but this is overkill [by default rsync already compares SOURCE/DEST timestamps and sizes as a "good enough" heuristic to decide if copying is nec].  In any case, rsync is far better than 'cp -p -u' [update only if newer] timestamp games, towards avoiding wear+tear [writing] to microSD on every boot, i.e. some heuristic to... Copy Only If Nec.)

2) What's a good stage/role and tasks/*.yml to wget the 3 firmware files (7.45.18.0 & 7.45.98.65 with its own .clm_blob) downloaded from http://d.iiab.io/packages directly to /lib/firmware/brcm ?  (So they're very visible to any operator poking around there, keeping the explanatory suffixes e.g. *_compile-date_firmware-version_os's-that-also-use-this etc).  Possibly [roles/1-prep/tasks/raspberry_pi.yml](https://github.com/iiab/iiab/blob/master/roles/1-prep/tasks/raspberry_pi.yml) for such essential driver/firmware?  Or in [roles/network/tasks/rpi_debian.yml](https://github.com/iiab/iiab/blob/master/roles/network/tasks/rpi_debian.yml) though the Debian name in rpi_debian.yml appears out of place (?) since we want this to run on _all_ Rapberry Pi OS's.

3) Of course with some kind of opt-out var in /etc/iiab/local_vars.yml (e.g. wifi_hotspot_capacity_rpi_fix: True -> False) for those who really want the OS's latest firmware(s), or their own hand-installed firmware(s) etc.

holta, 2020-09-12:
Good enough for now, with IIAB Raspberry Pi's once again being are to serve 30-32 Wi-Fi client devices!

This ticket can be re-opened for [IIAB 8.0](https://github.com/iiab/iiab/milestone/8) if later nec.

closed 2020-09-12 by holta

jvonau, 2021-04-20:
https://blog.quarkslab.com/reverse-engineering-broadcom-wireless-chipsets.html

pelwell, 2021-07-09:
FYI I've opened an issue requesting feedback about a streamlined firmware for using Pi 4 (and 3B+) as an AP: https://github.com/iiab/iiab/issues/2853