iiab/iiab #827: Enable easy backup & duplication of IIAB microSD's

holta, 2018-06-06:
@georgejhunt is working hard to design this: so that even schools without modern equipment and IT staff should be able to back up and duplicate their Internet-in-a-Box microSD cards.

Alongside @tim-moody is outlining educators/operators to copy Content Packs to/from USB memory sticks and USB-C Nexter storage.

georgejhunt, 2018-06-08:
- I happened to have a Samsung UHS1 128GB card. The first time I wrote to it, I got 10MB write speed, which seems to be the spec, using rpi3. But after the first experiment, the write speed decreased to 5MB.
- At this rate, it will take 7+ hours to copy a 128GB source.
- Adam you have the most experience duplicating, and were suggesting  USB3 on a recent laptop was required (or were you speaking of UHS3 - the card spec?).  I think perhaps the limits I have seen are built into SD spec. What are the specs/make of the 128GB that you use?
- I believe that 10MB/sec (new card) will copy 128GB in less than 4 hours.  I think that is good enough.

holta, 2018-06-08:
> At this rate, it will take 7+ hours to copy a 128GB source

Hmm I've never seen a copy take that long.  Even when using RPi3 to perform the copying, as I used to do in late 2016 or early 2017.

holta, 2018-06-11:
1) In addition to minifying/truncating (allows copying SD cards to any brand/make/model) let's also consider:
  A) deleting /etc/iiab/uuid "on target not source"
  B) modifying /etc/iiab/handle to become "COPIED-FROM-$oldstring"
  C) de-identifying all possible personal/private data and logs "on target not source"

2) Excellent info on SD card write/read speeds here:

https://havecamerawilltravel.com/photographer/fastest-microsd-cards/amp/

holta, 2018-06-12:
Cleanly-designed ("no Cisco training required") UI/UX is a central challenge in the developing world, in NGO's, and in government communities — as we are constantly reminded:

 #743 ("Instructions to create sdcard on mac")

holta, 2018-06-20:
See @georgejhunt's ongoing [research/notes](https://github.com/georgejhunt/iiab/wiki/SD-Card-Duplication-Explorations).

holta, 2018-07-08:
Seems everyone except for me uses http://Etcher.io so 2 things I should have realized:

1. While Etcher.io claims to be multiplatform it does not *yet* run well on the ARM version of Raspbian:

   https://github.com/resin-io/etcher/issues/1992 "Trying to install on a Raspberry Pi"
   https://github.com/resin-io/etcher/issues/1993 "Fix ARM build tooling"

2. https://etcher.io/cli/ is a beautiful set of tools we may want to keep our eye on... (and/or contribute to?)

arky, 2018-07-08:
Few  more resources, I like the design of rpi-clone. 

https://github.com/billw2/rpi-clone

https://github.com/raspberrypi-ui/piclone

http://sysmatt.blogspot.com/2014/08/backup-restore-customize-and-clone-your.html

https://pastebin.com/qVcWiVsz

holta, 2018-07-27:
Please see @georgejhunt's prototype emerging here:
https://github.com/iiab/iiab-factory/tree/master/box/rpi/imager/

holta, 2018-08-06:
ASIDE:

[PiShrink](https://github.com/Drewsif/PiShrink) is a [new, popular] bash script that automatically shrink a pi image that will then resize to the max size of the SD card on boot.

holta, 2018-08-06:
@georgejhunt & All, fyi @nataren is eager to assist you getting this:
- _cleanly translated into Spanish!_
- natively running on Windows (if a very small binary might be achievable, for those who are unable to download ~500 MB TinyCore Linux)
- if the above is possible, a fully-contained executable for MacOS and Linux newbies (true newbies) might also fly?!

georgejhunt, 2018-08-06:
Will be great to have help on Spanish translation. I created a pot file at
https://github.com/georgejhunt/iiab-factory/blob/i18n/box/rpi/imager/imager.pot

> @georgejhunt <https://github.com/georgejhunt> & All, fyi @nataren
> <https://github.com/nataren> is eager to assist you getting this:
>
>    - *cleanly translated into Spanish!*
>    - natively running on Windows (if a very small binary might be
>    achievable, for those who are unable to download ~500 MB TinyCore Linux)
>
> Running natively on a variety of platforms seems problematic to me. More
code means more chance to screw up. Code does not become reliable without
extensive use. Maintenance (bug fixing, and moving the goal posts) is
simpler if it is all in one script.

My initial choice was to install a Tiny Core GUI, and to make the user
experience as close to what she already knows on windows. If that is not
the target, it may be possible to reduce the Tiny Core download to 80MB.

>
>    - if the above is possible, a fully-contained executable for MacOS and
>    Linux newbies (true newbies) might also fly?!
>
> Of course the bash script will run on rpi, and an external hard disk is a
short $50 stretch.

I have read that UEFI is used on the MAC, but have no experience there. I
think there are linux drivers for the MAC file system, just as there are
for ntfs.

Maybe I'm missing something. What would be the primary advantage or running
natively on windows? or Mac?

holta, 2018-08-06:
> it may be possible to reduce the Tiny Core download to 80MB.

Fascinating: a native Windows installer is likely not necessary if ~500MB is reducible to ~80MB !?

> I have read that UEFI is used on the MAC, but have no experience there. I
think there are linux drivers for the MAC file system, just as there are
for ntfs.
>
> Maybe I'm missing something. What would be the primary advantage or running
natively on windows? or Mac?

Native or not, the Eurocentric NGO community almost all use MacBooks and iPhones.

A very-high-level technology decider within the Internet Archive (http://archive.org) personally asked me about this last Thursday.  As IIAB grows, we fundamentally need more ways for the "rich helpers" to implement IIAB very quickly & efficiently when they're on the road — in all countries.

As confounding as this is: the polarization between rich and poor stares us in the face when Peace Corps Volunteers (and similarly sincere international helpers) are _generally_ "addicted" to their expensive iOS / MacOS "Mercedes" devices — even when working directly among global citizens who _generally_ use Windows and Android "scooters..."

georgejhunt, 2018-08-06:
Success loading TinyCoreLinux on my MacBook! (just hold down option key
during reboot, and select UEFI)

As an experiment, I plugged in the USB stick developed for Windows, (with
UEFI boot capability), and was surprised to load the Tiny Core linux
kernel, and modules. It got stuck somewhere in bringing up the gui, and the
autostart menu.

But I think a one size, pared-down zipped image, less than 80MB, that runs
on Windows, rpi, and MacBooks, seems achievable.

Is that closer to what we really need?

Caveats:

   1. Will not run on older hardware, where Bios will not accept USB as
   boot device.
   2. Might require internet, and another script to load required
   dependencies as raspbian packages to run natively on the rpi (they are
   already part of the package in the Tiny Core image).
   3. The "help" function for the SD imager is currently using X11, and a
   browser called dillo. I wonder if lynx would work.

holta, 2018-08-06:
> Success loading TinyCoreLinux on my MacBook! (just hold down option key during reboot, and select UEFI)

Amazing!

> But I think a one size, pared-down zipped image, less than 80MB, that runs on Windows, rpi, and MacBooks, seems achievable. Is that closer to what we really need?

Yes: let's try to make it as self-documenting as possible in English & Spanish both, so that Mac/Windows/RPi folk can survive &mdash; even when the IT person disappears &mdash; and they literally lose all paper instructions, per 3. below!

> 1. Will not run on older hardware, where Bios will not accept USB as boot device.

I'm not worried at all.  Even in the poorest parts of Haiti/Mexico, there is almost always *someone* in the neighborhood with a more modern Windows laptop, willing to help make Internet-in-a-Box happen for a nearby school.

> 2. Might require internet, and another script to load required dependencies as raspbian packages to run natively on the rpi (they are already part of the package in the Tiny Core image).

Internet is impossible for many folks in the field :-(

So if v1.0 requires Internet, let's try for a v2.0 that does not require Internet later this year ?

> 3. The "help" function for the SD imager is currently using X11, and a browser called dillo. I wonder if lynx would work.

Cool.  The README / can't-go-wrong explanation needs to be Impossible To Miss (a.k.a. unavoidable ;) ...whatever approach is chosen!

holta, 2018-08-07:
@georgejhunt a giant win for Imager v2.0 would be something like "kolibri manage deprovision", as introduced by Kolibri 0.10.0 a month ago:

This removes users' private/personal data and configurations, leaving content intact...thereby broadly enabling ethical content cloning / system duplication of existing schools' installations or remixes.

As outlined/implemented within https://github.com/iiab/iiab/tree/master/roles/kolibri

georgejhunt, 2018-08-07:
> tree/master/roles/kolibri

I think it would be better to make a copy function that does only that. And
create  reset functions that function on the target machine to do the
sanitizing. The function that does the work could be added to rc.local.

holta, 2018-08-11:
> I think it would be better to make a copy function that does only that. And create reset functions that function on the target machine to do the sanitizing. The function that does the work could be added to rc.local.

Obviously a copy is a copy &mdash; but there's a much larger ethical question at stake here, facing us directly:

Promoting a tool that mass-copies kids/patients' personal data quickly becomes unacceptable, if privacy-by-design is not built in, prior to broad promotion & dissemination of such a tool.  For such a tool to be ethical, respected, and broadly adopted, we will need to realign (or certainly repackage!) the defaults with a 180-degree turn towards Privacy-By-Design over the coming year or so.

No major school system or medical system worldwide will tolerate such an invasive (and downright dangerous) tool if the defaults out-of-the-box are not realigned as such.  In Short: we'll need to find ways to clean out private/personal data of others _prior to_ (or during?) the duplication of entire systems, towards facilitating genuinely ethical and collaborative uses among schools/medical clinics/families/etc.

ASIDE: Not to mention many other less-well-intentioned players, that we need not mention here, who _will_ be more than happy to use "mere backups" to harvest/sell/ransom personal/private educational and health/medical data.  Recap: copying enterprise systems in schools and medical establishments is hardly an innocent act in the 21st Century, as it might have been in past decades, when abusable/exploitable personal (meta)data is inherently growing all around us.  Hence the reason it's not only unethical, but increasingly illegal to permit (or even accidentally encourage) exploitation/expropriation of citizen data e.g. in educational environments and medical systems especially.

CONCLUSION: We cannot bury our heads in the "dd is just dd" sand when we want broad adoption (and a clean UX with community support, that will lead us there...)  So let's find a solution co-integrating & co-evolving the critical human needs of all students, patients (and teachers!) worldwide over the coming year or so &mdash; even if it will certainly take time, as conscientious Privacy-by-Design ain't easy!  But certainly our efforts will be wasted if it [Privacy-by-Design] is dropped off the radar and left as a systems design afterthought~

holta, 2018-08-21:
_Many thanks to @nataren who tried this out on MacOS, and provided feedback as follows &mdash; replies from @georgejhunt are inlined below:_

>    Imager (README) feedback
>
>    We need to start with stating the problem statement that the tool helps solve.
>    A visual representation of the problem and the solution, it would be good to clarify things.
>    An items list of required equipment would be great. As in: Computer, USB stick, target SD card.
>    Seems like it is a hard requirement to have the SD cards in a USB compatible adapter. Needs to be documented on requirements section.
>    This Mac machine (not laptop), has an SD card reader, it might not be trying to use it.
>
>    Feedback:
>    First attempt:
>    Plugged in USB stick
>    Booted, and curses based UI showed up
>    I used the up arrows and it dropped me into bash. Why?
>
>    Seconds attempt:
>    I selected the card copy (first option) by just pressing Enter
>    And it displayed a verification error. See attachment.
>
>    General feedback:
>    I was able to see the bash history once dropped there, we probably don’t want to allow that in ‘production’.
>
>    More generally,  George, would you mind explaining why did you:
>
>    1) choose to make a program a bootable program instead of native windows or linux or mac executable?

The choice of a bootable program derived from the following requirements:

   1. Need to shrink an SD card whenever the destination SD is smaller than the source (etcher.io runs on all platforms, and covers the need for same sized copy)
   2. To shrink an ext4 partition, I only know how to do it in linux.
   3. A single program, in this case a bash script, that runs on all platforms, is more supportable long run. 

>    2) choose to implement the application in Bash instead of a cross platform higher level language like Go/Rust/C++?

Bash, on top of UEFI or legacy windows loaders, on top of tiny core linux, seems the simplest path to a menu driven application that runs everywhere. A single program, used everywhere, seems most likely to be bug free in the long run. 

>    I am just trying to understand the goals and motivations of the design.
>
>    Thanks for creating the tool,
>    [@nataren]

I appreciate your comments about help, and explaining the context better.

The idea of running on a mac surfaced late. Initially, [@holta] thought that running on Windows was good enough. I have not started looking at hfs+ drivers, to see if they work.

holta, 2018-08-23:
A huge thank you to Matt J who will be testing IMAGER on Windows in coming days, putting himself in the shoes of an overseas teacher testing http://download.iiab.io/packages/imager/ &mdash; i.e. using a Windows machine to test these 2 most common usage scenarios:

1. microSD-to-microSD card direct IIAB copying (try copying from one brand to another, to see if it works in both directions, when one microSD card is ever-so-slightly-smaller than the other!)

2. a) microSD-to-HDD "backup" of IIAB

    b) shrink the IIAB backup image (just created on the Windows HDD) to become a "master IIAB image" (hopefully auto-renamed elegantly???)

    c) HDD-to-new-microSD (many times if necessary...techie teacher would typically use this to create IIAB clones...for other teachers and nearby schools!)

holta, 2018-08-23:
Prelim feedback, thanks to Matt J:
- He could not figure out which file to download ([install instructions](http://download.iiab.io/packages/imager/README.html) need to be tight, short, and crystal clear for teachers!)
- Download URL needs to be much shorter and more descriptive:
  - Avoid multi-line URL's like http://download.iiab.io/packages/imager/LOCAL-0.1-ghunt-slimmer-180808-1841-g1c39553.img.zip (71MB)
  - Try for http://download.iiab.io/packages/imager-0.1.img or http://download.iiab.io/packages/imager.img shorter! (82MB; avoid zipping which creates headaches for less technical teachers)
- Clarify USB flash drive size requirements!  (e.g. are ancient USB sticks with 1GB or even 100MB sufficient ?)

holta, 2018-08-23:
- Many Windows machines cannot be booted from USB.  But Matt has found two that might work &dash; one is 32-bit, and the other 64-bit.  (I did not know 32-bit Windows 10 even existed!)
- Windows PCs with encrypted filesystems surely cannot work, right?

holta, 2018-08-24:
FYI @georgejhunt release a new version that contains support for "sudo vpn-on" to investigate problems live, as remote users begin to test this:

http://download.iiab.io/packages/imager/LOCAL-0.1-ghunt-openvpn-180823-1911-g008f806.img

holta, 2018-08-24:
@georgejhunt clarifies:

> I think I should unpublish the 500mb version, and spend some time getting a version that boots with vpn enabled out of the box. It may be a few days.

georgejhunt, 2018-08-24:
The next phase of development calls in question some of the recent changes
I made to the code base. In order to reduce the size of the download, and
also to have some hope on integration with MAC OS, I did the following:

   1. Removed openvpn
   2. Removed X11 desktop
   3. Removed all the drivers and firmware for wifi devices (this was
   approx 300MB)

When Adam asks for simple explanations for getting to the terminal, I begin
to question the wisdom of removing the desktop. I think it is more
important to supply an interface that is familiar to main line users than
it is to run on a MAC.

For the next phase, I think debugging a few more hardware configurations is
the highest priority. And GUI, and openvpn on by default, would be a good
interim step. Whether we run on a MAC might be something that waits for the
next release, after we are confident that the program functions with most
of the likely hardware in the target settings.

holta, 2018-08-24:
> simple explanations for getting to the terminal

Is this really nec once OpenVPN is automatic?

Certainly: those who already know Linux / command-line will find it on their own.  Without live coaching, the CLI does nothing for teachers & the general public.  (If however live/phone coaching will be a common use case, then perhaps this is necessary?)  Y/N

holta, 2018-08-24:
Very preliminary: but some of Matt's and my README.md initial suggestions are within #1032, as we all seek better ways for non-technical teachers to support themselves.

holta, 2018-09-10:
Many thanks to @georgejhunt who published IMAGER version 0.2 over recent days here!

http://download.iiab.io/packages/imager/

FYI it's intended mainly for developing world situations where only Windows PC's are available.  As you can see in the new README (click above) a Mac version might be possible in future, if demand truly requires that.  And manual instructions are also included for Linux.

Many Thanks to @nataren and Matt J and others whose [UX] testing keeps helping get this right, one week at time...

holta, 2018-09-12:
Early reactions/questions re: IMAGER 0.2:

1. @floydianslips  asks if can use IMAGER on Linux (ext4 presumably) machines not just Windows machines?

   (What all filesystems does it support, unencrypted NTFS only?)

2. Matt J also spent quite some time with it today, and it caused his Windows Netbook to go into an endless reboot cycle.  He'll try to send more details later.  Very scary even to a technically-proficient experienced operator like him :)

3. Matt J. and @floydianslips both found IMAGER's menus to be frightening for teacher-type operators.  As such people (who just want to copy "microSD A" to "microSD B" in the correct direction!) don't have a clue what a Linux mount point is...and so will be "universally" overwhelmed with any language mentioning "mmcblk" or "sda" or "sdb" or "sdh" or such.

   Unless they have in-person training first.  So: some serious UX (re?)thinking is needed here as, as to how to make adoption real among the senior/rural teachers types who need this most &mdash; _and naturally won't remember how to use it 6 months later, even if they are lucky enough to be trained_ :/

UX conclusions...as we cannot possibly parachute in a technically strong Linux IT helper every single time IMAGER is required...to provide very real moral support as such folks indeed require:

  - there's a very real need to translate the UI/UX into non-technical language.
  - it seems there might very well be a need to HIDE many/most technical details so less-technical users are not overwhelmed...?
  - technical users may prefer command-line tools [min-sd and cp-sd](https://github.com/iiab/iiab-factory/blob/master/box/rpi/howto-mkimg.txt)

holta, 2019-01-02:
@georgejhunt I'm deferring IMAGER refinements to IIAB 7.0 for now &mdash; but please help me keep this in the foreground for any & all NGO's/groups needing a proper arsenal of _rapid-deployment techniques_ as they always do!

holta, 2019-10-21:
Please re-open as necessary, especially if you can contribute, Thanks!

closed 2019-10-21 by holta