iiab/iiab #828: Copy Content Packs to/from USB sticks/storage

holta, 2018-06-06:
@tim-moody is working hard to design this: so educators/operators can copy Content Packs to/from USB memory sticks and other USB-connected storage.

This will likely be a part of the https://github.com/iiab/iiab-admin-console repo.

Alongside @georgejhunt is outlining a way (#827) for schools without modern equipment and IT staff to be able to quickly back up and duplicate entire Internet-in-a-Box microSD cards.

holta, 2018-06-11:
Do IPFS (InterPlanetary File System) or DAT ([Decentralized Archive Transport](https://docs.datproject.org/faq)) and/or http://archive.org have a role to play here?

holta, 2018-09-09:
Somewhat related...IIAB's UX redevelopment plans are now beginning!

This also includes an increasingly more serious look at Sneakernet-of-Alexandria sharing within (and between) offline educational communities in developing countries especially.

@arky & @darkenvy & All please help everyone digest/refine several possible strategies @tim-moody has outlined here:

https://github.com/iiab/iiab-menu/issues/9

holta, 2018-09-25:
FYI https://github.com/iiab/iiab-admin-console/pull/88 and the associated https://github.com/iiab/iiab-admin-console/pull/89 ("Add priority to long running jobs") have been merged to ramp up broader testing.

ericnitschke, 2018-11-22:
I think this functionality will be very useful and important.

A few issues/ideas....

- ZIM file sizes are incorrect
- Had to delete library.xml then re-insert USB to have updated list of ZIMs on USB.
- Possible to change "Copy /media/pi/xyz..." to "Install Selected"?
- Got the error "COPY-ZIMS FAILED and reported Invalid Source or Destination." Ejected and reinserted USB and tried again. Job status says succeeded, but there's no indication of success in Manage Content>Installed Content.   Also no ZIMs show in http://box.lan/kiwix/

ericnitschke, 2018-12-01:
thanks, Tim.  Let me try again next week and forward more details.

holta, 2019-01-08:
Possible expansion of this functionality in 2019, if ZIM files can become more portable/resilient/exchangeable between teachers on USB sticks:

#857 "Auto-display of ZIM files when Teachers insert USB sticks/drives"

(Much as lesson plan ideas and xeroxes are commonly exchanged between teachers today...)

holta, 2019-01-16:
@ericnitschke is IIAB 6.7 (http://d.iiab.io/6.7) looking solid enough for you as we prepare to release in coming days?

For now I'm moving this ticket to [IIAB 7.0's milestone](https://github.com/iiab/iiab/milestone/5) just in case further polish is needed here, Ok?

ericnitschke, 2019-01-21:
All good. In the long run it would be interesting to see ZIMs update between IIAB and mobile devices (IIAB pulls ZIMs from Kiwix mobile app and vice versa).

holta, 2019-03-02:
@tim-moody wrote:
> Probably we will move to allowing the zim external app to be installed from iiab in place of the zim embedded one and provide functionality to download a zim from the server to the device.

Field teachers/operators/students would also LOVE to have more tangible grasp and curatorial control over their own content.  

Usually they are asking to backup/copy content onto one of these 3 devices:

1. FAT32/exFAT USB stick or NTFS USB drives
2. Windows laptop
3. Android phone

Perhaps we should focus on just 1 or 2 of the above 3, thereby enabling genuine participation from "non geek" teachers/field operators in 2019?

In any case, the consistent pattern is that HDD or flash storage space has been pre-purchased in one of the 3 above flavors &mdash; almost always in in a low-income community that cannot purchase additional storage &mdash; and is deathly afraid of reformatting e.g. any of their 3 above devices deviating from their local/cultural norms (allowing immediate "Sneakernet of Alexandria" content interchange, without confusion).

Here are just a few ways (among many others!) IIAB might increasingly deliver on this longstanding goal of putting grassroots TLC's (technology learning coordinators) back in the driver's seat:
- [ ] https://github.com/iiab/iiab/issues/1002#issuecomment-462183401 Field editing of Content Pack descriptions/logos by non-technical operators
- [ ] #1530 Calibre-Web: fix cover/title thumbnails for PDF's and other docs/books
- [ ] #1538 How does a teacher install a ZIM file from Windows to IIAB? (using a FAT/FAT32 USB stick if poss?)
  - [ ] #857 Auto-display of ZIM files when Teachers insert USB sticks/drives

PS @m-anish & @deldesir who use IIAB in more-or-less remote communities should help inform and expand upon these designs!

holta, 2019-07-01:
I've tagged this ticket as "Help Wanted" to invite people to work with @tim-moody trying out "external disks" with IIAB.

Such that a screencast / video can hopefully come together in coming months (or weeks!?) paving the way for new kinds of Sneakernet-of-Alexandria grassroots content exchanges &mdash; e.g. between talented teachers exchanging vital lesson plans proven useful in different remote communities.

Kolibri will surely be a part of this too in coming years...but for now things like https://github.com/holta/iiab-admin-console/blob/master/roles/console/files/help/InstContent.rst#manage-content are probably the place to start?!

holta, 2019-07-15:
Screencasts/videos are needed here, perhaps in the month after IIAB 7.0 is released in July 2019 hopefully?

I'm moving this tkt to the [IIAB 7.1 Milestone](https://github.com/iiab/iiab/milestone/6) so that UX / usability (actionable, modern, video documentation) remains our focus in the IIAB 7.1 release cycle.

jvonau, 2021-07-21:
basic framework in place close

closed 2021-07-21 by holta