iiab/iiab #829: is Spanish WikiHow's search fixable? (works on RACHEL site!)

holta, 2018-06-06:
Online Version: See the top-right of http://oer2go.org/mods/es-wikihow/content/Portada.html where it DOES work.

Offline Version: See the top-right of http:// iiab.me /modules/es-wikihow/content/Portada.html where it does NOT work.  Instead it shows: "Resultados de la búsqueda - wikiHow ES".  CLARIF: Autocomplete always works (in the search box, top-right of the uppermost page above) but the search itself does not.

Peru schools (and others) very much want this fixed, if that's possible!  @georgejhunt & @tim-moody do you have ideas?

holta, 2018-06-14:
Search Error is:

> how/content/search/search.php:72\nStack trace:\n#0 {main}\n  thrown in /library/www/html/modules/es-wikihow/content/search/search.php on line 72, referer: http://iiab.me/modules/es-wikihow/content/Categoria%3AArte-y-entretenimiento.html

> [Thu Jun 14 12:03:29.077326 2018] [php7:error] [pid 2639] [client 192.168.0.56:52294] PHP Fatal error:  Uncaught Error: Call to undefined function stem() in /library/www/html/modules/es-wikihow/content/search/search.php:72\nStack trace:\n#0 {main}\n  thrown in /library/www/html/modules/es-wikihow/content/search/search.php on line 72, referer: http://iiab.me/modules/es-wikihow/content/Categoria%3AVida-familiar.html

holta, 2018-06-14:
@georgejhunt & @tim-moody are investigating if locale or a PECL / PEAR issue might be the cause?  

Is a word stemming pkg like https://github.com/marc-mabe/php-stem missing?

holta, 2018-06-14:
@tim-moody observes that search work on PHP 5 sites like http://oer2go.org/mods/es-wikihow/content/Portada.html

Can we make this (php-stem etc, originally from Google) happen on PHP 7.x, if we can build it?  (There is discussion online that PHP 7 compatibility has been restored...so there is hope!)

holta, 2018-06-14:
Ongoing/Related Links from @tim-moody and @georgejhunt:

https://bugs.php.net/patch-display.php?bug_id=71631&patch=update-for-php7&revision=latest
http://svn.php.net/viewvc/pecl/stem/
https://github.com/rachelproject/rachelplus/issues/51

holta, 2018-06-14:
@tim-moody  appears to have fixed this on RPi, by compiling stem.so

Thanks Tim for publishing stem.so if poss?

holta, 2018-06-14:
> @tim-moody appears to have fixed this on RPi, by compiling stem.so
>
> Thanks Tim for publishing stem.so & instructions if poss?

Plz test, those who can:
http://download.iiab.io/packages/php-stem.x64.tar
http://download.iiab.io/packages/php-stem.rpi.tar
http://download.iiab.io/packages/php-stem.src.tar

holta, 2018-06-14:
> @tim-moody clarifies: "put the tar in / (not /root) and untar"

Addendum: reboot or run `systemctl restart apache2`

holta, 2018-06-14:
@tim-moody please close this issue if you have a plan to upstream this fix to RACHEL?

georgejhunt, 2018-06-14:
I don't think this is a rachel problem

holta, 2018-06-14:
> I don't think this is a rachel problem

Ok.  How do we address it on our end, within IIAB?

So IIAB implementers can install http://oer2go.org/viewmod/es-wikihow with search (of these 55,000 articles).

holta, 2018-07-19:
@tim-moody untar'ing the 3 files in http://download.iiab.io/packages/php-stem.x64.tar to `/` and then restarting Apache does not work on Ubuntu 18.04 &mdash; when following the analogous routine that works on Raspbian with http://download.iiab.io/packages/php-stem.rpi.tar

The search at the top of http://box/modules/es-wikihow/content/Portada.html does not work on Ubuntu 18.04, regardless whether the untar is done manually (per https://github.com/iiab/iiab/issues/829#issuecomment-397442078) or automatically (per #925, using [php-stem.yml](https://github.com/iiab/iiab/blob/master/roles/httpd/tasks/php-stem.yml#L24-L32)).

So after untar'ing to Ubuntu 18.04 I tried the following...which also did not work:
- cp -p /etc/php/7.0/mods-available/stem.ini /etc/php/7.2/mods-available/
- ln -s /etc/php/7.2/mods-available/stem.ini /etc/php/7.2/apache2/conf.d/20-stem.ini
- systemctl restart apache2

In both above attempts, the permissions seem correct:

>root@box:~# ls -l /etc/php/7.?/mods-available/stem.ini
> -rw-r--r-- 1 root root 70 Jun 14 15:54 /etc/php/7.0/mods-available/stem.ini
> -rw-r--r-- 1 root root 70 Jun 14 15:54 /etc/php/7.2/mods-available/stem.ini
>
> root@box:~# ls -l /etc/php/7.?/apache2/conf.d/20-stem.ini
> lrwxrwxrwx 1 root root 36 Jun 14 15:55 /etc/php/7.0/apache2/conf.d/20-stem.ini -> /etc/php/7.0/mods-available/stem.ini
> lrwxrwxrwx 1 root root 36 Jul 19 01:10 /etc/php/7.2/apache2/conf.d/20-stem.ini -> /etc/php/7.2/mods-available/stem.ini

Finally I tried the following...which also did not work:
- rm /etc/php/7.0/mods-available/stem.ini /etc/php/7.0/apache2/conf.d/20-stem.ini
- systemctl restart apache2

Ideas?

holta, 2018-07-23:
@tim-moody & georgejhunt if http://download.iiab.io/packages/php-stem.x64.tar is not fixable for IIAB 6.6 (?) then I'll document this as a known issue & let's see if a recompilation is possible for IIAB 6.7

tim-moody, 2018-07-24:
on u18 (with php 7.2) I think the files should be:

/usr/lib/php/20170718/stem.so
/etc/php/7.2/mods-available/stem.ini
/etc/php/7.2/apache2/conf.d/20-stem.ini -> /etc/php/7.2/mods-available/stem.ini

php -i | grep extension_dir

might be a basis for softcoding the 20170718 part

tim-moody, 2018-07-24:
php -i | grep 'extension_dir => /usr/lib' | cut -d " " -f 3

holta, 2018-07-24:
> on u18 (with php 7.2) I think the files should be:
>
> /usr/lib/php/20170718/stem.so
> /etc/php/7.2/mods-available/stem.ini
> /etc/php/7.2/apache2/conf.d/20-stem.ini -> /etc/php/7.2/mods-available/stem.ini

Hmm the above did not work on my Ubuntu 18.04 :(

FWIW adding/removing these 2 (below) also did not help:

- /etc/php/7.0/mods-available/stem.ini
- /etc/php/7.0/apache2/conf.d/20-stem.ini -> /etc/php/7.0/mods-available/stem.ini

(Clarif: `systemctl restart apache2` was used in each case, prior to searching on top of http://box/modules/es-wikihow/content/Portada.html ...rebooting also did not work.)

tim-moody, 2018-07-24:
what does  
php -i |grep extension_dir
 show?
what version of php?

holta, 2018-07-24:
> what does
> php -i |grep extension_dir
> show?

root@box:~# php -i | grep extension_dir
extension_dir => /usr/lib/php/20170718 => /usr/lib/php/20170718
sqlite3.extension_dir => no value => no value

> what version of php?

root@box:~# apt -a list php
Listing... Done
php/bionic,bionic,now 1:7.2+60ubuntu1 all [installed]

(Both above being Ubuntu 18.04 standards, as far as I can tell anyway.)

holta, 2018-07-27:
On Ubuntu 18.04 all 3 files seem to be properly in place but /var/log/apache2/error.log showed:

> http://10.8.0.26/modules/es-wikihow/content/search/search.php?query=enamorar&sa=Buscar error pasted in from /var/log/apache2/error.log :
>
> [Fri Jul 27 10:30:41.338852 2018] [php7:error] [pid 19141] [client 10.8.0.66:52316] PHP Fatal error:  Uncaught Error: Call to undefined function stem() in /library/www/html/modules/es-wikihow/content/search/search.php:72\nStack trace:\n#0 {main}\n  thrown in /library/www/html/modules/es-wikihow/content/search/search.php on line 72, referer: http://10.8.0.26/modules/es-wikihow/content/Portada.html

On Ubuntu 18.04, http://10.8.0.26/info.php fails to show a section for `stem`

On Raspbian, http://10.8.0.10/info.php does show a section for `stem`
```
stem support | enabled
-- | --
version | 2.0.0-dev

"Original Porter ISO-8859-1" | enabled
"Danish UTF-8" | enabled
"Dutch UTF-8" | enabled
"English UTF-8" | enabled
"Finnish UTF-8" | enabled
"French UTF-8" | enabled
"German UTF-8" | enabled
"Hungarian UTF-8" | enabled
"Italian UTF-8" | enabled
"Norwegian UTF-8" | enabled
"Portuguese UTF-8" | enabled
"Romanian UTF-8" | enabled
"Russian UTF-8" | enabled
"Russian UTF-8 (old unicode)" | enabled
"Spanish UTF-8" | enabled
"Swedish UTF-8" | enabled
"Turkish UTF-8" | enabled
"Turkish UTF-8 (old unicode)" | enabled
```

holta, 2018-07-27:
I noticed that both these files have 755 permissions (including executable bits) when they should have 644:

- 10.8.0.10:/usr/lib/php/20151012/stem.so (RPi)
- 10.8.0.26:/usr/lib/php/20170718/stem.so (Ubuntu 18.04)

(Removing the executable bits and then running `systemctl restart apache2` does not solve the situation however.)

holta, 2018-08-15:
Let's defer to IIAB 6.7 and close as necessary (if a volunteer is not found to compile support for Ubuntu 18.04?)

jvonau, 2020-04-29:
#2349

PR #925 (2018-07-18): php-stem fixes search @ http://box/modules/es-wikihow
Fixes #829

@tim-moody @georgejhunt @jvonau please confirm what permissions the unarchive command's `mode: ????` should use for these 3 files:
```
/etc/php/7.0/mods-available/stem.ini
/usr/lib/php/20151012/stem.so
/etc/php/7.0/apache2/conf.d/20-stem.ini
```
Thanks!
closed 2020-04-29 by holta