iiab/iiab #835: Pi zero not working with iiab

Piflyer, 2018-06-12:
### Expected behavior and actual behavior.
To have it boot up to Raspbian with iiab in it
### Steps to reproduce the problem.
Download on a Raspberry Pi 3 B and transfer it to a Pi zero. It was in ‘panic mode’ and I lot would kill init.

tim-moody, 2018-06-12:
I do this regularly without any problems, but generally with small installs. Also, I have not yet tried 3 B+.

holta, 2018-06-12:
@kananigit can you please comment on your own experience today building up IIAB 6.5 within RPi 3 B+ to create microSD's that you then insert into RPi Zero W?

PS this has worked reliably for me in recent months.  I would like @Piflyer to provide photos of the error(s) in question, excerpt from systems logs (using the journalctl command) and context as to what exact local_vars.yml he/she is using from http://download.iiab.io/6.5

Piflyer, 2018-06-12:
[image]

Piflyer, 2018-06-12:
It said, `[ 1.187737] Kernel panic - not syncing: Atyempted to kill init! exit code 0x0000000b`

holta, 2018-06-13:
@jvonau : somehow @Piflyer 's screen photo from yesterday isn't posted above, so am re-posting — any ideas?

[image]

tim-moody, 2018-06-13:
I guess we need local_vars.yml to see what might be installed to cause this. Is this during fsck?

holta, 2018-06-13:
>  we need local_vars.yml to see what might be installed to cause this

Yes: @Piflyer please post your /opt/iiab/iiab/vars/local_vars.yml

Clarif: this is usually derived from _{min, medium, or big}_ local_vars.yml here...

http://wiki.laptop.org/go/IIAB/local_vars.yml

holta, 2018-06-13:
@Piflyer RE-confirm whether you've prepared the microSD in a:
- 3 B (RPi released 2016-02-29)
or a...
- 3 B+ (RPi released 2018-03-14)

Thanks!

holta, 2018-06-13:
Context:
* 3 B WORKS for me, to prepare IIAB microSD's for Zero W.
* 3 B+ does NOT work for me, to prepare IIAB microSD's for Zero W.

jvonau, 2018-06-13:
Can you post the contents of cmdline.txt on the first partition of the SDcards there were prepared on the 3B and the 3B+ please...

holta, 2018-06-13:
> Can you post the contents of cmdline.txt on the first partition of the SDcards there were prepared on the 3B and the 3B+ please...

microSD prepared on **3 B** has /boot/cmdline.txt :

> dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=8e908af0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

microSD prepared on **3 B+** has /boot/cmdline.txt :

> dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=05f17ccc-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

holta, 2018-06-13:
@jvonau observes: "seems like the wrong overlay is being loaded... I don't have time till the week end to investigate further.."

holta, 2018-06-13:
Clarif: adding " quiet splash plymouth.ignore-serial-consoles" to the end of /boot/cmdline.txt (as prepared on 3 B+) is _not_ sufficient to get it to boot in a Zero W.

holta, 2018-06-13:
1) Unusual Result: @tim-moody indicates that IIAB microSD's prepared with 3 B+ *can successfully* be run in 3 B, even if we haven't found a way to get them to run in Zero W.

   This is unexpected as 3 B and Zero W are so similar (same WiFi chip anyway!)

2) @jvonau suggests we try prep'ing IIAB images on Raspbian 2018-04-18 without "apt update" and without "apt dist-upgrade" (using a 3 B+) to see if those might possibly work in Zero W???

kananigit, 2018-06-13:
Interesting, i prepared my image in a 3B and used medium local vars, and i have the same exact problem piflyer has [end Kernel panic -not syncing: Attempted to kill init! exitcode=0x0000000b]

In the medium local vars i only made a few changes, turn off elgg install and turn off ka lite install

holta, 2018-06-13:
> i prepared my image in a 3B and used medium local vars, and i have the same exact problem

Did you ever run the microSD inside a 3 B+, prior to inserting it into a Zero W?

kananigit, 2018-06-13:
Not at all, don't even own a 3 B+. Just pulled it out of a 3B and inserted it to a pi zero w

holta, 2018-06-13:
> Not at all, don't even own a 3 B+. Just pulled it out of a 3B and inserted it to a pi zero w

@kananigit what exact version of Raspbian are you using?  Did you run "apt update" and "apt dist-upgrade" by hand or within a script from http://download.iiab.io/6.5 ?

Both Raspbian Lite 2018-04-18 and Raspbian Desktop 2018-04-18 worked for me inside Zero W &mdash; I used a RPi 3 B (NOT B+) to install IIAB 6.5 on May 22nd (based on a [medium-sized local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml)) using http://download.iiab.io/6.5/load.txt (which ran "apt update" and "apt -y dist-upgrade").

Both working copies of /boot/cmdline.txt are below if it helps others.

On Raspbian Lite, /boot/cmdline.txt is:

> dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=36d137ba-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

On Raspbian Desktop, /boot/cmdline.txt is:

> dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=8e908af0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

kananigit, 2018-06-13:
I  ran 2018-04-18 raspbian lite.  I ran  "apt update" and "apt dist-upgrade"  via the script.

Piflyer, 2018-06-13:
I downloaded all the content on a RPi 3 B, run it, put it in a 3 B+, and currently runs iiab. The first time, however, the 3 B+ couldn't boot the sd card correctly, and was stuck on the splash screen. So I downloaded it onto the 3 B, and worked flawlessly. After the download, the 3 B+ was able to reconize the sd card, and boot up to Raspbian with no problem. However, any time I boot up in the 3 B, or 3 B+, it said that sugarizer failed to work, but iiab still worked. Hopefully this can help further explain how Nextcloud takes forever to login, and some apps don't load after I reboot my Pi? `# This is local_vars_big.yml -- copy it to local_vars.yml then...

# Put variables herein to override /opt/iiab/iiab/vars/default_vars.yml

# PLEASE READ http://wiki.iiab.io/local_vars.yml

# Orig Idea: branch github.com/xsce/xsce-local for your deployment/community

# Ansible's default timeout for "get_url:" downloads (10 seconds) often fails
download_timeout: 200

# Users and Passwords

iiab_admin_user: iiab-admin
# Obtain a password hash with:
#    python -c 'import crypt; print crypt.crypt("<plaintext>", "$6$<salt>")'
# iiab_admin_passw_hash:
admin_install: True

# Set admin_install: False if you don't want iiab_admin_user & wheel group
# auto-created in roles/iiab-admin/tasks/main.yml, thereby disabling sudo-based
# warnings on use of published passwords like pi/raspberry & iiab-admin/g0adm1n

# If admin_install: False, set iiab_admin_user (above) to an existing Linux
# user that has sudo access, so you can login to Admin Console http://box/admin

iiab_hostname: box
iiab_domain: lan

# Set to /home or /wordpress or /mediawiki or /wiki (for DokuWiki)
iiab_home_url: /home

# Raspbian requires WiFi country since March 2018.  Please set it here:
host_country_code: US
host_ssid: "Internet in a Box"
host_wifi_mode: g
host_channel: 6
hostapd_secure: False
hostapd_password: changeme

dns_jail_enabled: False

# Enables "campus access" to kiwix (3000), kalite (8008) & calibre (8010 or
# 8080) on WAN side of server. See network/templates/gateway/iiab-gen-iptables
# within github.com/iiab/iiab/blob/master/roles/
services_externally_visible: True

# Make this True if client machines should have access to WAN/Internet:
iiab_gateway_enabled: False

# Make this False to disable http://box/common/services/power_off.php button:
allow_apache_sudo: True

# Stages 3 & 4 must be run (using iiab-install or runrole) if changing these:
squid_install: True
squid_enabled: True

dansguardian_install: True
dansguardian_enabled: True

# Unmaintained as of October 2017: https://github.com/iiab/iiab/pull/382
# wondershaper_install: False
# wondershaper_enabled: False

# 1-PREP

# 2-COMMON

# 3-BASE-SERVER

[... 127 lines cut ...]
phpmyadmin_enabled: False

# Unmaintained (better to install from http://teamviewer.com)
teamviewer_install: False
teamviewer_enabled: False

vnstat_install: True
vnstat_enabled: True

# Unmaintained
# sugar_stats_install: False
# sugar_stats_enabled: False

# Unmaintained
# xovis_install: False
# xovis_enabled: False

# Unmaintained
# schooltool_install: False
# schooltool_enabled: False

# Unmaintained
# debian_schooltool_install: False
# debian_schooltool_enabled: False
`

holta, 2018-06-13:
FYI inserting these 2 (working) IIAB 6.5 microSD cards into a 3 B+ and then running "apt update; apt dist-upgrade" does not harm them.  Just for good measure I also rebooted these microSD's within a 3 B+.  (I cannot cause them to fail, as much as I try!)

They (IIAB 6.5 based on Raspbian Lite and Raspbian Desktop, originally prepared in a 3 B) continue to work flawlessly in all 3 kinds of RPi, _even after all updates are applied:_
- Zero W
- 3 B
- 3 B+_

Conclusion: This Is A Solvable Problem Even If We Don't Understand It Just Yet :)

Piflyer, 2018-06-13:
After that, it still didn't work.

holta, 2018-06-14:
I too have now seen microSD's fail to boot in Zero W even when they were prepared on original RPi 3 B's.

New Theory: are "apt update" and "apt dist-upgrade" SOMETIMES causing this failure in recent weeks?

(Even while "apt update" and "apt dist-upgrade" do NOT cause the failure when applied to my working microSD's from late May!)

Piflyer, 2018-06-14:
I agree, maybe it is. I remember doing the airplay script after the update, and it didn't work.

Piflyer, 2018-06-14:
But it still doesn't work before hand.

holta, 2018-06-14:
@georgejhunt and I have confirmed that [MIN-sized](http://download.iiab.io/6.5/load-min.txt) IIAB microSD's work in Zero W.

This is astonishing as [MEDIUM-sized](http://download.iiab.io/6.5/load.txt) microSD's fail in the Zero W, using the same physical hardware (same microSD, and same RPi 3 B to prepare it).

So hopefully folks can help us bissect where MEDIUM-sized installations of IIAB as going off the rails (for Zero W, unlike ~3 weeks ago when this worked!!)

Piflyer, 2018-06-14:
Definitely large size like mines would not work, but it confuses me how many other programs work but this doesn’t?

holta, 2018-06-15:
@georgejhunt & others please continue to help me bissect.

I've reconfirmed that *upgrading* a MIN-sized IIAB 6.5 to a MEDIUM-sized IIAB 6.5 (by swapping in a new /opt/iiab/iiab/vars/local_vars.yml and then running "./iiab-install --reinstall") destroys the microSD card's ability to boot in a Zero W (unlike 3 weeks ago when this worked!)

So now at least it's a straightfoward matter of analyzing the difference between these 2:
- http://wiki.laptop.org/go/IIAB/local_vars_min.yml =
   https://github.com/iiab/iiab/blob/master/vars/local_vars_min.yml
- http://wiki.laptop.org/go/IIAB/local_vars.yml =
   https://github.com/iiab/iiab/blob/master/vars/local_vars_medium.yml

holta, 2018-06-15:
If it helps everyone bisect (isolate) the root cause, please start with a MIN-sized IIAB and try adding each one of these packages using ./runrole [PKG NAME] until the microSD is no longer bootable in Zero W:
```
root@box:/opt/iiab/iiab/vars# diff local_vars_min.yml local_vars_medium.yml     1c1
< # This is local_vars_min.yml -- copy it to local_vars.yml then...
---
> # This is local_vars_medium.yml -- copy it to local_vars.yml then...
94c94
< cups_install: False
---
> cups_install: True
127,128c127,128
< calibre_install: False
< calibre_enabled: False
---
> calibre_install: True
> calibre_enabled: True
146,147c146,147
< elgg_install: False
< elgg_enabled: False
---
> elgg_install: True
> elgg_enabled: True
152,153c152,153
< nextcloud_install: False
< nextcloud_enabled: False
---
> nextcloud_install: True
> nextcloud_enabled: True
155,156c155,156
< wordpress_install: False
< wordpress_enabled: False
---
> wordpress_install: True
> wordpress_enabled: True
172,173c172,173
< osm_install: False
< osm_enabled: False
---
> osm_install: True
> osm_enabled: True
180,181c180,181
< sugarizer_install: False
< sugarizer_enabled: False
---
> sugarizer_install: True
> sugarizer_enabled: True
```

holta, 2018-06-15:
Just a historical reference to #661 "IIAB 6.5 microSD's cannot be moved from RPi3 to RPi Zero W?" which was resolved around March 2018 &mdash; symptoms are almost exactly the same a few months later &mdash; but the root cause appears very different this time around ?

holta, 2018-06-16:
INSTALL WORKAROUND, if you need your microSD to work in RPi Zero W: fix both these [2 lines in local_vars.yml (approx Lines 127-128)](https://github.com/iiab/iiab/blob/master/vars/local_vars_medium.yml#L127-L128) to False, in order to prevent installation of Calibre:
```
calibre_install: False
calibre_enable: False
```

CLARIFICATION: an "antidote/immunization" for Zero W is clearly possible (as I did in late May) resulting in complete microSD cards that work in Zero W, but is not easy to reproduce.

i.e. What Did Work was installing Calibre 3.23 (using [scripts/calibre-install-latest-rpi.sh](https://github.com/iiab/iiab/blob/master/scripts/calibre-install-latest-rpi.sh)) PRIOR to applying Debian .debs (using [scripts/calibre-install-latest.sh](https://github.com/iiab/iiab/blob/master/scripts/calibre-install-latest.sh)) to upgrade to 3.24.2 or 3.25.

However this "magical approach" is no longer viable since 3.23 seems to have been removed from sites holding Raspbian .debs, i.e. sites like http://raspbian.raspberrypi.org/raspbian/pool/main/c/calibre/ and http://archive.raspbian.org/raspbian/pool/main/c/calibre/ :(

holta, 2018-06-16:
Possibly Relevant:

Calibre 3.26.1 was released yesterday at https://calibre-ebook.com/whats-new

Whether or not it's Raspbian .deb installers proves workable at http://raspbian.raspberrypi.org/raspbian/pool/main/c/calibre/ (as was the case up to 3.23).

holta, 2018-06-17:
CLARIF: https://github.com/iiab/iiab/pull/833 is the immediate root cause.

@jvonay suggests we try installing via calibre-install-latest-rpi.sh first, an then calibre-install-latest.sh

Another approach (for now, demonstrated to work) would be to install Calibre 3.23 from RPi .deb's (if we can recover those) and then upgrade using calibre-install-latest.sh

holta, 2018-06-17:
> @jvonau suggests we try installing via calibre-install-latest-rpi.sh first, and then calibre-install-latest.sh

Oops the above is unlikely to work, given the failures @kananigit and I experienced 10 days ago (calibre-install-latest-rpi.sh does not complete since Calibre 3.24.2).

> Another approach (for now, demonstrated to work) would be to install Calibre 3.23 from RPi .deb's (if we can recover those) and then upgrade using calibre-install-latest.sh

Has more hope?

Piflyer, 2018-06-17:
I'm starting to feel like it's a bug.

holta, 2018-06-18:
Those who care about the Zero W need to take action, working conscientiously with those upstream to generate/identify a version of Calibre that installs on Raspbian, starting in place like this:

https://www.mobileread.com/forums/showthread.php?p=3707933

Reminder: those who don't care about Zero W have a workaround/solution today (https://github.com/iiab/iiab/blob/master/scripts/calibre-install-latest.sh).

holta, 2018-06-19:
Ref:
PR #840 Proposed IIAB 6.5.1: restoring Calibre, Zero W runnability (PR #839), Fixed IP (PR#837)

holta, 2018-06-19:
While not really a bug (Calibre is not expected to run on RPi Zero W, given its 512MB limitation) PR #839 now allows Calibre 3.25 to install as part of IIAB in a more flexible way, permitting such microSD's to at least boot in Zero W computers.

holta, 2018-06-19:
Clarif: PR #839 has been merged into IIAB 6.6/master, and its equivalent will likely be merged into IIAB 6.5.1 in the coming week.

closed 2018-06-19 by holta