iiab/iiab #841: Kolibri playbook for IIAB, from Learning Equality?

holta, 2018-06-19:
@ivanistheone began working on this, according to @tim-moody.

What ports does Kolibri prefer?

Could Internet-in-a-Box install both (Kolibri and KA Lite) in the early days?  (Or is that a very bad idea?)

holta, 2018-06-30:
> What ports does Kolibri prefer?

@ivanistheone suggests he will use port 8009, how does this sound @tim-moody ?

arky, 2018-07-09:
@holta @tim-moody I have similar request from Cambodia. Let me pitch in and help with this ansible role.

holta, 2018-07-09:
> @holta @tim-moody I have similar request from Cambodia. Let me pitch in and help with this ansible role.

Thanks Arky!  Note @ivanistheone is traveling during much of July, and so will not be able to help until later.

arky, 2018-07-09:
@holta Quickly hacked together a role for kolibri. Will continue working on it when I have time.

ivanistheone, 2018-07-09:
@arky [Good stuff](https://github.com/arky/iiab/commit/373d14d10c2490206620b2260ca7febe193a49ed)... I was going to try `.deb` approach, but I see `pip` approach works too. Note you'll need to set the `KOLIBRI_HTTP_PORT` env var otherwise kolibri will run on `8080`.

I'll try to publish my playbook as soon as I get it to work (testing on debian first in a VM, but want it to work on Rπ3B).

arky, 2018-07-09:
Thanks @ivanistheone I think using pip is ideal choice for django applications.  Is there a way to load configuration settings from a file instead of passing them as environment variables?

holta, 2018-07-09:
FWIW the `.deb` approach (though not without its own risks) helps cultivate ongoing participation i.e. has fostered more ongoing community "eyes on the prize" QA over the years.

arky, 2018-07-10:
@ivanistheone @holta Please kindly take a peek and give some feedback. Thanks! 

https://github.com/arky/iiab/tree/kolibri

holta, 2018-07-10:
@arky does it work with Kolibri 0.10.0 on Raspbian and/or Ubuntu 18.04 do you know?

https://github.com/arky/iiab/blob/kolibri/roles/kolibri/tasks/main.yml#L13 should prob be `state: latest` ?

Consider writing up upgrade/reinstall recommendations for implementers who depend on http://FAQ.IIAB.IO Item 23 ("Can I upgrade or reinstall server apps?")   Sometimes (re)moving a central trigger file or central configuration file is best, to indicate that a complete reinstall or update (i.e. of Kolibri) should/can safely go ahead the next time you the implementer/operator runs `./runrole kolibri` or `./iiab-install --reinstall`

PS I don't understand the Django issues unfortunately!  But I'm glad to see Learning Equality now officially supports updates (for those intermittently online, whether we use this or not) which they recommend happen using:
```
apt install software-properties-common python-software-properties
add-apt-repository ppa:learningequality/kolibri
apt update
apt install kolibri
```

arky, 2018-07-10:
Smoke-test results on Ubuntu 9 

`***************************************************
2018-07-10 07:34:09,401 p=7990 u=root |  skipping: [127.0.0.1]
2018-07-10 07:34:09,450 p=7990 u=root |  TASK [0-init : STAGE 0 HAS COMPLETED ======================================] ***********************************************************************************
2018-07-10 07:34:09,496 p=7990 u=root |  skipping: [127.0.0.1] => (item={u'option': u'is_VM', u'value': u'yes'}) 
2018-07-10 07:34:09,545 p=7990 u=root |  TASK [kolibri : Create Kolibri folder to store data and configuration files.] **********************************************************************************
2018-07-10 07:34:09,896 p=7990 u=root |  changed: [127.0.0.1] => (item=/library/kolibri)
2018-07-10 07:34:09,945 p=7990 u=root |  TASK [kolibri : Install kolibri using pip on all OS's] *********************************************************************************************************
2018-07-10 07:36:38,075 p=7990 u=root |  changed: [127.0.0.1]
2018-07-10 07:36:43,287 p=7990 u=root |  TASK [kolibri : Create kolibri systemd file] *******************************************************************************************************************
2018-07-10 07:36:53,032 p=7990 u=root |  changed: [127.0.0.1] => (item={u'dest': u'/etc/systemd/system/kolibri.service', u'src': u'kolibri.service.j2', u'mode': u'0644'})
2018-07-10 07:36:53,115 p=7990 u=root |  TASK [kolibri : Enable kolibri service] ************************************************************************************************************************
2018-07-10 07:37:25,891 p=7990 u=root |  changed: [127.0.0.1] => (item={u'name': u'kolibri'})
2018-07-10 07:37:25,943 p=7990 u=root |  TASK [kolibri : Disable kolibri service] ***********************************************************************************************************************
2018-07-10 07:37:26,051 p=7990 u=root |  skipping: [127.0.0.1] => (item={u'name': u'kolibri'}) 
2018-07-10 07:37:26,103 p=7990 u=root |  TASK [kolibri : Add 'kolibri' to list of services at /etc/iiab/iiab.ini] ***************************************************************************************
2018-07-10 07:37:26,873 p=7990 u=root |  changed: [127.0.0.1] => (item={u'option': u'name', u'value': u'kolibri'})
2018-07-10 07:37:27,226 p=7990 u=root |  changed: [127.0.0.1] => (item={u'option': u'description', u'value': u'"Kolibri is an open-source educational platform specially designed to provide offline access to a wide range of quality, openly licensed educational contents in low-resource contexts like rural schools, refugee camps, orphanages, and also in non-formal school programs."'})
2018-07-10 07:37:27,534 p=7990 u=root |  changed: [127.0.0.1] => (item={u'option': u'enabled', u'value': True})
2018-07-10 07:37:27,547 p=7990 u=root |  PLAY RECAP *****************************************************************************************************************************************************
2018-07-10 07:37:27,552 p=7990 u=root |  127.0.0.1                  : ok=27   changed=10   unreachable=0    failed=0   `

holta, 2018-07-12:
> > https://github.com/arky/iiab/blob/kolibri/roles/kolibri/tasks/main.yml#L39-L41
>
> @holta Is there a good way to format multiline ansible commands?

Please read about the difference between `>` and `|` to fold together multiple lines:
https://stackoverflow.com/questions/40230184/how-to-do-multiline-shell-script-in-ansible

Supposedly `|` (merges lines with "\n") is preferred for multi-line shell scripts.

But if you prefer to merge lines with " " then use `|`.

ivanistheone, 2018-07-13:
Hi all,

I think @arky has a good grasp on the setup/install procedure so there is not much I can contribute at this point. Also I will be mostly offline for the next few weeks so can't github effectively. When I come back I can contribute testing and docs for the playbook, but overall you guys seem to be on top of things.

Some comments and tips:

  - @arky you might want to add `--verbosity 0` to the [provisiondevice command](https://github.com/learningequality/kolibri/pull/3775/files#diff-89dfc299b6380a7f7cc9dc159241fb8aR149) to make it less chatty

  - Another thing you might want to do is set create a var `channels_to_import` and then run [`importchannel` and `importcontent`](https://github.com/learningequality/kolibri/pull/3775/files#diff-89dfc299b6380a7f7cc9dc159241fb8aR161) during the install phase. This would take a long time, but might be useful for fully automated setup.

  - There is also a [`language setdefault` command](https://github.com/learningequality/kolibri/pull/3775/files#diff-89dfc299b6380a7f7cc9dc159241fb8aR231), which I always do before fist starting Kolibri, but it doesn't do much since Kolibri respects the connecting browser's language preferences.

  - For the `.deb` install approach, I haven't tested recently but [these instructions](https://github.com/learningequality/kolibri-docs/blob/develop/docs/install/_rpi.rst#install) should work.

  - Last but not least, you should consider making apache serve the static files found at  ``{{KOLIBRI_HOME}}/content/storage`` under the path ``/content/storage``. Here is [an example nginx config file](https://github.com/learningequality/cloud-kolibri-demo/blob/master/config/nginx_site.template.conf#L13) that you can adapt to apache syntax. This will boost the performance significantly for deployments where many kids are connecting simultaneously since it cuts out Python out of the picture completely.

holta, 2018-07-27:
Moving remaining tasks to #977

closed 2018-07-27 by holta