iiab/iiab #842: New Logo Discussion

holta, 2018-06-19:
Very unfortunately GitHub deleted the prior ticket, artwork and discussion (these had been at https://github.com/iiab/iiab/issues/789) so here is a new ticket to relaunch the invitation for new logo design proposals.

gcarlosr, 2018-07-15:
Hi I have created this entry to place some design proposals for your project logo, I would like to contribute, I am a designer and I make logo design contributions to apps and repo open source.

gcarlosr, 2018-07-18:
Here something of what they had seen in the previous discussion but I remain to decide what would be the pro I hope your comments.
[image]

holta, 2018-07-18:
Thanks you so much Carlos for persisting!

FYI the correct spelling has `a` not `an` and appears as:
- "Internet-in-a-Box" (when in a sentence)
- "Internet in a Box" (when not in a larger sentence)

gcarlosr, 2018-07-18:
Hi @holta  ok here is updated.
[image]

arky, 2018-08-31:
Adding this as task list for #1055