iiab/iiab #857: Auto-display of ZIM files when Teachers insert USB sticks/drives

holta, 2018-06-28:
@tim-moody: expanding on [Line 7 of /etc/systemd/system/kiwix-serve](https://github.com/iiab/iiab/blob/master/roles/kiwix/templates/kiwix-serve.service.j2#L7) @jvonau, @deldesir and I are curious if you think the following might work:

> kiwix-serve [...] --library /library/zims/library.xml /library/www/html/local_content/USB*/library/www/html/*.zim

Another equally effective could be:

> kiwix-serve [...] --library /library/zims/library.xml /media/usb*/library/www/html/*.zim

...to load **temporary ZIM files** from teachers' USB sticks, after "systemctl restart kiwix-serve" [would be] triggered by [usb-lib](https://github.com/iiab/iiab/tree/master/roles/usb-lib) of course, upon insertion (and upon removal?) of any USB media.

Another option might be to try to auto-display ALL ZIM files on inserted USB sticks/drives.

Either Way: building upon teachers' requests to have more tactile/immediate control of their own content &mdash; directly in their own hands &mdash; as we've begun to offer here:

> http://FAQ.IIAB.IO Item 4 = _"Can teachers display their own content?"_

holta, 2018-06-28:
I did not address auto-display of OER2GO/RACHEL modules on inserted USB media.

Perhaps I'm wrong (?) but my assumption was that these are already very easily visitable by students when a teacher inserts her USB stick/drive (using http://box/usb) and so no additional logic is required.

holta, 2018-06-28:
Related:

#827 Enable easy backup & duplication of IIAB microSD's (@georgejhunt  is working hard on this)
#828 Copy Content Packs to/from USB sticks/storage (@tim-moody is working hard on this)

tim-moody, 2018-06-28:
kiwix rendering zims on a usb stick that can be inserted and removed. what could possibly go wrong.

holta, 2018-06-28:
> kiwix rendering zims on a usb stick that can be inserted and removed. what could possibly go wrong.

It's worked reliably in Haiti with SD cards inserted into XO's.

So I don't personally understand the cynicism at all.

Particularly when teachers (e.g. in Mexico, and other places) have demonstrated they love this model &mdash; retaining full control of "their own" IIAB, while full well to insert USB strick/drive at the beginning of the lesson (or similar).

holta, 2018-06-28:
Potentially Massive Opportunity: saving ZIM files to cheap USB sticks (using @tim-moody's upcoming #828, in the hands of hands of the best "TLC" teacher/techies) which may well powerfully spark peer distribution of ZIM files among other IIAB's.

Always saving ZIM files to the USB media's /library/zims/* (as per #828) seems like the right way to go, for these reasons and others.

The clincher is that most IIAB's do not have IT staff, as training (AKA professional development) is the Most Expensive Part of any implementation/deployment, as we witnessed every single day in Mexico schools, during both trips to different parts of Mexico earlier this year.

_Yet educators (despite having little to no IT skills) will fight for and defend the content they themselves have gathered on their own USB sticks, propelling forward a very thoughtful "Sneakernet-of-Alexandria"._

holta, 2019-03-02:
Related: (to field operators & implementers' desire for more curatorial control)

https://github.com/iiab/iiab/issues/828#issuecomment-468933366 Copy Content Packs to/from USB sticks/storage
https://github.com/iiab/iiab/issues/1002#issuecomment-462183401 Field editing of Content Pack descriptions/logos by non-technical operators
#1530 Calibre-Web: fix cover/title thumbnails for PDF's and other docs/books
#1538 How does a teacher install a ZIM file from Windows to IIAB? (using a FAT/FAT32 USB stick if poss?)

holta, 2020-11-08:
Tangentially Related:

- #2626 "Copying zims manually without running iiab-make-kiwix-lib will cause Admin Console installation to fail (and kiwix not to work)"
- PR #2627 Kiwix [role improvements for pre-positioning of ZIM files into /library/zims/content, etc]

jvonau, 2020-11-10:
Think the metric needed to evaluate is how long does iiab-make-kiwix-lib take to complete with various numbers and sizes of zim files. Once the execution time can be best guessed then look at if it is indeed practical to explore auto re-indexing of library.xml upon insertion of a usbdrive. Altering kiwix-serve's execution line with the extra directories if detected is an option to explore also should it take a long time creating library.xml.

tim-moody, 2020-11-10:
admin console relies heavily on library.xml to know what is installed (vs. what is in the catalog), so there could be side effects of scanning for zims and more especially for abandoning library.xml in favor of passing kiwix names of directories.

manage content relies on a fixed directory structure, but it does create a library.xml on <usb>/library/zims, not when it is inserted, but when the manage content menu option is taken.

I think the metric needed is how long does the scan for zims take?

For me an even bigger question is what happens when the usb is removed.

holta, 2020-11-10:
> I think the metric needed is how long does the scan for zims take?

Yes!  Thankfully /usr/bin/iiab-make-kiwix-lib is now insanely fast, thanks to @tim-moody's improvements there in ~2019.

tim-moody, 2020-11-10:
I wouldn't say it is insanely fast, but the time is pretty linear with respect to number of changes, rather than to the total number of zims. Of course if all the zims a new to the index then it will take more time.

Another problem that just occurred to me is that we have lately had mismatches between kiwix-manage and zim version. I suppose that would have less of an effect if usbs contain older zims and kiwix-manage is backwards compatible. Any old style zims with external index would be out. Also any new zims that require a newer version of kiwix would probably cause errors.

There will also need to be logic to handle duplicates on different devices.

jvonau, 2020-11-10:
Start a new issues for RACHEL-like modules? Kolibri Channels? While there maybe some overlapping similarities let tackle these objectives individually, this issue started with zims/kiwix, lets focus on zims in this issue. I take it there would be a need to generate a menu-def for each zim also(with iiab-update-menus)?

> admin console relies heavily on library.xml to know what is installed (vs. what is in the catalog), so there could be side effects of scanning for zims and more especially for abandoning library.xml in favor of passing kiwix names of directories.
> 

I'm not suggesting library.xml be dropped or anything be altered just getting the facts laid out. Correct me if I'm mistaken but I think @holta is suggesting that kiwix-serve can be called with  `--library /library/zims/library.xml --library /somepath/library.xml` to merge both directories contents at runtime.

> manage content relies on a fixed directory structure, but it does create a library.xml on /library/zims, not when it is inserted, but when the manage content menu option is taken.
>
That I understood to be iiab-make-kiwix-lib and its supporting library under the hood that does the scanning for /library/zims/content/*.zims, is that correct?
 
> I think the metric needed is how long does the scan for zims take?

Scanning would be part of the time, rebuilding library,xml for additions and deletions when needed is additional time.
> 
> For me an even bigger question is what happens when the usb is removed.

Yes, that is a major question, think the least would be kiwix stops serving zims and at the worst a corrupted usbdrive.

> Another problem that just occurred to me is that we have lately had mismatches between kiwix-manage and zim version. I suppose that would have less of an effect if usbs contain older zims and [...]

Think the old zim format would be an issue anyway it is sliced. Based on the flow of iiab-make-kiwix-lib it appears that kiwix-manage is used when the kiwix-serve service is stopped? Is kiwix-manage backwards compatible with the older zims? 
>There will also need to be logic to handle duplicates on different devices.

I was thinking of all zims in the usbdrives's /zims/content/ directory or all zims on the sdcard, as an easier starting point.

holta, 2020-11-10:
As the person who opened this ticket almost 1/4 decade ago (where DOES the time go!) I hope that E-Books are taken very seriously, which is just a reminder that this is not just a technocratic (ZIM-only) design challenge, but also an inherently an exploration of how teachers+students+others can circulate learning materials (on their USB sticks/drives today) with less bureaucratic obstacles.

I don't have the answers, but that's the point: when a teacher inserts a USB stick/drive containing {PDF's, EPUB, Kindle, ZIM} format is detected, she/he often does that with the express intent to make those books + handouts making them available to her students.  Even after she/he removes the USB stick/drive, if there's a way to make that super-easy (eventually!?)

So there are tough but really important design conundrums around how to embrace teachers' nearly universal desires to share+cache such E-Books, PowerPoints, handouts, photos (all very compact, generally) on their various IIAB's.  And/or at least display them in the best possible ways (e.g. using something like Calibre-Web more than [Raspberry Pi's minimal Bookshelf](https://www.raspberrypi.org/blog/latest-raspberry-pi-os-update-may-2020/) software, or Calibre itself, etc ?)

In the end, most educators will _not_ care whether it's ZIM book(s) or EPUB book(s) or PDF book(s) or whatever &mdash; as she/he is a teacher/librarian trying to do her job, survive the complexity &mdash; and definitely avoid the technocratic details as much as humanly possible.  Many schools' teacher/staff rooms are the original Sneakernet after all (bringing inspiring learning materials in front of your students is basically the job description of a teacher...)

Practically speaking of course this is very hard, and this will might mean e-book caching tips and defaults + choices may need to be set during IIAB's install (and changeable afterwards too!) to help her-or-him guide the decisions as to how to save those those books -- from one IIAB/classroom to the next -- so the kids who need them get access.

cc: @metasj who may have many more thoughts on how to push this forward?!

tim-moody, 2020-11-10:
>Can teachers display their own content? 

in the faq has been answered and and implemented. I'm satisfied the answer.

jvonau, 2020-11-10:
Here is quick proof of concept of all zims on the usbkey:

1. create the /zims directory in the root of the usbstick with /content /index within /zims
2. populate the /zims/content/ with a few select zims
3. insert usbdrive into IIAB
4. find mountpoint used for the usbstick  ie /media/usb0 /media/usb1, etc.. 
5. `sudo mount  --bind /media/usb0/zims /library/zims`
6. `sudo iiab-make-kiwix-lib` which only needs to be run when the data set changes so I'm inclined not to automate this part
7. view http://box/kiwix and see if the zims are present

Step 4-5 can be automated with usbmount, I have that working, haven't mocked up stage 6 yet just concerned with the running time, and perhaps a boot time race/ordering (re)starting kiwix-serve.

holta, 2020-11-10:
As hoc ZIM files and OER2Go modules might (?) turn out be a lot easier if E-Books suffer from too many formats.

Which is sad (!) as exchanging handouts / lesson materials / PDFs (and books) is nearly the very core of what education/learning is, and those smaller files may represent the low-hanging fruit &mdash; for teachers trying to preserve/enact this collaborative workflow between classrooms "that constitutes teaching" in so many ways.

In any case, the teacher that "owns" the IIAB (in every sense, if she/he bought it) seems to want something closer to Calibre-Web than Nextcloud from what I can tell.  The ideal would be Calibre-Web (or something similar, quite simple/structured) that did little more than absorbing recent materials that s/he wants shared, week-by-week.  WITHOUT having to design an entire LMS to do this basic LRU (delete least-recently-used) caching/timestamping strategy...

If it were easy (probably not, unfortunately!!) IIAB could be set by such teachers to show the content chronologically week-by-week, of all handouts + lesson materials that were inserted week-by-week over the prior "semester".

Anyway a quite simple idea, that would be extremely powerful in the hands of many educators (and students) but needs thought to get it right.

jvonau, 2020-11-10:
Using mount --bind hides what the contents of the target /library/zims contains, a test.zim or any other contents would be hidden from view but still exists on the underlying filesystem until unmounted.

shanti-bhardwa, 2020-11-11:
Here is a scenario: Class full of student desktops (or Tablets or Smart Phones) , accessing content on Local IIAB server using their local browsers. Teacher inserts a USB stick with the content he wants to share. USB  gets mounted. On the USB, there is a specific folder such as SHARE or DOWNLOAD. Teacher  can put the documents  in this folder and they can be in any format like  PPT, PDF, MP4 . When the student clicks on the file name , it is downloaded rather than opened up in the relevant application. Is this not possible?
 
I mean it is possible to install kiwix or calibre on desktop as individual applications. Once the file is downloaded, it can be used on the Desktop or Tablet.

holta, 2020-11-11:
> Here is a scenario: Class full of student desktops (or Tablets or Smart Phones) , accessing content on Local IIAB server using their local browsers. Teacher inserts a USB stick with the content he [...]

Excellent idea.
Allowing teachers to make their intent very clear in advance.

_Similarly:_

The teacher could put ZIM files (or whatever) in a folder call "DISTRIBUTE" or "AUTOCOPY" on the USB stick — to request in advance that IIAB should _try_ to make a copy of (or cache) the Content Packs therein.

jvonau, 2020-11-11:
> Excellent idea.
> Allowing teachers to make their intent very clear in advance.

That 'SHARE or DOWNLOAD' functionality has been in place since the introduction of [usbmount](https://github.com/iiab/iiab/blob/master/roles/usb_lib/templates/mount.d/70-usb-library) way back when with the allowed directories mentioned, just need to toggle IIAB_USB_LIB_SHOW_ALL to False then the selective directory display becomes operational.

> _Similarly:_
> 
> The teacher could put ZIM files (or whatever) in a folder called "DISTRIBUTE" or "AUTOCOPY" on the USB stick — to request in advance that IIAB should _try_ to make a copy of (or cache) the Content Packs therein.

That is what the admin-console's copy function is for, there is zero change I would even attempt such a risky maneuver, great way to blindly max out the filesystem inadvertently,

holta, 2020-11-11:
> That 'SHARE or DOWNLOAD' functionality has been in place since the introduction of usbmount way back when

@shanti-bhardwa is suggesting more than that: if teachers wanting some content handled by server apps and other content handled by client apps, the question then becomes how realistic is that, towards friendly auto-handling of content, without drowning in MIME type handling etc.

Engineering a clean UX is never a walk in the park -- depends on server/client app subtleties too.

> That is what the admin-console's copy function is for

That's for large Content Packs, and a great thing too, during implementation.

Teachers in the field want something completely different -- tiny handouts replicated for all _without_ passwords.

They truly do want their proven decades-old workflows (photocopying of all the semester's tiny handouts) to be respected and hassle-free.

Looking at a teacher's life through her/his own eyes...may reveal that allocating 1GB (or some tiny fraction of the typical 128GB microSD card) for recent handouts is the real engineering challenge here...that few are yet willing to admit?  Even LRU (deletion of Least-Recently-Used files) doesn't need to be implemented.  Files could simply be deleted from a "1GB" recent handouts/files area chronologically, as new ones come in.

This kind of thing is a nearly universal request from teachers.  So I'll asking different teachers how about ground truth, around what "MVP" (minimum viable product) would be most useful for them there~

jvonau, 2020-11-11:
> @shanti-bhardwa is suggesting more than that: if teachers wanting some content handled by server apps and other content handled by client apps, the question then becomes how realistic is that, [...]

I interpreted "Teacher can put the documents in this folder and they can be in any format like PPT, PDF, MP4 . When the student clicks on the file name , it is downloaded rather than opened up in the relevant application." to mean just ad-hoc sharing, there is no mention of integration with IIAB. What is mentioned is "I mean it is possible to install kiwix or calibre on desktop as individual applications. Once the file is downloaded, it can be used on the Desktop or Tablet." as a modification to the connecting clients. 
> Engineering a clean UX is never a walk in the park -- depends on server/client app subtleties too.
>

What the opens the mime type or if the file is downloaded and saved is a function of the client connecting under the web-user's control. 
> > That is what the admin-console's copy function is for
> 
> That's for large Content Packs, and a great thing too, during implementation.
> 

In the field user changes are not part of implementation? 

> Teachers in the field want something completely different — tiny handouts replicated for all _without_ passwords.
> 
> They truly do want their proven decades-old workflows (photocopying of all the semester's tiny handouts) to be respected and hassle-free.

Seems like a chicken little argument, I fail to see what your concern is. One can toss the material in either the root of the  usbkey, /share/, or /usb/ and depending on how  IIAB_USB_LIB_SHOW_ALL is set the material is shown.

> Looking at a teacher's life through her/his own eyes...may reveal that allocating 1GB (or some tiny fraction of the typical 128GB microSD card) for recent handouts is the real engineering challenge [...]

With the material residing on the usbdrive the primary filesystem usage does not change, and can be user managed at will.

> 
> This kind of thing is a nearly universal request from teachers. So I'll keep asking different teachers how about ground truth, around what "MVP" (minimum viable product) would be most useful for them there~

Think culling old lessons should be left up to the individual teachers are they are in the best position to judge what she/he wants to preserve or delete.

jvonau, 2020-11-11:
> Ad hoc ZIM files and OER2Go modules might (?) turn out be a lot easier if E-Books suffer from too many formats.
> 

OER2GO is an order of magnitude harder to pull off, given each /library/www/html/modules/OER2GOmod/ lives under 'docroot', but doable as an all or none on the usbdrive with bindmount as shown with zims. The admin-console may have issues downloading and decompressing large content packs due to space constraints of the primary disk as in #2584 #2614. The space constraints could be temporary overcome by providing additional 'scratch space" to use via an usbdrive that has the same directory structure as in /library/downloads|working and using [bindmount](https://github.com/jvonau/iiab/commit/4302b32e0d1121650b28cd5121560a26df4a4d5e) to access the additional space automatically, restoring the original layout when the usbdrive is removed in between reboots. Don't just yank the usbkey, the machine should be shut down when the usbdrive is removed.

holta, 2020-11-11:
> culling old lessons should be left up to the individual teachers are they are in the best position to judge what she/he wants to preserve or delete

Excellent point.  Also if the teacher wants to revise a horrible typo in a handout from 2 weeks ago.  Providing an escape valve for teachers who make horrible mistakes, as they're human like everyone, and might need to yank a document later worst case.

The challenge is to keep talking to more teachers to identify what MVP (minimum viable...caching of recent handouts) teachers most want.

We know teachers absolutely despise the complexity of any LMS (e.g. Nextcloud, Moodle unless they management forces them to use it) so the challenge is whether "just save/circulate recent handouts for everyone" can be kept dead simple.

Possibly forcing these recent handouts into Calibre-Web, if teacher types prefer that makes over an archaic FTP-style listing of recent files ?

In any case: it needs thought and would definitely need to be something that does not interrupt a teacher's normal workflow, as he s/he tries to survive each 50min lesson.

holta, 2020-11-11:
> OER2GO is an order of magnitude harder to pull off

Good to know.

ZIM files and small handouts should probably be the priority targets.

Also because new ZIM files and new handouts can appear very often (e.g. daily/weekly/etc) -- whereas new RACHEL/OER2Go modules appear quite rarely (e.g. a few times per year?) so these are much less of a priority generally.

And RACHEL/OER2Go modules could always be added later, as these general Sneakernet-of-Alexandria learning community flows become validated in the field.

jvonau, 2022-04-05:
#2739