iiab/iiab #865: [i586] Kiwix-tools-src doesn't contain bin/ directory

arky, 2018-07-02:
Kiwix move fails on i586 architecture as  [kiwix-tools_linux-i586-2018-05-24](http://download.iiab.io/packages/kiwix-tools_linux-i586-2018-05-24.tar.gz ) doesn't contain a bin directory. 

TASK [kiwix : Move /tmp/kiwix-tools_linux-i586-2018-05-24/bin/* to permanent location /opt/iiab/kiwix/bin (i686)] *************************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "mv /tmp/kiwix-tools_linux-i586-2018-05-24/bin/* /opt/iiab/kiwix/bin/", "delta": "0:00:00.004278", "end": "2018-07-02 02:54:06.155023", "msg": "non-zero return code", "rc": 1, "start": "2018-07-02 02:54:06.150745", "stderr": "mv: cannot stat '/tmp/kiwix-tools_linux-i586-2018-05-24/bin/*': No such file or directory", "stderr_lines": ["mv: cannot stat '/tmp/kiwix-tools_linux-i586-2018-05-24/bin/*': No such file or directory"], "stdout": "", "stdout_lines": []}
	to retry, use: --limit @/opt/iiab/iiab/iiab-stages.retry

Bug found on Debian 9 machine running i586 kernel.

tim-moody, 2018-07-02:
perhaps the flag to indicate whether kiwix for a particular os has bin or not, kiwix_src_contains_bin, should be false by default and only true for i686 in tasks/main.yml

holta, 2018-07-02:
> perhaps the flag to indicate whether kiwix for a particular os has bin or not, kiwix_src_contains_bin, should be false by default and only true for i686 in tasks/main.yml

The flag is no longer nec as at all (I think!)

See PR #866

holta, 2018-07-03:
@arky please RSVP confirming the PR #866 (now part of IIAB 6.6/master) works on i586/i686 architectures, Thanks!

arky, 2018-07-08:
@holta PR #866 resolves the kiwix-tools problems on i586.

closed 2018-07-08 by arky