iiab/iiab #867: [Debian 9] Calibre-Serve Service Fails to start: Missing arguments "--log" "TimeoutStartSec'

arky, 2018-07-02:
The 'Calibre-Serve' services fails to start with multiple warnings. 

$ calibre-server --version
calibre-server (calibre 2.75.1)

TASK [calibre : Remove calibre.conf link if disabled (debuntu)] ****************
skipping: [127.0.0.1]

TASK [calibre : Enable Calibre service -- runs calibre-server by Kovid Goyal] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service calibre-serve: Job for calibre-serve.service failed because of unavailable resources or another system error.\nSee \"systemctl status calibre-serve.service\" and \"journalctl -xe\" for details.\n"}
	to retry, use: --limit @/opt/iiab/iiab/run-one-role.retry

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=26   changed=6    unreachable=0    failed=1   

$ sudo service calibre-serve status
● calibre-serve.service - Provides the Calibre Book Server
   Loaded: loaded (/etc/systemd/system/calibre-serve.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2018-07-02 12:26:42 EDT; 2min 25s ago
  Process: 4672 ExecStart=/usr/bin/calibre-server --daemonize --log=/var/log/calibre.log --pidfile=/var/run/calibre.pid --port=8080 /library/calibre (

Jul 02 12:26:41 box.lan calibre-server[4672]: allows you to browse you calibre library by categories. You can also
Jul 02 12:26:41 box.lan calibre-server[4672]: access an interface optimized for mobile browsers at /mobile and an
Jul 02 12:26:41 box.lan calibre-server[4672]: OPDS based interface for use with reading applications at /opds.
Jul 02 12:26:41 box.lan calibre-server[4672]: The OPDS interface is advertised via BonJour automatically.
Jul 02 12:26:41 box.lan calibre-server[4672]: Whenever you pass arguments to calibre-server that have spaces in them, enclose the arguments in quotati
Jul 02 12:26:41 box.lan calibre-server[4672]: calibre-server: error: no such option: --log
Jul 02 12:26:42 box.lan systemd[1]: calibre-serve.service: Control process exited, code=exited status=2
Jul 02 12:26:42 box.lan systemd[1]: Failed to start Provides the Calibre Book Server.
Jul 02 12:26:42 box.lan systemd[1]: calibre-serve.service: Unit entered failed state.
Jul 02 12:26:42 box.lan systemd[1]: calibre-serve.service: Failed with result 'exit-code'.

Jul 02 12:37:42 box.lan calibre-server[6584]: access an interface optimized for mobile browsers at /mobile and an
Jul 02 12:37:42 box.lan calibre-server[6584]: OPDS based interface for use with reading applications at /opds.
Jul 02 12:37:42 box.lan calibre-server[6584]: The OPDS interface is advertised via BonJour automatically.
Jul 02 12:37:42 box.lan calibre-server[6584]: Whenever you pass arguments to calibre-server that have spaces in them, enclose the arguments in quotati
Jul 02 12:37:42 box.lan calibre-server[6584]: calibre-server: error: no such option: --log
Jul 02 12:37:42 box.lan systemd[1]: calibre-serve.service: Control process exited, code=exited status=2
Jul 02 12:37:42 box.lan systemd[1]: Failed to start Provides the Calibre Book Server.
Jul 02 12:37:42 box.lan systemd[1]: calibre-serve.service: Unit entered failed state.
Jul 02 12:37:42 box.lan systemd[1]: calibre-serve.service: Failed with result 'exit-code'.
Jul 02 12:38:16 box.lan systemd[1]: [/etc/systemd/system/calibre-serve.service:11] Unknown lvalue 'TimeoutStartSec' in section 'Install'

holta, 2018-07-02:
@arky Calibre 2.75.1 is no longer supported by IIAB, for almost a year now.

Please help us find a way to make sure Calibre 3.x is installed using [py-installer.yml](https://github.com/iiab/iiab/blob/master/roles/calibre/tasks/py-installer.yml) which installs the very latest 3.26+ as called by [Lines 21-23 of roles/calibre/tasks/main.yml](https://github.com/iiab/iiab/blob/master/roles/calibre/tasks/main.yml#L21-L23).

This works on Ubuntu 18.04 LTS (and worked on Ubuntu 16.04 LTS, and should still).

ASIDE: the mass-market (Raspberry Pi) installs Calibre 3.26+ using Raspbian .deb's instead of Python, i.e. using [calibre-install-latest-rpi.sh](https://github.com/iiab/iiab/blob/master/scripts/calibre-install-latest-rpi.sh).  Those particular Raspbian .deb's will not work on Debian 9 however.  But if you really don't like the Python approach, you can try Debian 10's experimental .debs using [calibre-install-latest.sh](https://github.com/iiab/iiab/blob/master/scripts/calibre-install-latest.sh).  If you want to move in that direction, consider modifying the last 2 lines of [vars/debian-9.yml](https://github.com/iiab/iiab/blob/master/vars/debian-9.yml) from:
```
calibre_via_debs: False
calibre_via_python: True
```
...to:
```
calibre_via_debs: True
calibre_via_python: False
```

PS there are 3 stanzas to keep an eye on at the end of [roles/calibre/tasks/debs.yml (Lines 30-40)](https://github.com/iiab/iiab/blob/master/roles/calibre/tasks/debs.yml#L30-L40) &mdash; make sure the correct one is being run on Debian 9, if you go this route.

holta, 2018-07-02:
CLARIF: many of us do not have access to Debian 9.  So a big thanks in advance to @arky for trying to uncover why Calibre 3.x (3.26 for now) is oddly/apparently not installing onto Debian 9 ?  Despite the fact this (installation of Calibre 3.x) is happening properly on other OS's like Ubuntu and Raspbian.

holta, 2018-07-02:
Request from @jvonau and I:

@arky can you please post all lines from /opt/iiab/iiab/iiab-install.log relevant to calibre?  Also please confirm you were installing IIAB 6.6/master on the latest Debian 9.4 or similar?  Thanks !

Specifically that'll shed crucial light on:
1. IF [py-installer.yml](https://github.com/iiab/iiab/blob/master/roles/calibre/tasks/py-installer.yml) indeed is running on Debian 9 (as it should)
2. IF [py-installer.yml](https://github.com/iiab/iiab/blob/master/roles/calibre/tasks/py-installer.yml) is possibly failing...
   a. Possibly due to Debian 9's own Calibre 2.75.1 not quite bringing in enough dependencies?
   b. Possibly due to some compiler-like issue with Python here?

arky, 2018-07-08:
After hours of debugging, I found that somehow Calibre script or calibre debian package leave behind these broken symlinks in /usr/bin/ folder.  This confuses ansible scripts, they think calibre is present on the system.

$ ls /usr/bin/calibre*
/usr/bin/calibre            /usr/bin/calibre-parallel
/usr/bin/calibre-customize  /usr/bin/calibre-server
/usr/bin/calibredb          /usr/bin/calibre-smtp
/usr/bin/calibre-debug      /usr/bin/calibre-uninstall

For now I am going to close this bug because the correct version of calibre is install if you remove any remaining files in /usr/bin/ . 

Here is the full logs 
`TASK [0-init : Now changing FQDN] **********************************************
skipping: [127.0.0.1]

TASK [0-init : STAGE 0 HAS COMPLETED ======================================] ***
skipping: [127.0.0.1] => (item={u'option': u'is_VM', u'value': u'yes'}) 

TASK [calibre : Check if /usr/bin/calibre exists] ******************************
ok: [127.0.0.1]

TASK [calibre : Install Calibre via OS's package installer (IF /usr/bin/calibre MISSING)] ***
changed: [127.0.0.1] => (item=[u'calibre', u'calibre-bin'])

TASK [calibre : Install Calibre experimental .debs IF calibre_via_debs (AND /usr/bin/calibre WAS MISSING)] ***
skipping: [127.0.0.1]

TASK [calibre : Install Calibre via calibre-installer.py IF calibre_via_python (AND /usr/bin/calibre WAS MISSING)] ***
included: /opt/iiab/iiab/roles/calibre/tasks/py-installer.yml for 127.0.0.1

TASK [calibre : Download latest linux-installer.py from GitHub to calibre-installer.py] ***
changed: [127.0.0.1]

TASK [calibre : Check if calibre-installer.py exists in /opt/iiab/downloads] ***
ok: [127.0.0.1]

TASK [calibre : FAIL (force Ansible to exit) IF /opt/iiab/downloads/calibre-installer.py OR Internet ARE MISSING] ***
skipping: [127.0.0.1]

TASK [calibre : Run calibre-installer.py to install Calibre programs into /usr/bin] ***
changed: [127.0.0.1]

TASK [calibre : Copy template userdb to /library/calibre/users.sqlite (IF /usr/bin/calibre WAS MISSING)] ***
changed: [127.0.0.1]

TASK [calibre : Create calibre-serve.service and calibre.conf (IF /usr/bin/calibre WAS MISSING)] ***
changed: [127.0.0.1] => (item={u'dest': u'/etc/systemd/system/calibre-serve.service', u'src': u'calibre-serve.service.j2', u'mode': u'0644'})
ok: [127.0.0.1] => (item={u'dest': u'/etc/apache2/sites-available', u'src': u'calibre.conf', u'mode': u'0644'})

TASK [calibre : Force systemd to reread configs (IF /usr/bin/calibre WAS MISSING)] ***
ok: [127.0.0.1]

TASK [calibre : Stop Calibre service -- calibre-server by Kovid Goyal] *********
ok: [127.0.0.1]

TASK [calibre : Check if /library/calibre/metadata.db exists] ******************
ok: [127.0.0.1]

TASK [calibre : Create database (required since Calibre 3.x) with a sample book] ***
included: /opt/iiab/iiab/roles/calibre/tasks/create-db.yml for 127.0.0.1

TASK [calibre : Create /library/calibre (mandatory since Calibre 3.x)] *********
ok: [127.0.0.1]

TASK [calibre : Check if sample book exists in /opt/iiab/downloads] ************
ok: [127.0.0.1]

TASK [calibre : Download sample book (mandatory since Calibre 3.x)] ************
skipping: [127.0.0.1]

TASK [calibre : Check if sample book exists in /opt/iiab/downloads] ************
ok: [127.0.0.1]

TASK [calibre : Incorporate sample book into Calibre DB (mandatory since Calibre 3.x)] ***
changed: [127.0.0.1]

TASK [calibre : Make /library/calibre/metadata.db writable for Calibre client SW] ***
changed: [127.0.0.1]

TASK [calibre : Create calibre.conf link for UNTESTED http://box/books etc (debuntu)] ***
ok: [127.0.0.1]

TASK [calibre : Remove calibre.conf link if disabled (debuntu)] ****************
skipping: [127.0.0.1]

TASK [calibre : Enable Calibre service -- runs calibre-server by Kovid Goyal] ***
changed: [127.0.0.1]

TASK [calibre : Forcing apache to reread configs] ******************************
changed: [127.0.0.1]

TASK [calibre : Add 'calibre' to list of services at /etc/iiab/iiab.ini] *******
changed: [127.0.0.1] => (item={u'option': u'name', u'value': u'Calibre'})
changed: [127.0.0.1] => (item={u'option': u'description', u'value': u'"Calibre is an extremely popular personal library system for e-books."'})
changed: [127.0.0.1] => (item={u'option': u'url', u'value': u'https://raw.githubusercontent.com/kovidgoyal/calibre/master/setup/linux-installer.py'})
changed: [127.0.0.1] => (item={u'option': u'database', u'value': u'/library/calibre'})
changed: [127.0.0.1] => (item={u'option': u'port', u'value': 8080})
changed: [127.0.0.1] => (item={u'option': u'enabled', u'value': True})

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=43   changed=15   unreachable=0    failed=0   

`

closed 2018-07-08 by arky

holta, 2018-07-08:
Congrats @arky , I'm presuming "calibre --version" now shows 3.27.1?

Question: have you found http://FAQ.IIAB.IO Item 11 (Calibre passwords for teacher & students) as well as http://FAQ.IIAB.IO Item 23 (How to UPGRADE-or-REINSTALL Internet-in-a-Box apps/services like Calibre?)

arky, 2018-07-09:
@holta Yes, Calibre 3.x is installed as expected.  
 
No, I haven't looked into user accounts as this machine is yet ready for deployment.