iiab/iiab #868: Calibre-Serv service template missing "--with-library" argument

arky, 2018-07-02:
Perhaps the calibre-serve service is missing the "--with-library" argument before the path to the folder to serve with the content. 

https://github.com/iiab/iiab/blob/b04a6147e960f7e4ec6e7e9454cd46389146ff87/roles/calibre/templates/calibre-serve.service.j2#L9

holta, 2018-07-02:
Please see https://github.com/iiab/iiab/issues/867#issuecomment-401885523 going into the details of IIAB no longer supporting Calibre 2.x.

IIAB now supports Calibre 3.x's new syntax for calibre-server, as shown in [Line 9 of /etc/systemd/system/calibre-server.service](https://github.com/iiab/iiab/blob/master/roles/calibre/templates/calibre-serve.service.j2).

NOTE: the above systemd unit file flags might be enhanced in coming weeks, per #830 (auto-enable teacher's ability to add/remove/edit books) and/or @darkenvy & @nataren's request that we move to a fancier Calibre front-end known as calibre-web (#816).

holta, 2018-07-03:
I'm closing this issue as it's fundamentally a duplicate of #867.

Please @arky when you can, do respond to the central question @ https://github.com/iiab/iiab/issues/867#issuecomment-401918405.  Specifically why isn't Calibre 3.26+ installing on Debian 9?

Send us "calibre" section of  your /opt/iiab/iiab/iiab-install.log, Thanks!

closed 2018-07-03 by holta