iiab/iiab #869: [Debian 9] Sugarizer.service: Failed to parse output specifier, ignoring: syslog

arky, 2018-07-02:
Seems to be harmless parsing error. Filing for documentation purposes only. 

https://github.com/iiab/iiab/blob/b04a6147e960f7e4ec6e7e9454cd46389146ff87/roles/sugarizer/templates/sugarizer.service.j2#L12
https://github.com/iiab/iiab/blob/b04a6147e960f7e4ec6e7e9454cd46389146ff87/roles/sugarizer/templates/sugarizer.service.j2#L13

Errors
Jul 02 13:10:37 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:1] Unknown section 'unit'. Ignoring.
Jul 02 13:10:37 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:10] Failed to parse output specifier, ignoring: syslog               # Output to syslog
Jul 02 13:10:37 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:11] Failed to parse output specifier, ignoring: syslog                # Outut to syslog
Jul 02 13:10:37 box.lan systemd[1]: apt-daily-upgrade.timer: Adding 20min 11.919953s random time.
Jul 02 13:10:37 box.lan systemd[1]: apt-daily.timer: Adding 10h 56min 48.251582s random time.
Jul 02 13:18:36 box.lan systemd[1]: Reloading.
Jul 02 13:18:36 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:1] Unknown section 'unit'. Ignoring.
Jul 02 13:18:36 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:10] Failed to parse output specifier, ignoring: syslog               # Output to syslog
Jul 02 13:18:36 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:11] Failed to parse output specifier, ignoring: syslog                # Output to syslog
Jul 02 13:18:36 box.lan systemd[1]: apt-daily.timer: Adding 6h 30min 40.149135s random time.
Jul 02 13:18:36 box.lan systemd[1]: apt-daily-upgrade.timer: Adding 8min 50.496711s random time.
Jul 02 13:22:05 box.lan systemd[1]: Reloading.
Jul 02 13:22:05 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:10] Failed to parse output specifier, ignoring: syslog               # Output to syslog
Jul 02 13:22:05 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:11] Failed to parse output specifier, ignoring: syslog                # Outut to syslog
Jul 02 13:22:05 box.lan systemd[1]: apt-daily-upgrade.timer: Adding 3min 53.600637s random time.
pi@box:/opt/iiab/iiab$ sudo nano /etc/systemd/system/sugarizer.service

holta, 2018-07-02:
@georgejhunt, @jvonau & @llaske please help review below, thanks!

> Seems to be harmless parsing error. Filing for documentation purposes only.
>
> iiab/roles/sugarizer/templates/sugarizer.service.j2
>
> Line 12 in b04a614
 StandardOutput=syslog 
>
>iiab/roles/sugarizer/templates/sugarizer.service.j2
>
> Line 13 in b04a614
> StandardError=syslog 
>
> Errors
Jul 02 13:10:37 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:1] Unknown section 'unit'. Ignoring.
Jul 02 13:10:37 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:10] Failed to parse output specifier, ignoring: syslog # Output to syslog
Jul 02 13:10:37 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:11] Failed to parse output specifier, ignoring: syslog # Outut to syslog
Jul 02 13:10:37 box.lan systemd[1]: apt-daily-upgrade.timer: Adding 20min 11.919953s random time.
Jul 02 13:10:37 box.lan systemd[1]: apt-daily.timer: Adding 10h 56min 48.251582s random time.
Jul 02 13:18:36 box.lan systemd[1]: Reloading.
Jul 02 13:18:36 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:1] Unknown section 'unit'. Ignoring.
Jul 02 13:18:36 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:10] Failed to parse output specifier, ignoring: syslog # Output to syslog
Jul 02 13:18:36 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:11] Failed to parse output specifier, ignoring: syslog # Output to syslog
Jul 02 13:18:36 box.lan systemd[1]: apt-daily.timer: Adding 6h 30min 40.149135s random time.
Jul 02 13:18:36 box.lan systemd[1]: apt-daily-upgrade.timer: Adding 8min 50.496711s random time.
Jul 02 13:22:05 box.lan systemd[1]: Reloading.
Jul 02 13:22:05 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:10] Failed to parse output specifier, ignoring: syslog # Output to syslog
Jul 02 13:22:05 box.lan systemd[1]: [/etc/systemd/system/sugarizer.service:11] Failed to parse output specifier, ignoring: syslog # Outut to syslog
Jul 02 13:22:05 box.lan systemd[1]: apt-daily-upgrade.timer: Adding 3min 53.600637s random time.
pi@box:/opt/iiab/iiab$ sudo nano /etc/systemd/system/sugarizer.service

llaske, 2018-07-03:
Not sure to understand what is this "sugarizer.service".
Never seen this error. Could you be more precise about Sugarizer settings you are using ?

holta, 2018-07-03:
> Never seen this error. Could you be more precise about Sugarizer settings you are using ?

Sugarizer 1.0 is installed according to @georgejhunt PR #814 here:
https://github.com/iiab/iiab/pull/814/files

@georgejhunt modified sugarizer.ini settings as follows:
https://github.com/iiab/iiab/pull/814/files#diff-5ac0e21313f92d985f3d025ce3ca48cf

> Not sure to understand what is this "sugarizer.service".

@georgejhunt modified sugarizer.service as follows:
https://github.com/iiab/iiab/pull/814/files#diff-a8678bc2b46d75db06ebaac2a1683cbd

llaske, 2018-07-04:
Okay, I see where is the issue.
It's related to settings file.
The way Sugarizer Server look for settings file (.INI) has changed with version 1.0.
It look for a file `env/sugarizer.ini` except if something else is mentioned in the NODE_ENV variable.
Because your script indicate:
`
      Environment=NODE_ENV=production`

Sugarizer Server look for a file `env/production.ini`. So just copy/rename your `sugarizer.ini` to `production.ini` should solve the problem. I'm going to update the documentation to mention this feature.

holta, 2018-07-04:
> Because your script [indicates]:
>
> `Environment=NODE_ENV=production`

Currently this line is commented out.  As seen in both:
https://github.com/iiab/iiab/pull/814/files#diff-a8678bc2b46d75db06ebaac2a1683cbdR17
https://github.com/iiab/iiab/blob/master/roles/sugarizer/templates/sugarizer.service.j2#L17

(Do you recommend we UNcomment this line?)

> Sugarizer Server look for a file env/production.ini and can't find it here. So just copy/rename your sugarizer.ini to production.ini should solve the problem.

If nec we certainly can do that.  (What is best in the end?)  Given your safest-for-the-long-term recommendation(s) for `/opt/iiab/sugarizer-server-1.0/env/sugarizer.ini`

> I'm going to update the documentation to mention this feature.

Can you send a URL to this section of the documentation when it's ready?  Thanks so much @llaske!

llaske, 2018-07-04:
> (Do you recommend we UNcomment this line?)

At least it will show us if the issue come from the value of NODE_ENV variable.
Else you could try to print the current value of the variable.

> Can you send a URL to this section of the documentation when it's ready?

Will do it and will update this thread.

georgejhunt, 2018-07-05:
I think the first thing that needs to happen is to replicate the failure we are trying to respond to.  I have a vague memory of the sugarizer.service file having [unit] in the first line, rather than [Unit]. This fix is already in the code.
I do not think NODE_ENV is related to a parsing error of a systemd uniit file

holta, 2018-07-05:
@arky can you please confirm you've running the latest IIAB 6.6/master and/or have done a "cd /opt/iiab/iiab" and "git pull" very recently?

arky, 2018-07-08:
@holta  Here is the root cause of this problem. On Debian 9 the rsyslog service was disabled, thus causing the parsing error.    

$ sudo service rsyslog status
sudo: unable to resolve host box.lan
● rsyslog.service
   Loaded: masked (/dev/null; bad)
   Active: inactive (dead)

$ sudo service rsyslog start
sudo: unable to resolve host box.lan
Failed to start rsyslog.service: Unit rsyslog.service is masked.

Here is the workaround, if anyone hits the problem in future. 

$ sudo systemctl unmask rsyslog.service
Removed /etc/systemd/system/rsyslog.service.

Closing this bug as 'works-for-me'.

holta, 2018-07-08:
> Here is the workaround, if anyone hits the problem in future.
>
> $ sudo systemctl unmask rsyslog.service
> Removed /etc/systemd/system/rsyslog.service.

@georgejhunt & @llaske should this be implemented on Debian 9 and possibly also Ubuntu 18.04?

llaske, 2018-07-09:
I'm not familiar with systemd unit file format. BTW this comment is not pertinent since Sugarizer Server now wait for MongoDB before starting.

llaske, 2018-07-14:
FYI, Sugarizer Server [README](https://github.com/llaske/sugarizer-server/blob/master/README.md#server-settings) updated with mention of NODE_ENV variable to change settings file.

holta, 2018-07-21:
@arky I did not see this problem/warning when installing BIG-sized http://download.iiab.io/6.6 onto Debian 9.5 yesterday.

(Is it possible this is not serious?  Or am looking in the wrong place!)

holta, 2018-11-04:
@arky please re-open if this is still important!

closed 2018-11-04 by holta