iiab/iiab #871: KA Lite's mandatory English Pack takes hours/days to download (2 major US ISP's)

holta, 2018-07-03:
Please help us better understand why KA Lite installs seem to be failing for those using AT&T and Time Warner/Spectrum as their home-or-work ISP:

https://github.com/learningequality/ka-lite/issues/5575

This is the command that is failing to complete: (in a reasonable time)
```
kalite manage retrievecontentpack download en
```
Workaround for those willing to break out of IIAB's KA Lite install, when its download freezes:
```
wget http://pantry.learningequality.org/downloads/ka-lite/0.17/content/contentpacks/en.zip
kalite manage retrievecontentpack local en en.zip
```
As documented at http://FAQ.IIAB.IO #27 under ["Mandatory English Pack taking too long to download?"](http://wiki.laptop.org/go/IIAB/FAQ#KA_Lite_Administration:_What_tips_.26_tricks_exist.3F)

holta, 2018-07-03:
In the interim, we might consider revising IIAB's install scripts replacing its last/functional line...
```
kalite manage retrievecontentpack download en
```
With these 4 lines...
```
cd /opt/iiab/downloads    # Or /library/downloads if en.zip is considered stub content :)
wget http://pantry.learningequality.org/downloads/ka-lite/0.17/content/contentpacks/en.zip
kalite manage retrievecontentpack local en en.zip
#rm en.zip                # It's highly recommended you later remove this 0.9GB file!
```

holta, 2018-07-03:
FYI the following 4 lines were implemented in all 3+3 install scripts at http://download.iiab.io/6.6 :
```
cd /opt/iiab/downloads/    # Or /library/downloads if you prefer
wget http://pantry.learningequality.org/downloads/ka-lite/0.17/content/contentpacks/en.zip
kalite manage retrievecontentpack local en en.zip
# "rm /opt/iiab/downloads/en.zip" (929,916,955 bytes) after testing KA Lite!
```
We can revert to the 1-line solution if this is later resolved upstream (learningequality/ka-lite#5575).

holta, 2018-07-03:
_@deldesir & All Who Bandwidth-Constrained:_

You can AVOID downloading 0.9GB every time you install IIAB if you put a local copy of KA Lite's en.zip in /tmp (from http://pantry.learningequality.org/downloads/ka-lite/0.17/content/contentpacks/en.zip) **right before** running any of these [IIAB 6.6/master install scripts](http://download.iiab.io/6.6/).

This also means a dramatic speed-up compared to KA Lite's very problematic downloader, and financial savings for IIAB implementers in the developing world.

A reminder (STEP 5) has been placed at the top of all 3+3 install scripts in http://d.iiab.io/6.6

Here is the code at the bottom of each install script, to make sure en.zip is 929,916,955 bytes before proceeding:
```
cd /tmp/    # Or /opt/iiab/downloads or /library/downloads if you prefer
if [ -f en.zip ]; then
    if [ $(wc -c < en.zip) -ne 929916955 ]; then
        echo "ERROR: /tmp/en.zip must be 929,916,955 bytes to proceed." >&2
        exit 1
    fi
else
    wget http://pantry.learningequality.org/downloads/ka-lite/0.17/content/contentpacks/en.zip
fi
kalite manage retrievecontentpack local en en.zip
# "rm /rmp/en.zip" (929,916,955 bytes) after testing KA Lite!
```

holta, 2018-07-05:
FYI @deldesir explanations overhauled (further tightened up) within 3+3 Install Scripts @ http://d.iiab.io/6.6

holta, 2018-07-07:
FYI KA Lite's underlying issue remains (learningequality/ka-lite#5575 where Matt & I hope to continue to helping @benjaoming understand why this is happening) but the original problem is fully solved for IIAB with the 2-stage approach, so I'm closing this particular ticket here.

arky, 2018-07-08:
@holta  I noticed that "kalite manage retrievecontentpack download en" doesn't resume partially downloaded packages (akin to wget -c,  --continue ) 

Can someone verify this. This becomes an issue in networks with high latency.

holta, 2018-07-08:
> @holta I noticed that "kalite manage retrievecontentpack download en" doesn't resume partially downloaded packages (akin to wget -c, --continue )
>
> Can someone verify this. This becomes an issue in networks with high latency.

Correct.  This is one of several serious problems we've observed with "kalite manage retrievecontentpack download en"

I don't know if @benjaoming has time to fix these, as he has many other commitments.

Meantime I agree 100% everyone who can should use "wget -c http://pantry.learningequality.org/downloads/ka-lite/0.17/content/contentpacks/en.zip"

benjaoming, 2018-07-08:
@arky the `retrievecontentpack` command is already built to retry.

arky, 2018-07-23:
Am going to reopen this bug because it is impossible to download english contentpack from remote location (Phnom Penh, Cambodia)
$ $ sudo kalite manage retrievecontentpack download en
[INFO] [2018-07-24 00:31:26,657] kalite.i18n.base: Downloading content pack from http://pantry.learningequality.org/downloads/ka-lite/0.17/content/contentpacks/en.zip
(....... time out) 

$ traceroute pantry.learningequality.org
traceroute to pantry.learningequality.org (162.209.63.41), 30 hops max, 60 byte packets
 1  192.168.0.1 (192.168.0.1)  0.538 ms  0.574 ms  0.500 ms
 2  162.209.63.41 (162.209.63.41)  2.600 ms  2.533 ms  2.462 ms
 3  192.168.11.2 (192.168.11.2)  2.607 ms  2.508 ms  2.461 ms
 4  111.92.241.110 (111.92.241.110)  3.464 ms  3.646 ms  3.573 ms
 5  * * *
 6  * * *
 7  * * *
 8  * * *
 9  * * *
10  * * *
11  * * *
12  * * *
13  * * *
14  * * *
15  * * *
16  * * *
17  * * *
18  * * *
19  * * *
20  * * *
21  * * *
22  * * *
23  * * *
24  * * *
25  * * *
26  * * *
27  * * *
28  * * *
29  * * *
30  * * *

benjaoming, 2018-07-24:
> Am going to reopen this bug because it is impossible to download english contentpack from remote location (Phnom Penh, Cambodia)

That sounds like a firewall issue, though? Something being blocked! I'm sorry that we don't have a mirror structure for this.

benjaoming, 2018-07-24:
@arky just got a nice tip from @aronasorman that a Firewall is likely blocking ICMP so you're not getting anything back beyond `111.92.241.110`

Could you try just to reach `http://pantry.learningequality.org` ?

aronasorman, 2018-07-24:
@arky could you also try using tcptraceroute: https://linux.die.net/man/1/tcptraceroute

holta, 2018-08-06:
@arky please give us an update in the coming 10 days if possible!

holta, 2018-08-15:
[IIAB 6.6](https://github.com/iiab/iiab/wiki/IIAB-6.6-Release-Notes)'s August release is approaching so perhaps we need to defer this to [IIAB 6.7](https://github.com/iiab/iiab/milestone/4) in coming days.

holta, 2018-09-13:
@arky please re-open as necessary -- so far others haven't seen problems here since July, so it's looking good.

closed 2018-09-13 by holta