iiab/iiab #872: Reboot + Auto-Continue...IIAB Install Scripts (for OS updates etc)

holta, 2018-07-05:
IIAB "1-liner" Install Scripts @ http://d.iiab.io/6.6 should:

1. Install OS security updates e.g. "apt update; apt -y dist-upgrade" then...
2. Reboot then...
3. Auto-continue with IIAB installation

Potentially followed by:

4. Reboot upon IIAB install completion
5. MAYBE: Display confirmation message that IIAB install was successful??

As part of [IIAB 6.7](http://wiki.laptop.org/go/IIAB/6.7) or [7.0](https://github.com/iiab/iiab/milestone/5)?  (Unlikely this will be ready in time for IIAB 6.6 in coming weeks!)

holta, 2018-07-13:
_EVEN BETTER:_

0. IIAB's 1-line installers should pre-populate local_vars.yml based on the preferences of the implementer/operator, before (or ideally WHILE!) running "apt update; apt -y  dist-upgrade" etc.

holta, 2018-07-18:
Intimately (if indirectly) related to streamlining installs:
- #799 Standardize upgrade/reinstall of IIAB services (server apps)
- #884 Automate torrenting+installing of KA Lite videos in 7 languages

holta, 2018-07-19:
Who knows clever tricks exist to make /etc/rc.local auto-continue [install scripts](http://download.iiab.io/6.6) ?

 It's time to make this happen.

(Am far from the only one with better things to do than watch "apt update; apt -y dist-upgrade; reboot" every time ;)

holta, 2018-07-19:
During today's community/team call (http://minutes.iiab.io) @jvonau & @georgejhunt helped discuss revising [IIAB 6.6 install scripts](http://download.iiab.io/6.6/) so that a 4th hassle-free option emerges alongside beyond MIN-sized, MEDIUM-sized and BIG-sized.

That 4th option would take as input a custom [local_vars.yml](http://wiki.iiab.io/local_vars.yml) file.

It will not matter if that custom local_vars.yml file is:
- generated by a Public Web GUI
- generated by a Local TTY GUI
- generated by a tiny bash script asking you 4-quick-n-easy-questions
- generated by an in-country colleague who hand edited local_vars.yml to be regionally-polished-n-perfect
- downloaded from an official site like https://github.com/iiab/iiab/tree/master/vars
- scraped from a community forum or site like http://wiki.iiab.io/local_vars.yml
- stolen from any random Facebook posting

CORE IDEA: de-couple the generation of the IIAB definition file (local_vars.yml) from the provisioning/manufacturing process, so that community/curation experts (those most atune to community fieldback/needs) can then circulate the very best local_vars.yml files that best suit those grassroots AND global needs.

TOUGHER IDEA: this IIAB-definition file (local_vars.yml for now) might then later expand to pre-define what content needs to downloaded, along with very basic configurations?

holta, 2018-07-21:
Related:

PR #931 "/etc/iiab/local_vars.yml puts /opt/iiab/iiab/vars/local_vars.yml out of its misery"
PR #934 "softcode local_vars path" which puts `iiab_local_vars_file: /etc/iiab/local_vars.yml` in default_vars.yml

holta, 2018-08-21:
FYI the Raspbian OS from 2018-06-27 currently requires you to do the following TWICE before its kernel stabilizes... from 4.14.50-v7+ to 4.14.52-v7+ ...and then again after reboot, on to kernel 4.14.62-v7+:
```
apt update
apt -y dist-upgrade
reboot
```
Which makes automating the install sequence that much harder (not knowing in advance how many reboots might be required ;)

holta, 2018-09-03:
- Raspbian fixed its need for a double-twist 2 weeks ago
- @jvonau has ideas for implementing this auto-reboot-and-continue-installing...using a long-running systemd unit file?!

holta, 2018-09-04:
> @jvonau has ideas for implementing this auto-reboot-and-continue-installing...using a long-running systemd unit file?!

See this evolving prototype: https://github.com/iiab/iiab-factory/pull/33

holta, 2019-10-16:
Is @jvonau's PR #2008 related?

Should this issue #872 be closed now?

holta, 2020-09-03:
Good enough!

closed 2020-09-03 by holta

jvonau, 2020-09-26:
hum... 1028, 1041 still open deferred to 8.0 
--reinstall question open...