iiab/iiab #877: Package vector-based OSM maps

holta, 2018-07-07:
@georgejhunt is working on this @ https://github.com/iiab/iiab-factory/pull/32

Related Demo Sites:
http://medbox.iiab.me/modules/en-osm-omt-min/
http://medbox.iiab.me/modules/en-osm-omt-central-am/

holta, 2018-07-18:
- [ ] Excellent progress as written up in DRAFT https://github.com/iiab/iiab/wiki/IIAB-6.6-Release-Notes
- [ ] OSM instructions need review with a fresh set of eyes, to see if a newbie installer can understand, follow, succeed!

holta, 2018-08-21:
Good News Matt J will looking into this in the coming week, so that a fresh set of eyes guarantees instructions are crystal clear and simplified for non-technical and (less-technical) implementers if IIAB 6.6.

Draft starting points:
- paragraph 3 of https://github.com/iiab/iiab/wiki/IIAB-6.6-Release-Notes
- https://github.com/iiab/iiab-factory/blob/master/content/vector-tiles/README.md
- http://download.iiab.io/content/OSM/vector-tiles/

holta, 2018-08-24:
Matt J & I want to make this cleanly understandable, so everyday implementers can support themselves.

Here's a small start in that direction: #1033, iiab/iiab-factory#39

holta, 2018-08-24:
Of course few read anything (nevermind RTFM!) as they used to in the lazy era of "just Google it", so we strive for a crisp/clean UX, allowing folks to support themselves wherever possible~

STILL: implementer docs are still sometimes nec (and indexed by Google!) so Matt J offers this as an model/example &mdash; what we should strive for:

> digital ocean really got things right trying to distill down technical stuff and provide how-tos for all levels of experience. one example is https://www.digitalocean.com/community/tutorials/how-to-install-the-apache-web-server-on-ubuntu-18-04-quickstart

holta, 2018-09-08:
We really need someone to share @georgejhunt's big step forward[*] at https://2018.stateofthemap.us in Detroit October 5-7.

And learn from others there what next steps make most sense for the coming decade.  Please see the program here: https://2018.stateofthemap.us/program/

The best would be if @georgejhunt attended himself.  But if nobody else is available I can try to make the trip, which won't be easy, but possibly with @mapmeld's assistance we can re-establish the technical/community connections we need to move forward broadly with authentic community mapping.

[*] offline vector maps that are customizable for every country/continent/etc, now documented at [FAQ.IIAB.IO Question 29](http://FAQ.IIAB.IO#What_can_I_do_with_E-books_and_Internet-in-a-Box.3F)

arky, 2018-09-08:
@holta I think a lightning talk would be apt if someone could make it.  Let me know if you need some help I usually lend a hand with communication and outreach of SOTM events

georgejhunt, 2018-09-08:
This will probably be useful to me. I'd like to find someone who knows
about tweaking the style so that there is less overlap of detailed
information at zoom levels approaching, but not yet at,  the max zoom.

I plan to attend.

holta, 2018-09-08:
@georgejhunt wrote:
> I plan to attend.

That would be INCREDIBLE.  I'll try to attend Detroit's https://2018.stateofthemap.us too (Fri Oct 5 - Sunday Oct 7) hoping @mapmeld and @tim-moody might consider too?

Great opportunity to catch up with NRiddle when on location there, and a close friend who recently started as a professor at UMich (Ann Arbor is ~30 min away).

holta, 2018-09-08:
"Call for local community lightning talks, and more workshops, at State of the Map US" (in Detroit)
https://www.openstreetmap.us/2018/08/community-call/

From OSM US's blog: https://www.openstreetmap.us/blog/

holta, 2018-09-08:
> "Call for local community lightning talks, and more workshops, at State of the Map US" (in Detroit)
https://www.openstreetmap.us/2018/08/community-call/

"The call is open until September 7." but we are only hours after the deadline so, we should submit for a lightning talk regardless, as they'll likely approve &mdash; and @arky who knows them might be able to help us out getting a place on the event's program!

holta, 2018-09-13:
I just now updated http://download.iiab.io/content/OSM/vector-tiles/en-osm-omt-central-am.zip based on @georgejhunt's August 2018 improvements making maps fill the entire screen/browser (https://github.com/iiab/iiab/issues/1035).

It took me forever to figure out that zip requires the `-y` flag (as well as the `-r` flag of course) to avoid including duplicate copies of every .mbtiles files (due to symlinks) but at last [IIAB 6.6](https://github.com/iiab/iiab/wiki/IIAB-6.6-Release-Notes) is now ready to ship at long last!

Documentation revised hereunder, to save the next person all the agony I just went through:
https://github.com/iiab/iiab-factory/blob/master/content/vector-tiles/README.md

PS these vector maps can be very slow to render :-(  _@georgejhunt we need to investigate this in person in [Detroit](http://2018.stateofthemap.us/), to bring this under control in coming years!_

holta, 2018-10-29:
@georgejhunt is continuing to making headway on https://github.com/iiab/iiab-factory/pull/52/ "Merge non-overlapping OSM .mbtiles regions (into a single .mbtiles file)"

holta, 2019-06-27:
Packaging of OSM is *dramatically* improved as of IIAB 7.0 to be released very shortly now, far and away more approachable than any other release to date!

Thank you @georgejhunt.  With assistance from @tim-moody 

(Of course implementer UX's are inherently hard, so the above checklist remains useful to track diverse options that might considered for IIAB 7.1 and 7.2...)

jvonau, 2021-07-21:
in place

closed 2021-07-21 by holta