iiab/iiab #878: Recruit someone to test IIAB 6.6 on Debian 9

holta, 2018-07-07:
IIAB 6.6 is well tested but Raspbian & Ubuntu 18.04

@arky & All: can you help us find someone to install/test Internet-in-a-Box on Debian 9.4+ on x86_64 ?

holta, 2018-07-08:
Any body who can help test Debian 9 — please run http://download.iiab.io/6.6 to check if the 3 issues below are ongoing:

#865 [i586] Kiwix-tools-src doesn't contain bin/ directory
#867 calibre-serve Service Fails to start: Missing arguments "--log" "TimeoutStartSec'
#869 Sugarizer.service: Failed to parse output specifier, ignoring: syslog

holta, 2018-07-09:
While IIAB 6.6/master work well on Debian 9.4 works well (largely due to Debian 9's libraries/dependencies are years old compared to Raspbian & Ubuntu 18.04 ;) we need someone to test Debian 9.5 after it's released "Saturday July 14th" &mdash; that official announcement will be @ https://lists.debian.org/debian-stable-announce/ in about 5 days from now.

_Thanks all for helping here & where you can!_

holta, 2018-07-24:
> I will test it.

Many Thanks @deldesir &mdash; testing on Debian 9.5 seems very solid (so far!) but please also consider talking to @arky about testing Debian 10 if you can!

deldesir, 2018-07-24:
No problem @holta. Possible to provide feedback for both of them. Let's tackle this together @arky.

deldesir, 2018-07-25:
@arky Is http://cdimage.debian.org/cdimage/daily-builds/sid_d-i/current/i386/iso-cd/debian-testing-i386-netinst.iso the best one to go with?

holta, 2018-07-25:
@arky & @deldesir doesn't http://cdimage.debian.org/cdimage/daily-builds/sid_d-i/current/amd64/iso-cd/debian-testing-amd64-netinst.iso look better to test Debian 10 pre-releases on modern 64-bit hardware and VMs?

holta, 2018-07-29:
> @arky & @deldesir doesn't http://cdimage.debian.org/cdimage/daily-builds/sid_d-i/current/amd64/iso-cd/debian-testing-amd64-netinst.iso look better to test Debian 10 pre-releases on modern 64-bit hardware and VMs?

@arky & @deldesir please if you can provide an update of IIAB-on-Debian-10 in coming days (testing branch or unstable branch!) so we can finalize IIAB 6.6 Release Candidate(s) in a way that encourages others to give this a try!

(Ideally we can then raise the profile of Debian 10 on https://github.com/iiab/iiab/wiki/IIAB-Platforms on IIAB 6.6 final release early in August possibly?)

holta, 2018-08-15:
@deldesir are there any updates on Debian 10?

(Now would be a really good time as we approach [IIAB 6.6](https://github.com/iiab/iiab/wiki/IIAB-6.6-Release-Notes) final release :-)

holta, 2018-08-22:
@deldesir it's the final days of [IIAB 6.6](https://github.com/iiab/iiab/wiki/IIAB-6.6-Release-Notes) before release, and we could really use your help this week testing [IIAB 6.6/master](http://download.iiab.io/6.6/) on any version Debian 10!

(Whether the testing branch, or the unstable branch of Debian 10, just specify which one you choose to test!)

arky, 2018-08-31:
Adding this for task list for #1055

holta, 2018-12-14:
@deldesir, @floydianslips & @arky if you can help find someone to test http://d.iiab.io/6.7 on Debian 10 prior to release in coming weeks that'd be really great!

floydianslips, 2018-12-14:
Just install on Debian 9.6 with no errors on first go. Will try to find time for 10.

holta, 2018-12-14:
> Just install on Debian 9.6 with no errors on first go. Will try to find time for 10.

Amazing.  Will you use Debian 10 / Buster's `testing` branch presumably?

(FWIW @arky used Debian Sid's `unstable` branch, which is even more bleeding-edge ;)

holta, 2019-01-02:
> use Debian 10 / Buster's `testing` branch presumably?
> 
> (FWIW @arky uses Debian Sid's `unstable` branch, which is even more bleeding-edge ;)

@deldesir are you able to test http://d.iiab.io/6.7 on either Debian 10 above?

In anticipation of IIAB 6.7's final release later this January 2019 !

PS Debian 10 Alpha 4 was just released 2018-12-15 if you prefer :)
https://www.debian.org/devel/debian-installer/

holta, 2019-05-04:
Thanks all for keeping on Debian 10 "Buster" pre-releases, i.e. testing http://d.iiab.io there, as its final release is now approaching fast!

(Though IIAB 7.0 might well come first, likely around early June or so!)

holta, 2019-06-14:
Closing in favor of #1737 !

closed 2019-06-14 by holta