iiab/iiab #880: Install script fails to create /opt/iiab/downloads folder

arky, 2018-07-08:
Doing a fresh install results in "Destination /opt/iiab/downloads does not exist" on Debian 9 machine. 

 $ curl download.iiab.io/6.6/load-big.txt | sudo bash

```
TASK [include_role : dokuwiki] ************************************************************************************************************************

TASK [dokuwiki : Install DokuWiki] ********************************************************************************************************************
included: /opt/iiab/iiab/roles/dokuwiki/tasks/install.yml for 127.0.0.1

TASK [dokuwiki : Download DokuWiki software] **********************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Destination /opt/iiab/downloads does not exist"}
	to retry, use: --limit @/opt/iiab/iiab/iiab-stages.retry

PLAY RECAP ********************************************************************************************************************************************
127.0.0.1                  : ok=24   changed=6    unreachable=0    failed=1   

```

holta, 2018-07-08:
@arky is it possible you had /etc/iiab/iiab.env already in place?  @jvonau's code (below) currently does not permit this, assuming a clean OS.

Please check in /opt/iiab/iiab/iiab-install.log if "Create various library directories" ran.

This code (in Stage 0!) has always run on Debian 9 in the past:
https://github.com/iiab/iiab/blob/master/roles/0-init/tasks/fl.yml#L12 called by
https://github.com/iiab/iiab/blob/master/roles/0-init/tasks/first_run.yml#L2 called by
https://github.com/iiab/iiab/blob/master/roles/0-init/tasks/main.yml#L16

holta, 2018-07-08:
Context: {{ downloads_dir }} is set to /opt/iiab/downloads for fl.yml here:

https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L14-L18

arky, 2018-07-08:
@holta You are right! The problem was instead the presence of "/etc/iiab". 

Going to close this bug as invalid for now. Thanks!

holta, 2018-07-19:
@kananigit and/or Matt J. also fell in this trap.

_Underlying Dilemma: precocious implementers think that "rm -rf /opt/iiab" means they're starting fresh &mdash; not realizing they also need to "rm -rf /etc/iiab"  for a (somewhat more!) clean start._

3 OPTIONS that we should TRY to communicate to Linux gurus: (but where do we alert them, where they will Actually Read It ??)
1. "rm -rf /opt/iiab/iiab" or,
2. "rm /opt/iiab /etc/iiab" or,
3. install a completely clean OS, so databases, /etc & everything are all truly clean!!

PS of course make a backup copy of your IIAB's /opt/iiab/iiab/vars/local_vars.yml a priori :)

holta, 2018-08-09:
The underlying awareness dilemma was partly solved in late July by moving /opt/iiab/iiab/vars/local_vars.yml to /etc/iiab/local_vars.yml

As a result: implementers are now/increasingly much more aware that /opt/iiab AND /etc/iiab BOTH need to be deleted, when trying to get back closer/towards a virgin OS.

holta, 2018-08-09:
@georgejhunt reminds us that existence of the file `/etc/iiab/iiab.env` is what prevents folders like /opt/iiab/downloads from being created: https://github.com/iiab/iiab/blob/master/roles/0-init/tasks/main.yml#L3-L17

@jvonau & @tim-moody remind us that they would prefer if implementers NOT delete /etc/iiab/iiab.ini unless absolutely necessary.  The reason is that it stores important history of that system's evolution.

_(Again: as stated several times on this ticket...better to install a completely clean OS if that's what you really want :-)_

holta, 2018-08-11:
Related Discussion on various redesign dreams around iiab.ini and iiab.env:

#796 "Rapid networking & other IIAB config changes that can't wait ~20min for Ansible"

closed 2019-07-26 by holta