iiab/iiab #884: Automate torrenting+installing of KA Lite videos in 7 languages

holta, 2018-07-10:
@arky & All: The overwhelming majority of IIAB implementers and schools want the download/installation of videos+thumbnails to be fully automated from http://pantry.learningequality.org/downloads/ka-lite/0.17/content/ (currently holds 7 languages, English is the largest at ~44GB) into KA Lite.  For their main language(s).

As microSD cards are plenty big enough (and affordable enough) to hold every compressed KA video in 2018.

So I've started a conversation with Learning Equality to see if they have suggestions as to how best to proceed and collaborate here, e.g. if for example this selection among the 7 cuurently torrentable KA languages (to be downloaded using above URL torrents) can be made increasingly foolproof AND future-proof ?

Applying to both KA Kite & Kolibri ?

And ideally applicable to other popular/bulk content that is yet to appear ?

Stayed tuned!  Keep in touch all interested in working towards this very-high-leverage install automation..as first discussed with @tim-moody earlier in 2018.

tim-moody, 2018-07-10:
adm cons should probably do this instead of just opening the kalite gui.

holta, 2018-07-10:
> adm cons should probably do this

Also /opt/iiab/iiab/vars/local_vars.yml can fully specify the primary (if not secondary) language, so that "overnight" installs become fully automated.

> instead of just opening the kalite gui

Currently KA Lite's GUI is useless to download/torrent compressed vids.  KA Lite's GUI is however very useful for those schools/implementers that want to REMOVE videos from the ~44GB English set.

arky, 2018-07-12:
I think this is easily done with systemd bittorrent daemon. The daemon could download and copy the finished download to the KA-Lite library.
 
The daemon could be triggered by ansible install commands or triggered by config file or download folder.

jamalex, 2018-07-13:
Heads up that in Kolibri, all KA videos are already pre-compressed, so the entire English channel is 36.5 GB, when downloaded through the regular methods (UI or management commands). It's also more frequently updated than KA Lite content, and hence torrents would get stale quickly (and hence dilute the seeder pool), making torrents less than optimal.

One option we've been looking at is https://ipfs.io (since it is file-level seeding, so even when the channel is updated, you don't lose seeders for existing files). We have also discussed integrating it with Kolibri more directly (when available), as an alternative "download source".

jamalex, 2018-07-13:
Also, we would now recommend not installing KA Lite by default, and instead loading only Kolibri, including the KA channel and whatever other content is desired. This will create a more unified experience for users, give lots more tools, and be easier to keep updated. We've written a blog post explaining why we're pushing people to migrate: https://blog.learningequality.org/startwithkolibri-58227e98dde

arky, 2018-08-16:
@holta I have cobbled together a systemd torrent playbook that downloads ka-lite torrents.  Please assign this bug to me.

arky, 2018-08-17:
@holta The playbook is shaping up nicely. Take a peek and give me feedback 

https://github.com/arky/iiab/commits/transmission

holta, 2018-08-29:
Thanks to @arky BitTorrent downloads into KA Lite have not been partly automated, in the implementer's chosen language(s).

This is a great proof-of-concept stepping up implementer efficiencies &mdash; just now pushed into IIAB 6.6, to see how people use it &mdash; even if Kolibri might not require this in future if its CLI and IPFS obviate the need for some of this?!

arky, 2018-09-21:
@holta Please kindly close this bug as fixed. https://github.com/iiab/iiab/pull/1044

holta, 2018-09-21:
> @holta Please kindly close this bug as fixed. #1044

Done.

Let's now transition the conversation around critical next steps to #1078 (BitTorrent and/or IPFS for education-centric grassroots-driven Sneakernets-of-Alexandria...)

closed 2018-09-21 by holta