iiab/iiab #902: [Debian 10]  ERROR! vars file vars/{{ ansible_local.local_facts.os_ver }}.yml was not found on the Ansible Controller.

arky, 2018-07-15:
### Expected behavior and actual behavior.
Installation of IIAB to start 
### Steps to reproduce the problem.
On Debian 10 

curl download.iiab.io/6.6/load-big.txt | sudo bash 

### Specifications like the version of the IIAB, operating system version, or hardware details.
Debian 10 

$ cat /opt/iiab/iiab/iiab-install.log 
2018-07-15 04:23:58,451 passlib.registry registered 'md5_crypt' handler: <class 'passlib.handlers.md5_crypt.md5_crypt'>
2018-07-15 04:24:31,763 p=6796 u=root |  127.0.0.1 | SUCCESS => {
    "ansible_facts": {
        "ansible_all_ipv4_addresses": [
            "192.168.0.103"
        ], 
        "ansible_all_ipv6_addresses": [], 
        "ansible_apparmor": {
            "status": "enabled"
        }, 
        "ansible_architecture": "i386", 
        "ansible_bios_date": "04/13/2007", 
        "ansible_bios_version": "F.26", 
        "ansible_cmdline": {
            "BOOT_IMAGE": "/boot/vmlinuz-4.16.0-2-686-pae", 
            "quiet": true, 
            "ro": true, 
            "root": "UUID=dd873783-f175-4a4f-b7b9-115e59c35009"
        }, 
        "ansible_date_time": {
            "date": "2018-07-15", 
            "day": "15", 
            "epoch": "1531643071", 
            "hour": "04", 
            "iso8601": "2018-07-15T08:24:31Z", 
            "iso8601_basic": "20180715T042431220010", 
            "iso8601_basic_short": "20180715T042431", 
            "iso8601_micro": "2018-07-15T08:24:31.220253Z", 
            "minute": "24", 
            "month": "07", 
            "second": "31", 
            "time": "04:24:31", 
            "tz": "EDT", 
            "tz_offset": "-0400", 
            "weekday": "Sunday", 
            "weekday_number": "0", 
            "weeknumber": "28", 
            "year": "2018"
        }, 
        "ansible_default_ipv4": {
            "address": "192.168.0.103", 
            "alias": "enp8s8", 
            "broadcast": "192.168.0.255", 
            "gateway": "192.168.0.1", 
            "interface": "enp8s8", 
            "macaddress": "00:16:d3:9c:64:d6", 
            "mtu": 1500, 
            "netmask": "255.255.255.0", 
            "network": "192.168.0.0", 
            "type": "ether"
        }, 
        "ansible_default_ipv6": {}, 
        "ansible_device_links": {
            "ids": {
                "sda": [
                    "ata-FUJITSU_MHW2080BH_PL_K10MT7427S2A", 
                    "wwn-0x500000e0404bdf66"
                ], 
                "sda1": [
[... 384 lines cut ...]
        "ansible_virtualization_role": "NA", 
        "ansible_virtualization_type": "NA", 
        "ansible_wlp5s0": {
            "active": false, 
            "device": "wlp5s0", 
            "macaddress": "2e:42:ea:70:24:cb", 
            "module": "iwl3945", 
            "mtu": 1500, 
            "pciid": "0000:05:00.0", 
            "promisc": false, 
            "type": "ether"
        }, 
        "gather_subset": [
            "all"
        ], 
        "module_setup": true
    }, 
    "changed": false
}
2018-07-15 04:24:33,724 passlib.registry registered 'md5_crypt' handler: <class 'passlib.handlers.md5_crypt.md5_crypt'>
2018-07-15 04:24:33,867 p=6877 u=root |  PLAY [all] *********************************************************************
2018-07-15 04:24:33,941 p=6877 u=root |  TASK [Gathering Facts] *********************************************************
2018-07-15 04:24:36,806 p=6877 u=root |  ok: [127.0.0.1]
2018-07-15 04:24:36,879 p=6877 u=root |  ERROR! vars file vars/{{ ansible_local.local_facts.os_ver }}.yml was not found on the Ansible Controller.
If you are using a module and expect the file to exist on the remote, see the remote_src option

holta, 2018-07-15:
@arky it looks like we need a new file /opt/iiab/iiab/vars/debian-10.yml

Can you begin to help us formulate that?

ASIDE: Debian 9.5 and Ubuntu 18.04 are now tested & looking great with http://download.iiab.io/6.6/load-big.txt !

arky, 2018-07-15:
@holta Now that I have finished the initial work on Kolibri playbook, I'll take a shot at this next week.

PR #904 (2018-07-16): [Debian 10] [Fixes #902] Generate correct os_ver for Debian 10/Buster Release 
### Fixes Bug
Allows IIAB to install on future releases of Debian 10 / Buster Release, by correctly generating os_ver variable from VERSION_ID in /etc/*release files. 
Fixes #902 
### Description of changes proposed in this pull request.
Adds a new file in /vars/debian-10.yml 
Adds a new switch routine in scripts/local_facts.fact

### Smoke-tested in operating system.
Early preview of Debian 10 / Bester / Testing 

### Mention a team member for further information or comment using @ name
@holta
commit 3716815e (2018-07-16): Merge pull request #904 from arky/Debian-10

[Debian 10] [Fixes #902] Generate correct os_ver for Debian 10/Buster Release
commit a6e8d8e7 (2018-07-16): [Debian 10]Add case switch for debian-10

Fixes #902
closed 2018-07-16 by holta