iiab/iiab #908: Kolibri playbook fails to pip install on Raspbian

holta, 2018-07-16:
@arky @jvonau is it possible Kolibri requires Python 3?

Kolibri playbook failed on Raspbian Lite (*) with "MemoryError" at the end of https://pastebin.mozilla.org/9089809

Surely the 1 GB RAM in Raspberry Pi's is meager but sufficient in the end?

(*) CONTEXT: http://download.iiab.io/6.6/load-big.txt was attempted on a fresh OS, fully updated with `apt update; apt -y dist-upgrade; reboot` on RPi 3 B +.  This error occurred regardless whether using /usr/bin or /usr/local/bin in roles/kolibri/defaults/main.yml (before and after PR #907).

QUESTION: this is currently blocking all installs on Raspberry Pi.  Is this quickly fixable &mdash; or should we change these to False/False:

https://github.com/iiab/iiab/blob/master/vars/local_vars_big.yml#L173-L174
https://github.com/iiab/iiab/blob/master/vars/local_vars_big_vpn.yml#L173-L174

Refs: #841 #903 #906 #907

arky, 2018-07-16:
@holta I have tested Kolibri with Python 2.7.15. It works as expected. 

Perhaps we need to use "pip --no-cache-dir" with Kolibri ?

arky, 2018-07-16:
Another possible solution is adding filesystem swap file to Raspberry Pi. According to python pip forum posts it prevent such out of memory errors while installing large packages. 

dd if=/dev/zero of=/swapfile bs=1024 count=524288
chown root:root /swapfile
chmod 0600 /swapfile
mkswap /swapfile
swapon /swapfile

Might be a good idea to report this problem to kolibri upstream bug tracker as well.

closed 2018-07-16 by holta

jvonau, 2018-07-17:
see that https://github.com/iiab/iiab/blob/master/roles/1-prep/tasks/raspberry_pi.yml is already creating a 500MB swapfile

holta, 2018-07-17:
> Might be a good idea to report this problem to kolibri upstream bug tracker as well.

@arky can you do that and/or should this tkt be re-opened?

In consideration of @jvonau's [observation](https://github.com/iiab/iiab/issues/908#issuecomment-405609947) that IIAB has a 500MB swap file on RPi, but this was not enough?!

arky, 2018-07-17:
@holta @jvonau I recommend leaving this kolibri bug closed. 

Let's track this pip cache out of memory issue in a new meta bug as it affects multiple packages: kolibri, kalite and OSM.

commit 5c9ad5a9 (2018-07-16): Adding 'no-cache-dir' extra argumuments

Perhaps using no-cache-dir will preview out-of-memory errors on
Raspberry Pi.

Fixes #908