iiab/iiab #911: Kolibri times out in browser on Ubuntu 18.04/Server but not Ubuntu 18.04/Desktop

holta, 2018-07-16:
### Expected behavior and actual behavior.

http://box:8009 times out on Ubuntu 18.04/Server and 16.04/Desktop, despite Kolibri's Ansible installation completing (`./runrole kolibri`).

This is extremely odd given that http://box:8009 works on 18.04/Desktop using the exact same steps!

### Steps to reproduce the problem.

On any Ubuntu machine with a very recent version of IIAB 6.6/master installed:

cd /opt/iiab/iiab
git pull
nano vars/local_vars.yml    # Ensure "kolibri_install: True" and "kolibri_enabled: True"
`./runrole kolibri` or `./iiab-install --reinstall`

Visiting http://box:8009 then causes the browser to "spin" and then time out (give up) about 15 seconds later.

_`systemctl status kolibri` shows "**active (exited)**" in green on all systems, both on those that are working (U18/Desktop, Raspbian) and on those that are not (U18/Server, U16/Desktop)._

Strangely nothing additional appears here on non-working OS's/systems after trying to visit http://box:8009 :
- more /library/kolibri/kolibri.log

FYI nothing additional appears on ANY of the OS's (working and non-working) after trying to visit http://box:8009 when checking logs like:
- more /library/kolibri/server.log
- journalctl -u kolibri --no-pager

### Specifications like the version of the IIAB, operating system version, or hardware details.

All 3 OS's (Ubuntu 18.04/Desktop, 18.04/Server and 16.04/Desktop) are running on their own Inter NUC.

Whereas the working Raspbian Lite was fully updated using "apt update; apt -y dist-upgrade; reboot" prior to running http://download.iiab.io/6.6/load-big.txt on a RPi3 B+.

holta, 2018-07-16:
This same failure now occurs on RPi (Raspbian, which used to show http://box:8009 but now times out like the other OS's).  This failure began after I reinstalled on Raspbian, using "./iiab-install --reinstall".

I tried a 3rd installation of Kolibri using `./runrole kolibri` which did not help: http://box;8009 still spins.

The error/warning below during Ansible's attempt to install Kolibri is likely not related?  As this error/warning (below) also occurs when re-running `./runrole kolibri` on Ubuntu 18.04/Desktop &mdash; where http://box:8009 continues to work both before and afterwards:

> TASK [kolibri : Create kolibri default facility name, admin account and language] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "export KOLIBRI_HOME=\"/library/kolibri\" && \"/usr/local/bin/kolibri\" manage provisiondevice --facility \"Kolibri-in-a-Box\" --superusername \"Admin\" --superuserpassword \"changeme\" --preset \"formal\" --language_id \"en\" --verbosity 0 --noinput", "delta": "0:00:08.104487", "end": "2018-07-16 16:21:55.105698", "msg": "non-zero return code", "rc": 1, "start": "2018-07-16 16:21:47.001211", "stderr": "\u001b[37mINFO     Running Kolibri with the following settings: kolibri.deployment.default.settings.base\u001b[0m\n\u001b[33mWARNING  Facility with name Kolibri-in-a-Box already exists, not modifying preset.\u001b[0m\nTraceback (most recent call last):\n  File \"/usr/local/bin/kolibri\", line 11, in <module>\n    sys.exit(main())\n  File \"/usr/local/lib/python2.7/dist-packages/kolibri/utils/cli.py\", line 635, in main\n    manage(command, args=django_args)\n  File \"/usr/local/lib/python2.7/dist-packages/kolibri/utils/cli.py\", line 425, in manage\n    execute_from_command_line(argv=argv)\n  File \"/usr/local/lib/python2.7/dist-packages/kolibri/dist/django/core/management/__init__.py\", line 364, in execute_from_command_line\n    utility.execute()\n  File \"/usr/local/lib/python2.7/dist-packages/kolibri/dist/django/core/management/__init__.py\", line 356, in execute\n    self.fetch_command(subcommand).run_from_argv(self.argv)\n  File \"/usr/local/lib/python2.7/dist-packages/kolibri/dist/django/core/management/base.py\", line 283, in run_from_argv\n    self.execute(*args, **cmd_options)\n  File \"/usr/local/lib/python2.7/dist-packages/kolibri/dist/django/core/management/base.py\", line 330, in execute\n    output = self.handle(*args, **options)\n  File \"/usr/local/lib/python2.7/dist-packages/kolibri/core/device/management/commands/provisiondevice.py\", line 119, in handle\n    interactive=options['interactive'],\n  File \"/usr/local/lib/python2.7/dist-packages/kolibri/core/device/management/commands/provisiondevice.py\", line 72, in create_superuser\n    FacilityUser.objects.create_superuser(username, password)\n  File \"/usr/local/lib/python2.7/dist-packages/kolibri/auth/models.py\", line 496, in create_superuser\n    raise ValidationError(_('An account with that username already exists'))\ndjango.core.exceptions.ValidationError: [u'An account with that username already exists']", "stderr_lines": ["\u001b[37mINFO     Running Kolibri with the following settings: kolibri.deployment.default.settings.base\u001b[0m", "\u001b[33mWARNING  Facility with name Kolibri-in-a-Box already exists, not modifying preset.\u001b[0m", "Traceback (most recent call last):", "  File \"/usr/local/bin/kolibri\", line 11, in <module>", "    sys.exit(main())", "  File \"/usr/local/lib/python2.7/dist-packages/kolibri/utils/cli.py\", line 635, in main", "    manage(command, args=django_args)", "  File \"/usr/local/lib/python2.7/dist-packages/kolibri/utils/cli.py\", line 425, in manage", "    execute_from_command_line(argv=argv)", "  File \"/usr/local/lib/python2.7/dist-packages/kolibri/dist/django/core/management/__init__.py\", line 364, in execute_from_command_line", "    utility.execute()", "  File \"/usr/local/lib/python2.7/dist-packages/kolibri/dist/django/core/management/__init__.py\", line 356, in execute", "    self.fetch_command(subcommand).run_from_argv(self.argv)", "  File \"/usr/local/lib/python2.7/dist-packages/kolibri/dist/django/core/management/base.py\", line 283, in run_from_argv", "    self.execute(*args, **cmd_options)", "  File \"/usr/local/lib/python2.7/dist-packages/kolibri/dist/django/core/management/base.py\", line 330, in execute", "    output = self.handle(*args, **options)", "  File \"/usr/local/lib/python2.7/dist-packages/kolibri/core/device/management/commands/provisiondevice.py\", line 119, in handle", "    interactive=options['interactive'],", "  File \"/usr/local/lib/python2.7/dist-packages/kolibri/core/device/management/commands/provisiondevice.py\", line 72, in create_superuser", "    FacilityUser.objects.create_superuser(username, password)", "  File \"/usr/local/lib/python2.7/dist-packages/kolibri/auth/models.py\", line 496, in create_superuser", "    raise ValidationError(_('An account with that username already exists'))", "django.core.exceptions.ValidationError: [u'An account with that username already exists']"], "stdout": "", "stdout_lines": []}
...ignoring

holta, 2018-07-17:
SUMMARY after 4 new/fresh OS installs...2 on Raspbian on RPi 3 B+...and 1 each on Ubuntu 18.04/Server and Debian 9.5 &mdash; and in all cases "apt update; apt -y dist-upgrade; reboot" was run (as root) followed by "curl http://download.iiab.io/6.6/load-big.txt | sudo bash" (as a non-root user) :

- http://box:8009 (Kolibri's web interface) worked perfectly on Ubuntu 18.04/Server and Debian 9.5/Desktop, even after re-installing Kolibri using "./runrole kolibri" !
- I've never again been able to get http://box:8009 working on Raspbian.  While Kolibri claims to install (its Ansible playbook completes successfully) its site http://box:8009 does not load anything in the end.

QUESTION/CONCERN: why does "systemctl status kolibri" respond `active (exited)` (instead of `active (running)`) on all above OS's?

arky, 2018-07-17:
@holta  Am looking into this issue this morning. Let me see what is causing this problem. When Kolibri is installed for the first time, it needs sometime to run all the migrations. When reinstalling kolibri_provision should be turned off. 

Is there any additional logs debug.log, install.log in /library/kolibri directory. 

Last night I report two upstream bugs, Will continue looking into this.  

https://github.com/learningequality/kolibri/issues/4032

https://github.com/learningequality/kolibri/issues/4031

PR #916 (2018-07-17): Open kolibri port in iptables, for on-campus browsing on WAN side of IIAB (e.g. http://192.168.0.x:8009)
VPN access to http://10.8.0.x:8009 and http://localhost:8089 work on RPi 3, but http://192.168.0.x:8009 doesn't(*) due to iptables.

Hopefully this PR Fixes #911 ?

(*) However http://192.168.0.x:8089 seems to work on VM's and similar, where 192.168.0.x is the "only" IP address (i.e. no AP hotspot exists, or none was configured...)
closed 2018-07-17 by holta