iiab/iiab #913: Kolibri should work http://box/kolibri not just http://box:8009

holta, 2018-07-16:
Thanks @arky for investigating if you can!

When Kolibri works as part of IIAB, it is available at http://box:8089 but unfortunately not at http://box/kolibri

I'd like to update http://FAQ.IIAB.IO Item 3 ("What can I do with E-books and Internet-in-a-Box?") with much friendlier info on Kolibri teachers, in coming days if possible~

arky, 2018-07-17:
@holta box/kolibri is not implemented. How are these short urls implemented?

holta, 2018-07-17:
> @holta box/kolibri is not implemented. How are these short urls implemented?

I'm new to this unfortunately!

1. A simple example is the code that makes http://box/wordpress work:
   - https://github.com/iiab/iiab/blob/master/roles/wordpress/tasks/install.yml#L115-L136
   - Creates file [/etc/apache2/sites-available/wordpress.conf](https://github.com/iiab/iiab/blob/master/roles/wordpress/templates/wordpress.conf.j2)
   - A link from /etc/apache2/sites-enabled/wordpress.conf is then created to the above actual file

2. A better example might be http://box/nextcloud ?
   - Implemented by https://github.com/iiab/iiab/blob/master/roles/nextcloud/tasks/nextcloud_enabled.yml#L1-L32
   - Creates file [/etc/apache2/sites-available/nextcloud.conf](https://github.com/iiab/iiab/blob/master/roles/nextcloud/templates/nextcloud.conf.j2)
   - A link from /etc/apache2/sites-enabled/nextcloud.conf is then created to the above actual file

If you get stuck, let's probe @georgejhunt and @jvonau who created this framework!

As mentioned in #898 ["Fix http://box/sugarizer (worked with Sugarizer 0.7, 0.8, 0.9)"] these memorizable/mnemonic URL's have humongous advantages to educators+kids+users of all kind &mdash; across many situations &mdash; whether the underlying service is http://10.8.0.x:8009 (over VPN), or http://172.18.96.1:8009 over AP/hotspot, or http://192.168.0.x:8009 (over the school's Intranet) or http://127.0.01:8009 (at the keyboard of a teacher's RPi with screen attached).

The main reason is: smartphones/tablet/laptops always pop up in the hands of enthusiastic people even in the most rural locations &mdash; alongside traditional computer labs &mdash; and so a repertoire of rapid usability tricks (demonstrated by these short URLs) allows local teachers to FEEL, to SHARE, and to BELIEVE IN a sense of mastery AKA _grassroots community self-reliance_.

arky, 2018-07-17:
Thanks for the pointers. Am going to take a shot at implementing Kolibri as a httpd service next week.  
Please kindly assign this task to me.

arky, 2018-07-18:
@holta @jvonau Kindly review this quick and dirty patch. It needs to be smoke-tested as well. That's something for the weekend I guess. 

https://github.com/arky/iiab/commit/dbe9b24ac87af94d35b36caa2d2f3b05f6aca5fb

holta, 2018-07-18:
> @holta @jvonau Kindly review this quick and dirty patch. It needs to be smoke-tested as well. That's something for the weekend I guess.
>
> arky@dbe9b24

@georgejhunt  please review above, for Thursday's 10AM NYC community/team call (http://minutes.iiab.io) if poss!

arky, 2018-07-20:
@holta @georgejhunt  After smoke-testing the patch, it seems to me that what is needed is proxypass not mod_rewrite. 

```
<VirtualHost *:*>
    ProxyPreserveHost On
    # Servers to proxy the connection, or;
    # List of application servers:
    # Usage:
    # ProxyPass / http://[IP Addr.]:[port]/
    # ProxyPassReverse / http://[IP Addr.]:[port]/
    # Example: 
    ProxyPass /kolibri http://0.0.0.0:8009
    ProxyPassReverse /kolibri http://0.0.0.0:8009/

   #ServerName localhost
</VirtualHost>

```

tim-moody, 2018-07-21:
https://github.com/iiab/iiab/blob/master/roles/kiwix/templates/kiwix.conf.j2

never got ProxyPassReverse to rewrite urls properly

arky, 2018-07-22:
@tim-moody Am traveling this week, when I return I am going to take a shot at getting Apache2 proxypass working.

ivanistheone, 2018-08-04:
Hi All,

TL;DR: remove trailing slashes and it works

___

I'm in a remote area with 10kb/sec internet so can't test latest iiab, but I had a VM on my machine so did some basic tests using `python3.5 -m http.server 8009` as surrogate for Kolibri.

Observations and comments:

1/
Debugging/testing RewriteRule and ProxyPass stuff is tough because CHROME WILL CACHE REDIRECTS, see https://superuser.com/a/304600   I spent 1h+ making various tests different configurations and wondering why my requests never hit the server... Working with Firefox in "private mode" or chrome in "incognito" mode seems to bypass this caching.

2/
Scoping directives to `<VirtualHost *:80>` has some unholy interaction with the `RedirectMatch ^/$ /home` in `sites-enabled/iiab-homepage.conf` (/kolibri opens the python server on port 8009 but /kolibri/ was sending me to /home/) UPDATE: this seems to have been because of extra slash, so maybe not a real problem. Please ignore the next two paragraphs:

This command should be useful for debugging virtual hosts `apachectl -S`. Right now seems there is only one virtual host `box` but maybe we could split things up so there is no chance one app will interfere with another? Maybe using Location or Alias/ScriptAlias directives? (I saw some of the PHP stuff uses these, but seems to be specific because pointing to directories with php code in them, which we don't need).

While reading apache docs, I saw the directive `ServerPath`, which seems to be some sort of legacy way to configure virtual hosts based on path, e.g. /kolibri. Not sure if we need since `ProxyPass` seem to be working. In general it might be a good idea to standardize on ProxyPass instead of RewriteRules when possible because seems to be simpler, no regex and flags like `P` and `L` to mess with.

3/
This seems to work:
```
<VirtualHost *:80>
    ProxyPreserveHost   On
    ProxyPass           /kolibri          http://127.0.0.1:8009
    ProxyPassReverse    /kolibri          http://127.0.0.1:8009
</VirtualHost>
```
The only change to @arky's config is to remove the trailing slashes after the port number.

4/
BTW, adding this to the apache config is good for visibility when debugging ProxyPass and RewriteRules:
```
LogLevel            alert rewrite:trace1 proxy:trace1
```
Then use `tail -f /var/log/apache2/error.log` to see what's going on.

Haven't tested with real Kolibri, so I hope y'all will continue	from here.

ivanistheone, 2018-08-04:
Django does some fancy stuff with trailing slashes, so in case the above doesn't work you can try this:
```
<VirtualHost *:80>
RemoteIPHeader X-Forwarded-For
    ProxyPreserveHost   On
    RedirectMatch       ^/kolibri$         /kolibri/
    ProxyPass           /kolibri/          http://127.0.0.1:8009/
    ProxyPassReverse    /kolibri/          http://127.0.0.1:8009/
</VirtualHost>
```
(same as above but with explicit slashes + keep the option of going to slashless `/kolibri`)

holta, 2018-08-05:
Profound thanks @ivanistheone & @arky as this will allow us to ramp up global community support for Kolibri in [IIAB 6.6](https://github.com/iiab/iiab/wiki/IIAB-6.6-Release-Notes) ASAP, and hopefully even more so in [IIAB 6.7](https://github.com/iiab/iiab/milestone/4) later this year.

arky, 2018-08-09:
There seems to be some limitations running kolibri on a subpath "/kolibri". For IIAB  I'll reimplement this under wsgi.

https://github.com/learningequality/kolibri/issues/4127

ivanistheone, 2018-08-09:
I tested today and I see the just the ProxyPass approach to 8009 doesn't work.

I think it will be difficult to pull off the subpath `/kolibri` URL scheme for Kolibri, since we can't know all the endpoints that Kolibri might access (in current version and future versions). There is a lot of logic that assumes a `{{hostname}}:{{port}}` scheme.

My recommendation would be to just just use something like:
```
    RedirectMatch       ^/kolibri$         http://{{ iiab_hostname }}.{{ iiab_domain }}
```
and let `box.lan:8009` remain the main endpoint (see below for more robust hostname alternative)

Another reason for not changing the URL scheme is that Kolibri has [data sync functionality](https://github.com/learningequality/morango/blob/master/morango/urls.py#L5) that allows Kolibri instances to talk to each other and I'm not sure if that will work if we use `{{host}}/kolibri/api...` instead of `{{host}}:{{port}}/api...` 

@arky If you can get it to work would be great, but just saying it might be a tough one since backend + frontend assumptions.

@holta I hear you about the need for "nice" short URLs, but maybe having the redirect for `/kolibri` offers enough of a shortcut? As for the multiple possible hostames that initial request can come in on, we could use something like this to redirect to the right host:
```
<Location "/kolibri">
    Redirect "http://%{HTTP_HOST}:8009"
</Location>
```
This should work for public ip, 127.0.0.1, box.lan, etc...  **Requires Apapche 2.4.19 or later** via https://stackoverflow.com/a/40291044

holta, 2018-08-21:
With IIAB 6.6 scheduled to be released next week, I'm moving this task IIAB 6.7.

Profound thanks to @arky and @ivanistheone for your very hard work &mdash; and the great news that support for http://box/kolibri is being worked as part of [Kolibri 0.11.0](https://github.com/learningequality/kolibri/milestone/22) !

arky, 2018-08-30:
Update: It seems kolibri developers are working to untangling the url problems. Not sure when this fix will land, we'll be ready when it does. 
https://github.com/learningequality/kolibri/pull/4176

holta, 2018-09-26:
@arky can http://box/kolibri be made to work now?

(Many countries/groups are going into production with IIAB over the coming weeks, in case this is hopefully possible!)

arky, 2018-09-26:
@holta I'll take a look at the new code fixes upstream and see if this could be resolved.

arky, 2018-09-26:
@holta  We need to wait for Kolibri version 0.11(Mid Oct) for this feature to available.

holta, 2018-09-26:
> @holta We need to wait for Kolibri version 0.11(Mid Oct) for this feature to be available.

Looks promising for 0.11 (https://github.com/iiab/iiab/issues/1110) Thanks!

Aside: Hopefully Kolibri's memory usage doesn't prevent other IIAB services on 1GB RAM platforms like RPi 3 and RPi 3 B+.  (@jvonau seems concerned as mentioned on #977 so let's pleaae keep an eye on it as Kolibri 0.10.3+ / #1129 becomes 0.11+)

arky, 2018-09-26:
@holta I think such concerns are better addressed upstream.

holta, 2018-11-10:
FYI Kolibri 0.11.0 was released earlier this week and has been confirmed to install as part of IIAB onto Raspbian Lite Ubuntu 18.04 (#1110).

If @arky and/or others can now get http://box/kolibri working (building off 0.11.0's new https://github.com/learningequality/kolibri/issues/4127 functionality) that'd be a tremendous service to teacher/implements everywhere!

PR #1412 (2019-01-18): Apache2 Reverse proxy subdirectory /kolibri
### Fixes Bug

Fixes #913 
https://github.com/iiab/iiab/issues/913

### Description of changes proposed in this pull request.

Better handling of top level URI rewrites in Kolibri. Introduces a new environmental variable KOLIBRI_URL_PATH_PREFIX in Kolibri systemd unit and Kolibri Apache2 configuration file to handle vhost.  

### Smoke-tested in operating system.
Ubuntu Cosmic
Debian Sid

### Mention a team member for further information or comment using @ name
@holta
closed 2019-01-18 by holta