iiab/iiab #920: Is 500MB swap file best on RPi? [pip cache memory issues; installing Kolibri, KA Lite & OSM?]

holta, 2018-07-17:
@arky requests we monitor this ongoing situation based on:
- https://github.com/iiab/iiab/issues/908#issuecomment-405345180 "Kolibri playbook fails to pip install on Raspbian"
- #910 "Adding 'no-cache-dir' extra arguments"
- https://github.com/learningequality/kolibri/issues/4040 "On Raspberry Pi, use ''--no-cache-dir" pip option to install Kolibri"

holta, 2018-07-29:
Moving to [IIAB 6.7](http://wiki.laptop.org/go/IIAB/6.7) as we finalize IIAB 6.6 for release.

arky, 2018-08-29:
@holta After some of work it is clear to me that swap size is not the problem.  So you could change the title of the bug. (I don't have the permission to do that. Also can assign this bug to me.) 

At this point it seem that Debian are making some substantial changes to python environment, it is possible that is source of the problem. I need to talk to upstream Debian python maintainers next.  
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=880472

holta, 2018-08-29:
> swap size is not the problem.

Great that there is so much progress, establishing a cleaner foundation for Debian 10!  @jvonau & all of us can definitely learn from you, whatever you uncover here~

> So you could change the title of the bug.

What should the title be?

arky, 2018-08-29:
@holta  Well my sleep deprived brain says like 'Install Pip tools from Pypa.io for Raspbian'. It seems that pip from OS maintainers tend to be problematic with numerous bugs. 

Perhaps one possible solution would be to install pip using 'get-pip.py' in the IIAB setup? 

>get-pip.py is a bootstrapping script that enables users to install pip, setuptools, and wheel in Python environments that don't already have them. You should not directly reference the files located in this repository and instead use the versions located at https://bootstrap.pypa.io/.

holta, 2018-08-29:
Idea from @arky:

> 'Install Pip tools from Pypa.io for Raspbian'. It seems that pip from OS maintainers tend to be problematic with numerous bugs.
>
> Perhaps one possible solution would be to install pip using [...]

arky, 2018-08-29:
@holta  I checked the code and we already have some code for this. Anyone test if using this over system provided pip solves this issue?

https://github.com/iiab/iiab/blob/50b792869d65114daccfc77ab290b16295e207a2/roles/2-common/tasks/packages.yml#L75-L79

arky, 2018-08-30:
(Continuing the conversation about @jvonau question here. 
https://github.com/iiab/iiab/issues/816#issuecomment-417381511 )
> (After looking into it last year, he believes Pypa.io can dangerously muddy the waters as to who is installing which Python package versions: #920 (comment))

@jvonau It seems no-one can give me a exact reason why this particular pip cache memory error occurs. Python developers put the blame on debian maintainers and vice versa. 

Lets continue digging deep into this problem and hope find a workable solution.

The search (for a solution) continue ....

holta, 2018-09-26:
Aside: Kolibri memory / RAM issues (@jvonau concerns) mentioned in passing at https://github.com/iiab/iiab/issues/913#issuecomment-424708939

holta, 2019-07-26:
I agree with @jvonau that we haven't seen almost any `pip` issues recently, so let's close this.

closed 2019-07-26 by holta