iiab/iiab #922: Verify Kolibri 0.10.0 -> 0.10.1

holta, 2018-07-18:
If [Kolibri 0.10.1](https://github.com/learningequality/kolibri/milestone/27) is released in time for IIAB 6.6?

holta, 2018-08-07:
Kolibri 0.10.1 was released 3 days ago, so give a shout anybody who's tried IIAB 6.6/master's BIG-sized install since then (which should pip install 0.10.1).

holta, 2018-08-09:
FYI @ivanistheone our priority Kolibri ticket at this time is @arky's #913 &mdash; where many are hopeful he will succeed in reimplementing Kolibri with wsgi &mdash; so that http://box/kolibri works (not just http://box:8009 !)

tim-moody, 2018-08-09:
fwiw @ivanistheone prefers a proxy to wsgi

holta, 2018-08-22:
It's like IIAB 6.6 will be released with basic support for Kolibri 0.11.1

i.e. release of [0.11.2](https://github.com/learningequality/kolibri/milestone/28) currently appears to be many days/weeks away.

holta, 2018-08-26:
Kolibri 0.10.2 released: https://github.com/learningequality/kolibri/releases

Basic testing should confirm this is sufficient for IIAB 6.6

holta, 2018-08-26:
Basic testing of Kolibri 0.10.2 successful after "./runrole kolibri" and login to http://box:8009 &mdash; completed on Ubuntu 18.04/Server, 18.04/Desktop, 16.04, Raspbian Lite and Raspbian Desktop.

closed 2018-08-26 by holta