iiab/iiab #923: Failing URL's—parent tkt to consolidate diverse issues

holta, 2018-07-18:
Internet-in-a-Box URL's are failing in many annoying ways.

So this ticket serves to consolidate all these glitches that drive teachers and implementer crazy.

Such that the failure patterns can be attacked together &mdash; more coherently building up a knowledge base of proven techniques &mdash; for developers to implement the clean/short URL's that teachers wants on Day 1:

#111 Nextcloud not accessible from VPN
#529 Calibre 2.x images & Calibre 3.x books don't appear @ http://box/books (http://box:8080 works!)
#852 Fix http://box:3000 for those who want privacy & original/clean/short URL
#898 Fix http://box/sugarizer (worked with Sugarizer 0.7, 0.8, 0.9)
#913 http://box/kolibri should work (not just http://box:8009)

holta, 2020-04-08:
Moving this tkt to IIAB 7.2's milestone as we can always do better, but we're in good shape for IIAB 7.1's release ETA 2020-04-30 / 2020-05-01!

jvonau, 2022-04-05:
#2739

jvonau, 2022-04-05:
stale

holta, 2022-04-05:
Good enough 4 years later.

An awful lot of work has gone into this &mdash; and more adjustments will always be needed as outlined here;

- https://github.com/iiab/iiab/blob/master/roles/nginx/README.md

closed 2022-04-05 by holta