iiab/iiab #930: Nextcloud 14 ETA 2018-08-29 - revise & test playbook

holta, 2018-07-19:
Monitor release here:
https://github.com/nextcloud/server/wiki/Maintenance-and-Release-Schedule
https://github.com/nextcloud/server/milestone/31

Playbook will need revising here:
https://github.com/iiab/iiab/blob/master/roles/nextcloud/defaults/main.yml#L11

holta, 2018-08-30:
Please see @floydianslips' suggestion to clean up security warnings here: https://github.com/iiab/iiab/issues/972#issuecomment-417374774

holta, 2018-09-02:
If we get greedy (and Nextcloud actually releases on time in 72-96 hours from now) we _might_ add this right as we're publishing IIAB 6.6's final release.

holta, 2018-09-04:
FYI @floydianslips successfully installed https://download.nextcloud.com/server/prereleases/nextcloud-14.0.0RC2.tar.bz2 on IIAB 6.6/master so we're looking great for a Thursday release of both Nextcloud 14 and IIAB 6.6 !

floydianslips, 2018-09-04:
That script is not for security issues, it was for removing unneeded files that caused an error after upgrading from an existing NC instance to 13.06.

closed 2018-09-06 by holta

holta, 2018-09-06:
FYI Nextcloud 14.0.0 on Ubuntu Server 18.04 (as part of IIAB 6.6) displays this 1 error + 3 warnings + 3 recommendations = 7 total:

- **_Your data directory and files are probably accessible from the Internet. The .htaccess file is not working. It is strongly recommended that you configure your web server so that the data directory is no longer accessible, or move the data directory outside the web server document root._**

- _Use of the the built in php mailer is no longer supported. Please update your email server settings ↗._

- _The PHP memory limit is below the recommended value of 512MB._

- _Accessing site insecurely via HTTP. You are strongly adviced to set up your server to require HTTPS instead, as described in the security tips ↗._

- No memory cache has been configured. To enhance performance, please configure a memcache, if available. Further information can be found in the documentation.

- The PHP OPcache is not properly configured. For better performance it is recommended to use the following settings in the php.ini:

    opcache.enable=1
    opcache.enable_cli=1
    opcache.interned_strings_buffer=8
    opcache.max_accelerated_files=10000
    opcache.memory_consumption=128
    opcache.save_comments=1
    opcache.revalidate_freq=1

- The "Referrer-Policy" HTTP header is not set to "no-referrer", "no-referrer-when-downgrade", "strict-origin" or "strict-origin-when-cross-origin". This can leak referer information. See the W3C Recommendation ↗.

holta, 2018-09-07:
Just to confirm that @floydianslips's PR #1095 (now merged, thx!) indeed got rid of our eternal red error in http://box/nextcloud -> Settings -> Administration/Overview:

> Your data directory and files are probably accessible from the Internet. The .htaccess file is not working. It is strongly recommended that you configure your web server so that the data directory is no longer accessible, or move the data directory outside the web server document root.

(As tested on a fresh install of IIAB 6.6/master on Ubuntu Server 18.04)

PR #1093 (2018-09-06): Nextcloud 13.x -> 14.x, as 14.0.0 is now released
@floydianslips tested RC2 earlier this week, and tests are ongoing on other OS's.

Resolves #930