iiab/iiab #935: [Debian 10] Kalite-serve.service fails to start

arky, 2018-07-20:
### Expected behavior and actual behavior.
Kalite server fails to start on Debian 10. Python virtual environment bug in Debian / Ubuntu ?  

● kalite-serve.service - Provides the kalite Server
   Loaded: loaded (/etc/systemd/system/kalite-serve.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2018-07-20 09:58:37 EDT; 28s ago
  Process: 2337 ExecStart=/usr/local/kalite/venv/bin/kalite start --port=8008 (code=exited, status=1/FAILURE)

Jul 20 09:58:37 box.lan kalite[2337]:   File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/dist/cherrypy/lib/cptools.py", 
Jul 20 09:58:37 box.lan kalite[2337]:     import logging
Jul 20 09:58:37 box.lan kalite[2337]:   File "/usr/lib/python2.7/logging/__init__.py", line 26, in <module>
Jul 20 09:58:37 box.lan kalite[2337]:     import sys, os, time, cStringIO, traceback, warnings, weakref, collections
Jul 20 09:58:37 box.lan kalite[2337]:   File "/usr/lib/python2.7/weakref.py", line 14, in <module>
Jul 20 09:58:37 box.lan kalite[2337]:     from _weakref import (
Jul 20 09:58:37 box.lan kalite[2337]: ImportError: cannot import name _remove_dead_weakref
Jul 20 09:58:37 box.lan systemd[1]: kalite-serve.service: Control process exited, code=exited status=1
Jul 20 09:58:37 box.lan systemd[1]: kalite-serve.service: Failed with result 'exit-code'.
Jul 20 09:58:37 box.lan systemd[1]: Failed to start Provides the kalite Server.
~

### Steps to reproduce the problem.

(venv) root@box:/opt# cd /usr/local/kalite/venv/
(venv) root@box:/usr/local/kalite/venv# ls
bin  include  lib  local  pip-selfcheck.json  share
(venv) root@box:/usr/local/kalite/venv# virtualenv . --system-site-packages
Running virtualenv with interpreter /usr/local/kalite/venv/bin/python2
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 21, in <module>
    import logging
  File "/usr/lib/python2.7/logging/__init__.py", line 26, in <module>
    import sys, os, time, cStringIO, traceback, warnings, weakref, collections
  File "/usr/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

### Specifications like the version of the IIAB, operating system version, or hardware details.

Debian 10 /  Buster / Testing 
IIAB Version: IIAB_RELEASE=6.6

jvonau, 2018-07-23:
We install the kalite-static package that should be completely self-contained give that we use the ansible option virtualenv_site_packages: no, so I'm confused about why there is am import with the path "/usr/lib/python2.7/logging/init.py".. This needs further investigation as to the root cause given there was a bump to 17.5 last week. Can we confirm that only Debian 10 is affected?

holta, 2018-07-23:
> We install the kalite-static package that should be completely self-contained given that we use the ansible option virtualenv_site_packages: no, so I'm confused about why there is an import with the path "/usr/lib/python2.7/logging/init.py"

@arky do you understand this?

> there was a bump to 17.5 last week.

FWIW KA Lite 0.17.5 is identical to 0.17.4 except for [3 PR's](https://github.com/learningequality/ka-lite/milestone/65?closed=1) that don't do a whole heckuva lot :)

> Can we confirm that only Debian 10 is affected?

So far after 10 days 0.17.5 works flawlessly installing on Raspian, Ubuntu 16.04/18.04 and Debian 9.5 &mdash; so it appears this only affects Debian 10.

arky, 2018-07-23:
@holta I do.  Need to dig into these Python packages on Debian 10. Debugging python package conflicts is a nightmare.

holta, 2018-07-27:
Thanks @deldesir for confirming whether this problem bites you too, on Debian 10 @ #878.

holta, 2018-08-22:
Deferring to 6.7 for now.

holta, 2018-11-04:
@arky is this still happening?

Can you clarify as we're all now trying to pull together loose ends for IIAB 6.7's December Release?

holta, 2019-01-13:
No longer an issue as documented in #1386 !

closed 2019-01-13 by holta